LibreELEC/LibreELEC.tv #4431: [BUG] Build options for libretro.pcsx-rearmed are outdated and are causing the Kodi to crash

KOPRajs, 2020-06-19:
### Describe the bug

Because of the changes in the upstream libretro.pcsx-rearmed core the LibreELEC build options need to be updated.

More info can be found here:
https://github.com/kodi-game/game.libretro.pcsx-rearmed/issues/16

I suggest the following changes in the /packages/emulation/libretro-pcsx-rearmed/package.mk for both master and LE9.2 branches:

1. Replace the no longer working USE_DYNAREC=1 with DYNAREC=ari64 for the ARM platform.

2. Use the new DYNAREC=lightrec for both Generic and Aarch64 builds.

More info about the new dynarec:
https://www.libretro.com/index.php/pcsx-rearmed-now-has-dynarec-support-across-multiple-platforms/

### To Reproduce
Steps to reproduce the behavior:
1. Run any PSX game using the current version of PCSX-reARMed core (22.0.0.9.1)
2. Kodi crashes

### Informations
 - LE Version: 9.2.3
 - Hardware Platform: x86 and ARM

### Additional context

KOPRajs, 2020-06-21:
When the update is done it should be build and pushed to the official repository ASAP because the current broken version is being distributed to users through addon auto-update and breaks working installations.

CvH, 2020-06-21:
I have no games to test, can you test it ? If so what arch/project do you need ?

KOPRajs, 2020-06-21:
I personally use CoreELEC on Amlogic S922X-H, so PROJECT=Amlogic DEVICE=AMLG12 ARCH=arm from master branch should work for me. This one should be build with the DYNAREC=ari64.

We should also test some build with the new DYNAREC=lightrec. I can test the Generic x86 build.

KOPRajs, 2020-06-21:
Or if you are building from LE9.2 branch then PROJECT=Rockchip DEVICE=RK3399 ARCH=arm should work for me as well.

CvH, 2020-06-21:
Here are for Generic a Master build with DYNAREC=lightrec and a RK build for 9.2 with  DYNAREC=ari64

[master-generic-game.libretro.pcsx-rearmed-22.0.0.9.1.zip](https://github.com/LibreELEC/LibreELEC.tv/files/4809655/master-generic-game.libretro.pcsx-rearmed-22.0.0.9.1.zip)

[9.2-arm-RK-game.libretro.pcsx-rearmed-22.0.0.9.2.zip](https://github.com/LibreELEC/LibreELEC.tv/files/4809656/9.2-arm-RK-game.libretro.pcsx-rearmed-22.0.0.9.2.zip)

KOPRajs, 2020-06-21:
Unfortunately the RK build crashes for me, sorry. Could be either not compatible build for my system or a problem in the addon itself. I could try the AMLG12 build from master if you can build it.

I'll test the Generic build tomorrow.

KOPRajs, 2020-06-21:
P.S. Looking through the Makefile.libretro more closely:
1. Shouldn't we also export _BUILTIN_GPU=neon_ when we export _HAVE_NEON=1_?
2. What does _platform=aarch64_ do? I can't see it defined in the Makefile.libretro.
3. The GIT_VERSION seems to be ignored and replaced by the Makefile.libretro, right?

CvH, 2020-06-21:
I can't answer any of that questions. We have currently no real maintainer for the whole emulator stuff so we just bump versions and cross fingers :)

KOPRajs, 2020-06-21:
Hmm:

1. For every _platform_ defined in the Makefile.libretro that has _HAVE_NEON=1_ there is also the _BUILTIN_GPU=neon_. If it is not defined, it is set by _BUILTIN_GPU ?= peops_ in the end. So I would say if we don't define the platform (using the fallback to _platform=unix_) and we are exporting the _HAVE_NEON=1_ manually, we should export the _BUILTIN_GPU=neon_ as well.

2. I don't know either and I don't have anything running aarch64 to test it.

3. At least it shouldn't break anything if it gets replaced, so I would leave it for now :-)

KOPRajs, 2020-06-24:
So I've put up an emergency build environment and tried to build a proper version for Amlogic-ng.arm in CoreELEC myself so I can test it, because the Rockchip build doesn't work for me.

Here is the patch I used:
https://pastebin.com/f185VKkR

The fixed build for Amlogic-ng.arm with the above patch is here:
http://koprajs.mx-net.cz/game.libretro.pcsx-rearmed-22.0.0.9.1.zip

I can confirm that this fixes the build for me and the PCSX-reARMed is now working fine again.

I'm not sure if the aarch64 build is correct, but if it is broken, then it was already broken before this patch.

shantigilbert, 2020-06-24:
Just FYI, it should be 

`make -f Makefile.libretro DYNAREC=lightrec platform=arm64 GIT_VERSION=$PKG_VERSION` 

as using `platform=aarch64` will not build

KOPRajs, 2020-06-25:
Yeah, I'm just not sure how the different GPU plugin (unai) will affect the build, but definitely _platform=arm64_ seems better choice than _platform=aarch64_, which is not defined at all. Another option might be to omit the plarform directive completely, but someone would have to test which one is better (or actually works).

shantigilbert, 2020-06-25:
Not really tested the addon, bur I know for sure that you need to define the platform, and it has to be arm64 as otherwise it will not build.

KOPRajs, 2020-06-26:
Have you tested the build without explicitly specifying the platform? In theory it should fallback to _platform=unix_ just like for the other ARCHs and inherit the compiler options from the PROJECT settings and the _BUILTIN_GPU=neon_ from the above patch.

shantigilbert, 2020-06-27:
it does not fallback to aarch64 by itself, but I noticed there is a patch included

https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/emulation/libretro-pcsx-rearmed/patches/libretro-pcsx-rearmed-0001-aarch64-support.patch

so aarch64 will work, unless the patch is also outdated

KOPRajs, 2020-06-29:
Nice catch, I've missed the patch as well. The patch contains USE_DYNAREC=1 in its context so it should be updated, but it should be a trivial update and then it will work with the platform=aarch64 as you say.

To sum it up:
Lets apply this patch to the package.mk:
https://pastebin.com/f185VKkR

And update the aarch64 patch here:
https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/emulation/libretro-pcsx-rearmed/patches/libretro-pcsx-rearmed-0001-aarch64-support.patch

KOPRajs, 2020-07-01:
FYI: The CoreELEC already accepted the relevant part of the changes until it is fixed upstream:
https://github.com/CoreELEC/CoreELEC/commit/c4cf39650821ec6599fe620be3ffa09b33cf2070

CvH, 2020-07-26:
https://github.com/LibreELEC/LibreELEC.tv/pull/4469#issuecomment-663958857

closed 2020-08-03 by CvH

Sidordroid, 2020-09-24:
PCSX REarmed 22.0.0.9.2 addon Kodi
How to enter the BIOS of the console? And how to change the disk on this emulator?

CvH, 2020-09-24:
pls use the https://forum.kodi.tv/ for those questions, the kodi wiki is likely completely outdated