LibreELEC/LibreELEC.tv #4457: Building le-master fails

Karl-opec, 2020-06-27:
### Describe the bug

Karl-opec, 2020-06-27:
Building LE-Master fails on x86_64 host for amlogic g12 with command

THREADCOUNT=10 ARCH=arm PROJECT=Amlogic DEVICE=AMLG12 make image
fails with
`configure: error: libcurses or libncurses are required!
configure: error: /home/rudolf/src/le-ma/build.LibreELEC-AMLG12.arm-9.80-devel/build/heimdal-7.7.0/lib/libedit/configure failed for lib/libedit
`
ANy Idea how to fix?

[73.log](https://github.com/LibreELEC/LibreELEC.tv/files/4840321/73.log)

CvH, 2020-06-27:
@Karl-opec did you a complete clean build after the gcc10 merge ? 
I needed to do a `cd .../LE/` and `rm -rf build.*` that it works again.

Karl-opec, 2020-06-27:
Yes, I did.but it tries to build libedit in heimdal, despite that it is 
disabled

package.mk.

Karl-opec, 2020-07-04:
I build the chewitt branch amlogic-upstream, which acording to #4467  as a follow up of 4461 shoud fix the error. It does not it gives the same error.

`configure: error: libcurses or libncurses are required!
configure: error: /home/rudolf/src/le-amup-20200704/build.LibreELEC-AMLG12.arm-9.80-devel/build/heimdal-7.7.0/lib/libedit/configure failed for lib/libedit
FAILURE: scripts/build heimdal:host during configure_host (default)
*********** FAILED COMMAND ***********
${PKG_CONFIGURE_SCRIPT} ${HOST_CONFIGURE_OPTS} ${PKG_CONFIGURE_OPTS_HOST}
**************************************
FAILURE: scripts/build heimdal:host has failed!

The following log for this failure is available:
  /home/rudolf/src/le-amup-20200704/build.LibreELEC-AMLG12.arm-9.80-devel/.threads/logs/73.log

>>> heimdal:host seq 73 >>>
[079/266] [FAIL] build   heimdal:host

The following log for this failure is available:
  /home/rudolf/src/le-amup-20200704/build.LibreELEC-AMLG12.arm-9.80-devel/.threads/logs/73.log

Parallel build failure - see log for details. Time of failure: Sat Jul  4 11:17:24 CEST 2020
make: *** [Makefile:12: image] Fehler 1
`

chewitt, 2020-07-04:
I've done a clean build of my `amlogic-upstream` branch without issues on an Ubuntu 18.04 host. After the gcc10 bump it needs to be a clean build - dirty builds _will_ fail.

Karl-opec, 2020-07-05:
I build clean on Opensuse Tumbleweed. I tried your amlogic-upstream branch to the above mentioned gcc10 fix in place. 
I know that this not an preferred build system but in the past it has worked. For me it seems to be an issue with automake, autoconf in the heimdal building process. configure for heimdal says not to build libedit but it tries to do it and then it fails.

chewitt, 2020-07-05:
> I know that this not an preferred build system but in the past it has worked.

We provide Docker templates in `tools/docker` for people with unsupported build environments to use.

the-dreamer, 2020-08-10:
Master is also failing for me on Fedora 32 x64 host for building generic image with the same error during heimdal build

the-dreamer, 2020-08-12:
after some research i am pretty sure, this pr is fixing the issue above: https://github.com/heimdal/heimdal/pull/714/files

davidgfnet, 2020-12-04:
I've got the same issue :/

lrusak, 2020-12-04:
> I've got the same issue :/

impossible to know without some kind of log

Karl-opec, 2020-12-04:
You find a log in the first message here :[73.log]https://github.com/LibreELEC/LibreELEC.tv/files/4840318/73.log

chewitt, 2021-01-16:
Closing after #4949 bumps everything to a current state again

closed 2021-01-16 by chewitt

Karl-opec, 2021-01-17:
with the newest tree I still have the same error.
`checking for tgetent in -lcurses... no
checking for tgetent in -lncurses... no
configure: error: libcurses or libncurses are required!
configure: error: /home/rudolf/src/le-che-20210117/build.LibreELEC-AMLGX.arm-9.80-devel/build/heimdal-7.7.0/lib/libedit/configure failed for lib/libedit
FAILURE: scripts/build heimdal:host during configure_host (default)
*********** FAILED COMMAND ***********
${PKG_CONFIGURE_SCRIPT} ${HOST_CONFIGURE_OPTS} ${PKG_CONFIGURE_OPTS_HOST}
**************************************
FAILURE: scripts/build heimdal:host has failed!
`

CvH, 2021-01-17:
@Karl-opec did you tried if it happens at other LE platforms too (RPi ...)? Its very likely a problem from host side.

You could try to run it at docker too, there it should work regardless of host system.

the-dreamer, 2021-01-17:
Please have a look to my comment above. It is still valid.

CvH, 2021-01-17:
@the-dreamer whats you host system ?

the-dreamer, 2021-01-17:
Fedora, but as the host side is build by build system and is using the lib it should happend on any host

Karl-opec, 2021-01-17:
I normaly run opensuse thumbleweed. On This host system it does not build, on ubuntu in a virtualbox maschine it does. But  the-dreamer's  comments is a valid statement. I also question myself, why is this heimdal thing is not generally fixable? Perhaps their might something problamatic hidden?

the-dreamer, 2021-01-17:
See edit2 from august. Ncurses has changed and heimdal is failing because of it. Next heimdal release will fix it  actually in git it is already fixed.

Karl-opec, 2021-01-17:
So the Headline should be corrected to "**Building le-master fails - Needs lastest heimdal git to fix**"

chewitt, 2021-01-17:
@the-dreamer what is the fixing commit in current git? .. i.e. which patch can we backport until heimdal pushes next release.

the-dreamer, 2021-01-18:
the fix is happend on more commits, i hope that is sufficient for successful build: 
1. https://github.com/heimdal/heimdal/commit/872222db35384f343a8a566a56a28c4eb73d38ed
2. https://github.com/heimdal/heimdal/pull/714/commits/180fc339c5874a7d82fa509d4cc63e17d80723a9

this is, what is working on my private repo: 

```
new file mode 100644
index 0000000000..4050f21f50
@@ -0,0 +1,20 @@
+--- a/lib/libedit/configure.ac	2020-08-20 09:39:16.032995156 +0200
++++ b/lib/libedit/configure.ac	2020-08-20 09:28:09.474773594 +0200
+@@ -18,6 +18,7 @@
+ AC_USE_SYSTEM_EXTENSIONS
+ 
+ AM_INIT_AUTOMAKE
++AM_MAINTAINER_MODE
+ AC_PROG_LIBTOOL
+ 
+ # libtool -version-info
+@@ -35,7 +36,8 @@
+ 
+ AC_CHECK_LIB(curses, tgetent,,
+    [AC_CHECK_LIB(ncurses, tgetent,,
+-      [AC_MSG_ERROR([libcurses or libncurses are required!])] )] )
++      [AC_CHECK_LIB(tinfo, tgetent,,
++         [AC_MSG_ERROR([libcurses, libncurses, or tinfo are required!])] )] )] )
+ 
+ 
+ # Checks for header files.

```

Karl-opec, 2021-01-19:
I have modified the above Patch to be useable in the Build System:

```
--- heimdal-7.7.0/lib/libedit/configure.ac	2019-06-06 04:56:55.000000000 +0200
+++ heimdal-7.7.0/lib/libedit/configure.ac	2021-01-19 11:14:16.305614445 +0100
@@ -18,6 +18,7 @@ AC_CONFIG_HEADER([config.h])
 AC_USE_SYSTEM_EXTENSIONS
 
 AM_INIT_AUTOMAKE
+AM_MAINTAINER_MODE
 AC_PROG_LIBTOOL
 
 # libtool -version-info
@@ -35,8 +36,8 @@ EL_MANTYPE
 
 AC_CHECK_LIB(curses, tgetent,,
    [AC_CHECK_LIB(ncurses, tgetent,,
-      [AC_MSG_ERROR([libcurses or libncurses are required!])] )] )
-
+      [AC_CHECK_LIB(tinfo, tgetent,,
+         [AC_MSG_ERROR([libcurses, libncurses, or tinfo are required!])] )] )] )
 
 # Checks for header files.
 AC_HEADER_DIRENT
```
I placed it as `packages/devel/heimdal/patches/0001-heimdal-compile.patch` in the build system
and building on opensuse tumbleweed works again.

For me this issue is solved now.

Thanks everybody.

Karl-opec, 2021-01-31:
@Chewitt or CvH
Could you be so kind to add the above Patch to the source. It really fixes the build issues under other systems. The compiled 
code runs without issues.