LibreELEC/LibreELEC.tv #4460: Libreelec on TF300T

Che-debarras, 2020-06-28:
Hi 

I wonder if there is a version of Libreelec for asus tablet like TF300T under tegra.
I saw one for Nvidia shield and I dont see tablets of Asus in the list of download.
Thanks to katkiss, this tablet has multiwindow that I wonder if it can be compatible with Chrome Addon.

Hence my questions is:
Is there a version of libreelec for asus TF300T 
If no, is it under developpment or can you make it, please
If yes, multiwindow of CFW Katkiss can be used for chrome addon?

Thank you

chewitt, 2020-06-28:
This is not a bug report, so closing. Questions can be asked in our forum.

closed 2020-06-28 by chewitt

Che-debarras, 2020-06-28:
I know but the answer will be quick from here.
If no version available for asus tablet, it will be a waste of time to register to a forum and never connect it.
Forum should be for users of LibreElec.
Anyway, it will takes 2 minutes from your time to reply and then delete or close the issue

Bye.