LibreELEC/LibreELEC.tv #4478: [BUG]ethernet of odroid-n2 or odroid-c4 does not work in actual master build

Karl-opec, 2020-07-08:
### Describe the bug
ethernet is not working for odroid c4 and odroid-n2 boards

### Informations
 - LE master commit bbbb72699e8f
 - odroid-n2 and odroid-c4

### Additional context
removing patch155-WIP and patch 156-WIP in amlogic linux fixes the bug

chewitt, 2020-07-08:
You need to better define "not working" because I've been working from N2/C4 in recent days without issues. Note that I am also using the current 2020.04 u-boot we bundle into images: http://ix.io/2rbQ

Karl-opec, 2020-07-08:
Ethernet not working means: it does not communicate over ethernet. interface eth0 is there but it can't get an ip from dhcp it assigns the dummy ip 169.xxx.xxx.xxx to eth0. The change of phy-Mode to rgmii-id seams to break something.

chewitt, 2021-01-16:
Closing after #4949 bumps everything to a current state again

closed 2021-01-16 by chewitt