LibreELEC/LibreELEC.tv #4505: Kodi on RPi3 cannot start

k0st1x, 2020-08-02:
### Describe the bug
Starting the device it shows only "LibreELEC (official): 9.2.3" (and the logo - smthng like this https://news-cdn.softpedia.com/images/news2/happy-pi-day-raspberry-pi-3-model-b-plus-support-comes-to-the-libreelec-embedded-os-520235-2.jpg ) and stop loading.
but the device is answered over the network by ssh and by samba.
so i've looked at the logs ( \\device.ip\Logfiles\ )

it is many lines with errors like "ERROR: SQL: [MyVideos116.db] SQLite error SQLITE_CORRUPT (database disk image is malformed)"
so i've tries to rename the file ~/.kodi/userdata/Database/MyVideos116.db -> MyVideos116.db.bak (i believed that file will be recreated automatically) but it does not work.

also the strange errors like "Backtrace stopped: previous frame identical to this frame (corrupt stack?)" - i dont know, was it before or not.

my log files i've attached here.
[log-2020-08-02-19.56.57.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5012970/log-2020-08-02-19.56.57.zip)

### To Reproduce
Steps to reproduce the behavior:
1. Start the device
2. See error

### Informations
 - LE Version: 9.2.3
 - Hardware Platform: RPi3

### Additional context
moved from https://github.com/xbmc/xbmc/issues/18243

vpeter4, 2020-08-03:
Maybe your SD card got corrupted that's why you should write new image to your SD card and start from fresh. Of course you will lose all your settings.

k0st1x, 2020-08-03:
yep, i'll try

closed 2020-08-03 by k0st1x