LibreELEC/LibreELEC.tv #4523: [BUG]

pen2paper, 2020-08-12:
Updated to latest official build using manual updater in Kodi, now no image on monitor following automatic reboot. Sound coming through okay. Have attempted rebooting since, but problem persists.

### Informations
 - LE Version: most recent
 - Hardware Platform: RPi4

CvH, 2020-08-12:
from what version did you updated ?
"most recent" means 9.2.4 released today ?

pen2paper, 2020-08-12:
Yes sorry, updated using today’s release from the last most-recent.

pen2paper, 2020-08-12:
Tried another monitor, and same thing happening — initial rainbow-spectrum screen on boot, then nothing. Start-up sound on menu.

CvH, 2020-08-12:
can you manually "downgrade" to 9.2.2 ?
you can put the SD card into a cardreader and just replace the system and kernel file
did you any manual changes (overclocking, edit the config ...)?

pen2paper, 2020-08-12:
Downgraded successfully. Tried updating again and still no screen, so have reverted once more. I do recall attempting to change edid settings a while back... Can’t say whether that’s the cause, but is there any way to restore all default picture settings without a complete reinstall? Very grateful again for your help, cheers

CvH, 2020-08-12:
Do you have a empty SD card around where you could try to install LE 9.2.4 from scratch ? That would answer the question if there is something to repair or not :)

> I do recall attempting to change edid settings a while back

that is possible, if you can login with ssh you can do a `dmesg | paste` then we could have a look.

> restore all default picture settings without a complete reinstall

no idea how the edid stuff works at RPi, @HiassofT might know

CvH, 2020-08-14:
https://forum.libreelec.tv/thread/22487-black-screen-after-boot/
sound similar this time Slice3 (Rpi3)

Izowiuz, 2020-08-14:
Same problem here. Only sound and no video after updating to the 9.2.4 version - manually form the 9.2.3
Here is my dmesg output: http://ix.io/2uaN

HiassofT, 2020-08-15:
@Izowiuz your monitor seems to have 1920x1080 interlaced set as preferred resolution which then causes issues. Please try setting 720p in config.txt
```
hdmi_group=1
hdmi_mode=4
```
If this doesn't work run `tvservice -d edid.dat` and attach the edid.dat file

Izowiuz, 2020-08-15:
@HiassofT Hmm, that's weird. It is a 48l3433dg Toshiba TV. I believe it supports up to 1080p resolutions. After downgrade to 9.2.3 and quick look at the settings menu I see that it runs in 1080p mode - which is consistent with the display mode reported with the TV itself when I choose RPi4 as input source.

I will try to reupgrade later during the weekend and check different resolution settings.

danioj, 2020-08-15:
FWIW - I have just upgraded from 9.2.3 to 9.2.4 via the GUI and after the upgrade my Plasma TV (Pioneer) (which supports 1080p) indicated it was booting into 1080i mode when Kodi restarted (usually it boots to the splash in 1080i then auto switches to 1080p when the GUI comes up) which gave the “Black Screen” issue people seem to be referring to.  Without access to the GUI I cannot try and cycle the modes to see it that fixes it (normally the TV can run in 1080i too). Grabbed the 9.2.3 File from the archive and dropped it into the update folder and “downgraded”. All works as it did before the attempted upgrade to 9.2.4.

HiassofT, 2020-08-15:
@danioj could you grab the edid file as well and attach it? You can do that on 9.2.3 as well if you already downgraded, that shouldn't make a difference.

I'm currently suspecting the firmware change to allow interlaced modes which was added in the Jul 2 firmware https://github.com/raspberrypi/firmware/commit/66fc5eaac3d0af1d5a7ffa616086cbfaefd72e98 might be the culprit. With your edids I should be able to reproduce what's going on.

Izowiuz, 2020-08-15:
Here is my edid.dat from 9.2.3

[edid.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5078640/edid.zip)

danioj, 2020-08-15:
> @danioj could you grab the edid file as well and attach it? You can do that on 9.2.3 as well if you already downgraded, that shouldn't make a difference.
> 
> I'm currently suspecting the firmware change to allow interlaced modes which was added in the Jul 2 firmware [raspberrypi/firmware@66fc5ea](https://github.com/raspberrypi/firmware/commit/66fc5eaac3d0af1d5a7ffa616086cbfaefd72e98) might be the culprit. With your edids I should be able to reproduce what's going on.

Sure thing. Already downgraded but here it is.

[edid.dat.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5078665/edid.dat.zip)

imforumman, 2020-08-15:
Same issue here. 

http://ix.io/2ueQ

[edid.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5078765/edid.zip)

Shamp0o, 2020-08-16:
Same here (rpi4).

Steps to downgrade:
1. put sd card into a sd card reader of a Linux machine. There should be 2 partitions. A FAT one called LIBREELEC and an ext4 one called STORAGE.
2. Mount the STORAGE partition.
3. Download the correct LibreELEC image file (e.g. LibreELEC-RPi4.arm-9.2.3.img.gz) from the [archive](http://archive.libreelec.tv/)
4. copy the image file to the .update directory in the STORAGE partition of your sd card as root
5. Unmount sd card
6. Put sd card into media box device and boot device
7. Wait for the downgrade process to finish (black screen). Your system should restart automatically and you should once again have an image.

It should also be possible to do all of this using ssh.

danioj, 2020-08-16:
> Same here (rpi4).
>
> Steps to downgrade:
>
> 1. put sd card into a sd card reader of a Linux machine. There should be 2 partitions. A FAT one called LIBREELEC and an ext4 one called STORAGE.
> 2. Mount the [...]

Or even easier (if you have SMB enabled on your Pi), access the Rpi4 from your pc/phone/tablet etc and copy the ‘LibreELEC-RPi4.arm-9.2.3.img.gz’ file referenced in the post above into the update dir (which is shared from the user home dir). Restart the Rpi4. Done.

HiassofT, 2020-08-17:
Thanks a lot to all of you who provided edids, this helped me to reproduce the issue. As suspected it's caused by interlaced modes.

Can you please test this build https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-9.2-devel-20200817130411-96056bf.tar and report back if it works fine? Also watch out for other issues.

This build reverts the RPi4 firmware back to the June 26 version before the interlaced changes and local testing was fine with your edids.

noejoe, 2020-08-18:
I can confirm that the build from @HiassofT is working for me.
I have a brand new Pi4 8GB and had the same problem: black screen, whereas sound was there. No update in my case brand new install of LibreELEV 9.2.4 and I had this problem. With the update placed in the appropriate folder and a reboot I got to see something. Thank you very much!

graysky2, 2020-09-12:
@HiassofT - I too can confirm that [your build](https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-9.2-devel-20200817130411-96056bf.tar) fixes the issue on my RPi4.  Is there a path to downgrade the firmware until the issue is solved so that the latest commits elsewhere in the build can be applied?

graysky2, 2020-09-12:
@HiassofT - I too can confirm that [your build](https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-9.2-devel-20200817130411-96056bf.tar) fixes the issue on my RPi4.  Is there a plan to downgrade the firmware in master until the issue is solved (firmware upstream I assume)?   It looks like the downgrade only made it into the [libreelec-9.2](https://github.com/LibreELEC/LibreELEC.tv/tree/libreelec-9.2) branch.  I tried a nightly build a few days ago and the black screen issue returned.

HiassofT, 2020-09-12:
The firmware downgrade was only needed as the 4.19 kernel in LE 9.2 didn't get interlaced support, and this mismatch caused the issues.

Master with current kernel and firmware were fine here when testing - please also check current master which includes latest kernel/firmware bump PR https://github.com/LibreELEC/LibreELEC.tv/pull/4531 (this was merged yesterday evening)

graysky2, 2020-09-12:
@HiassofT - Thanks for the info.  I will try `LibreELEC-RPi4.arm-9.80-nightly-20200911-16e68ab.tar` and report back.

graysky2, 2020-09-13:
@HiassofT - I updated to [LibreELEC-RPi4.arm-9.80-nightly-20200913-c782141](https://test.libreelec.tv/LibreELEC-RPi4.arm-9.80-nightly-20200913-c782141.tar) and still have this bug.  When the RPi4 boots, I do not see the rainbow splash screen, nor do I see any output.  The monitor is on, but displaying literally all black pixels.  I can ssh into the box and see that kodi is running in `top` but again, nothing on screen.

I did see an error in dmesg relating to firmware loading:
```
# dmesg| grep -i error
[    6.173564] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
```

[Here](https://gist.github.com/graysky2/f8869803f6ea35eaf7937cd2d3ee07ff) is a link the entire out of dmesg.

I am happy to provide any output to assist in debugging.

HiassofT, 2020-09-13:
@graysky2 please try with `disable_fw_kms_setup=1`in config.txt, forcing hdmi mode to 1080p60 should work, too.

FW chooses 4kp30 which isn't supported by kms and that causes black screen with LE/kodi (not sure ATM why rainbow isn't shown)

graysky2, 2020-09-13:
@HiassofT - Adding that line to `config.txt` worked, but as you pointed out, the resolution is now limited to 1080p60.  Is there no official support for 4kp30 currently or ever?  Maybe I am misunderstanding something.  Thanks!

HiassofT, 2020-09-13:
@graysky2 4k support is still missing in the KMS driver.

I tested the version in the rpi-5.9 tree a couple of days ago (test branch is here https://github.com/HiassofT/LibreELEC.tv/tree/le10-kernel-5.9) and was able to get 4kp30 GUI output with it but 4k video playback was distorted.

It still needs some work (esp 4kp60 is completely missing), therefore we carry a patch to limit the pixel clock and thus disable 4k modes ATM - see https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux/patches/raspberrypi/linux-001-hide-4k-modes.patch

graysky2, 2020-09-13:
@HiassofT - Thanks for the info!

HiassofT, 2020-11-21:
closing as already fixed via #4527

closed 2020-11-21 by HiassofT