LibreELEC/LibreELEC.tv #4525: [BUG] German Translation not correct at the "First Launch Setup Menu"

pavll, 2020-08-15:
### Describe the bug

It is a "translation error". On the last page of the LibreELEC "First Start Setup" on the RBPi3b+ (the Wellcome-Screen with the headline "Danke" and the title "Willkommen bei LibreELEC") the error i found is near "du wirst als nächstes Kodi zu geleitet, wo du deine..." The correct part of this sentence is "du wirst als nächstes zu Kodi weiter geleitet, wo du deine..."

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'the LibreELEC Installer on the RaspberryPi 3b+ but I think this error is not hardware specific'
2. Play 'The installer set it to "German" an configure it until you reach the page with title: "Willkommen bei LibreELEC"n'
3. See error

### Informations
 - LE Version: 9.2.1
 - Hardware Platform: RPi3 b+

### Additional context

DaVukovic, 2020-10-07:
Thanks for your report and sorry for the delay. 

This will be fixed by the next language update which I just created a PR for: 

https://github.com/LibreELEC/service.libreelec.settings/pull/180

I'll close this ticket after the PR is merged

DaVukovic, 2020-10-12:
merged

closed 2020-10-12 by DaVukovic