LibreELEC/LibreELEC.tv #4526: [BUG] need to re-select HDMI input, or got black screen after LibreELEC splash logo

efa, 2020-08-17:
### Describe the bug

Need to re-select the TV HDMI input to see the Kodi menu,
because at boot after the LibreELEC splash logo, I always got a permanent black screen

### To Reproduce
Steps to reproduce the behavior:
1. Boot the Raspberry Pi to LibreELEC
2. wait LibreElec splash logo
3. after a while, the TV become always black, and start a static noise
4. after some seconds, I can ear a 'ding' from TV
5. at this time I can press the TV input selector, and re-select the HDMI input (that is already selected), to see the Kodi menu

### Informations
 - LE Version:  v9.2.4 Leia
 - Hardware Platform: RPi1B with MPEG2 decoder unlocked

### Additional context

TV: Philips
Model ID: 22PFL3403D/10

efa, 2020-08-17:
Kodi resolution: 1680x1050p@60.00Hz

ChadTaljaardt, 2020-12-01:
@efa Did you ever find a workaround for this? I am having a similar issue in the issue above ^^

efa, 2020-12-01:
as the bug happen only when the same resolution is used for both LibreElec boot and Kodi, and I want to keep the max resolution in Kodi, as a work around I forced a lower resolution in LibreElec boot

efa, 2020-12-01:
"<LibreElecBootPartition>/config.txt"
hdmi_group=2 # DMT (Display Monitor Timings) standard used by monitors
hdmi_mode=46 # 1440x900 	60Hz 	16:10 	reduced blanking

efa, 2020-12-01:
take note that in my case after the rainbow and boot loader menù, I can always see the LibreElec splash logo, THEN after 40 s, the TV become black. I think your case is different, maybe related to 8 GB of ram and new RPi4 model, I'm using a RPi1

ChadTaljaardt, 2020-12-02:
Ill try test it today and will report back. Thanks for the info!

ChadTaljaardt, 2020-12-09:
Small update @efa, i tried the nightly build of libreelec and it fixed my issue, might be worth it to you to try it too :)

efa, 2020-12-09:
my RPi1 is a media player in holiday house, I can't reach it probably till next month, covid19 rule

chewitt, 2021-03-08:
I'm going to close this ticket as there will be no more LE 9.2 releases to push a fix into (even if the issue was understood) and there are no plans to release RPi0/1 images for LE10 - as the completely reworked video pipeline is not suited to the low RAM size on older boards (1GB is needed).

closed 2021-03-08 by chewitt