LibreELEC/LibreELEC.tv #4541: [BUG] tini is misconfigured for docker-init

ilya-fedin, 2020-08-26:
### Describe the bug
tini is configured with `-DMINIMAL=ON` what makes it unusable for docker
https://github.com/LibreELEC/LibreELEC.tv/blob/a709f0e43efe581c60b3aea3ee909d8ff3e5cb9c/packages/addons/addon-depends/tini/package.mk#L14

As a result, all containers that created with `docker run --init` return `exec -- failed: No such file or directory` error. When I googled, I found similar error on CentOS bugtracker: https://bugs.centos.org/view.php?id=15320

### To Reproduce
Steps to reproduce the behavior:
1. Create a container with `docker run --init`

### Informations
 - 9.2.3
 - RPi4

### Log file

### Additional context

heitbaum, 2020-10-06:
#4573 may fix this issue.

ilya-fedin, 2020-10-06:
> Fix #4573

How is this PR related to docker run --init? As I can see, it has only fixes for kodi-related scripts.

PR #4557 (2020-10-24): Don't define -DMINIMAL=ON for tini
Fixes #4541

Could that be backported to 9.2?
closed 2020-10-24 by CvH