LibreELEC/LibreELEC.tv #4547: Testing LibreELEC-settings addon

mglae, 2020-09-06:
This is a request for testing.

Changes to LibreELEC-settings addon can be tested by installing an update into /storage. No need to build an image for every platform.

### Installation
```shell
systemctl stop kodi
rm -rf /storage/.kodi/addons/service.libreelec.settings
unzip -q service.libreelec.settings-VERSION.zip -d /storage/.kodi/addons/
systemctl start kodi
```
### Deinstallation
```shell
systemctl stop kodi
rm -rf /storage/.kodi/addons/service.libreelec.settings
systemctl start kodi
```

### Test Installation Wizard
```shell
systemctl stop kodi
rm -f /storage/.kodi/userdata/addon_data/service.libreelec.settings/oe_settings.xml
systemctl start kodi
```

## Addon for LibreELEC 9.2
[service.libreelec.settings-9.0.0-20200906.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5180425/service.libreelec.settings-9.0.0-20200906.zip)

The file includes LibreELEC/service.libreelec.settings#164, LibreELEC/service.libreelec.settings#166, LibreELEC/service.libreelec.settings#168 and LibreELEC/service.libreelec.settings#172.

## Addon for LibreELEC 9.8 (current master)
[service.libreelec.settings-9.8.0-20200906.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5180429/service.libreelec.settings-9.8.0-20200906.zip)

The file includes LibreELEC/service.libreelec.settings#160, LibreELEC/service.libreelec.settings#165, LibreELEC/service.libreelec.settings#167, LibreELEC/service.libreelec.settings#169 and LibreELEC/service.libreelec.settings#171

edit4ever, 2020-09-15:
tested and merged

mglae, 2020-09-15:
PR175 added.

chewitt, 2021-03-08:
I think we can close this out now :)

mglae, 2021-03-08:
I still plan to keep the sources updated - although they are currently one week outdated.