LibreELEC/LibreELEC.tv #4563: [BUG] /usr/share/kodi/addons/service.libreelec.settings/oe.py/oe.py raise exception

tantra35, 2020-09-26:
### Describe the bugremote
In our logs we found folow

```
2020-09-27 01:58:15.151 T:3023    FATAL <general>: Traceback (most recent call last):
                                                     File "/usr/share/kodi/addons/service.libreelec.settings/oe.py/oe.py", line 828, in load_modules
                                                     File "/usr/share/kodi/addons/service.libreelec.settings/resources/lib/modules/xdbus.py", line 7, in <module>
                                                     File "/usr/lib/python3.8/site-packages/gobject/__init__.py", line 26, in <module>
                                                     File "/usr/lib/python3.8/site-packages/glib/__init__.py", line 22, in <module>
                                                   SystemError: initialization of _glib raised unreported exception
```
And not any upnp features, not working after that(not upnp discovery, not search)
Seems this is know issue https://bugs.busybox.net/show_bug.cgi?id=12286, appear after for python > 3.7

heitbaum, 2020-10-31:
Still erroring on latest master - devel-20201030100043-7f3b20d (Generic.x86_64)

2020-10-31 15:56:57.556 T:1570    FATAL <general>: ## LibreELEC Addon ## oe::MAIN(loadingModules)(strModule) ## ERROR: (SystemError('initialization of _glib raised unreported exception'))
2020-10-31 15:56:57.556 T:1570    FATAL <general>: Traceback (most recent call last):
                                                     File "/usr/share/kodi/addons/service.libreelec.settings/oe.py/oe.py", line 828, in load_modules
                                                     File "/usr/share/kodi/addons/service.libreelec.settings/resources/lib/modules/xdbus.py", line 7, in <module>
                                                     File "/usr/lib/python3.8/site-packages/gobject/__init__.py", line 26, in <module>
                                                     File "/usr/lib/python3.8/site-packages/glib/__init__.py", line 22, in <module>
                                                   SystemError: initialization of _glib raised unreported exception

jernejsk, 2020-10-31:
We know, no need to bump this, it won't be solved any faster.

CvH, 2020-11-21:
needs settings addon updates https://github.com/LibreELEC/LibreELEC.tv/issues/4669

closed 2020-11-21 by CvH

heitbaum, 2020-11-26:
Cross reference to https://github.com/LibreELEC/service.libreelec.settings/pull/188

This particular patch fixes th trace back - but will be bundled as a larger PR for the full fix. https://github.com/LibreELEC/service.libreelec.settings/commit/7e5e83b7b3ac1904fee8481460e467ff258e780a