LibreELEC/LibreELEC.tv #4599: [BUG] build of kernel aborts because of old patchset

damnms, 2020-10-13:
### Describe the bug

i tried to build an image for my odroid n2 with PROJECT=Amlogic DEVICE=AMLG12 ARCH=arm make image

### To Reproduce
Steps to reproduce the behavior:
1. check out latest sources
2. PROJECT=Amlogic DEVICE=AMLG12 ARCH=arm make image

### Informations
 - LE Version: master
 - Hardware Platform: Odroid N2

### Log file
patching file arch/arm64/kernel/cpuinfo.c
Hunk #1 succeeded at 148 (offset 1 line).
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0004-HACK-arm64-dts-meson-gx-add-ATF-BL32-reserved-memory.patch
patching file arch/arm64/boot/dts/amlogic/meson-gx.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0005-HACK-media-cec-silence-CEC-timeout-message.patch
can't find file to patch at input line 25
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From d7ba0960e3bc51f8368ef9d3b98ce33be91e3570 Mon Sep 17 00:00:00 2001
|From: Christian Hewitt <christianshewitt@gmail.com>
|Date: Tue, 7 Jan 2020 07:12:47 +0000
|Subject: [PATCH 005/156] HACK: media: cec: silence CEC timeout message
|
|If testing with an AVR that does not pass-through CEC state the system
|log fills with timeout messages. Silence this to stop the log rotation
|and ensure other issues are visible.
|
|[   42.718009] cec-meson_ao_cec: message ff 84 50 00 01 timed out
|[   45.021994] cec-meson_ao_cec: message ff 87 00 15 82 timed out
|[   47.325965] cec-meson_ao_cec: message 10 timed out
|[   49.630023] cec-meson_ao_cec: message 10 timed out
|[   51.933960] cec-meson_ao_cec: message 10 timed out
|
|Signed-off-by: Christian Hewitt <christianshewitt@gmail.com>
|---
| drivers/media/cec/cec-adap.c | 6 +++---
| 1 file changed, 3 insertions(+), 3 deletions(-)
|
|diff --git a/drivers/media/cec/cec-adap.c b/drivers/media/cec/cec-adap.c
|index 6a04d19a96b2..fcb29dba8de1 100644
|--- a/drivers/media/cec/cec-adap.c
|+++ b/drivers/media/cec/cec-adap.c
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
*********** FAILED COMMAND ***********
patch -d $(echo "${PKG_BUILD}" | cut -f1 -d\ ) -p1 1>&${VERBOSE_OUT}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/build linux:host has failed!

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLG12.arm-9.80-devel/.threads/logs/44.log

>>> linux:host seq 44 >>>
[044/265] [FAIL] build   linux:host

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLG12.arm-9.80-devel/.threads/logs/44.log

Parallel build failure - see log for details. Time of failure: Tue Oct 13 21:37:50 CEST 2020
make: *** [Makefile:12: image] Fehler 1

### Additional context

chewitt, 2020-10-14:
Use https://github.com/chewitt/LibreELEC.tv/commits/amlogic until I bump Amlogic non-support to Linux 5.9

damnms, 2020-10-14:
thanks, i tried it, unfortunately i had the same behaviour. did i do something wrong?

```
    APPLY PATCH (common - default)      packages/linux/patches/default/linux-058.05-hid_sony-add_autorepeat_for_PS3_remotes.patch
patching file drivers/hid/hid-sony.c
Hunk #1 succeeded at 837 with fuzz 1 (offset -283 lines).
Hunk #2 succeeded at 2807 with fuzz 1 (offset 416 lines).
    APPLY PATCH (common - default)      packages/linux/patches/default/linux-062-imon_pad_ignore_diagonal.patch
patching file drivers/media/rc/imon.c
Hunk #1 succeeded at 1427 (offset 83 lines).
    APPLY PATCH (common - default)      packages/linux/patches/default/linux-999-no-lzma-in-x86-perf-build.patch
patching file tools/perf/Makefile.config
Hunk #1 succeeded at 55 with fuzz 2 (offset 20 lines).
    APPLY PATCH (common - default)      packages/linux/patches/default/linux-999.02-0001-pm-disable-async-suspend-resume-by-default.patch
patching file kernel/power/main.c
Hunk #1 succeeded at 95 (offset 49 lines).
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0001-HACK-set-meson-gx-cma-pool-to-768MB.patch
patching file arch/arm64/boot/dts/amlogic/meson-gx.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0002-HACK-set-meson-g12-cma-pool-to-768MB.patch
patching file arch/arm64/boot/dts/amlogic/meson-g12-common.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0003-HACK-arm64-fix-Kodi-sysinfo-CPU-information.patch
patching file arch/arm64/kernel/cpuinfo.c
Hunk #1 succeeded at 148 (offset 1 line).
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0004-HACK-arm64-dts-meson-gx-add-ATF-BL32-reserved-memory.patch
patching file arch/arm64/boot/dts/amlogic/meson-gx.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/linux/amlogic-0005-HACK-media-cec-silence-CEC-timeout-message.patch
can't find file to patch at input line 25
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From d7ba0960e3bc51f8368ef9d3b98ce33be91e3570 Mon Sep 17 00:00:00 2001
|From: Christian Hewitt <christianshewitt@gmail.com>
|Date: Tue, 7 Jan 2020 07:12:47 +0000
|Subject: [PATCH 005/156] HACK: media: cec: silence CEC timeout message
|
|If testing with an AVR that does not pass-through CEC state the system
|log fills with timeout messages. Silence this to stop the log rotation
|and ensure other issues are visible.
|
|[   42.718009] cec-meson_ao_cec: message ff 84 50 00 01 timed out
|[   45.021994] cec-meson_ao_cec: message ff 87 00 15 82 timed out
|[   47.325965] cec-meson_ao_cec: message 10 timed out
|[   49.630023] cec-meson_ao_cec: message 10 timed out
|[   51.933960] cec-meson_ao_cec: message 10 timed out
|
|Signed-off-by: Christian Hewitt <christianshewitt@gmail.com>
|---
| drivers/media/cec/cec-adap.c | 6 +++---
| 1 file changed, 3 insertions(+), 3 deletions(-)
|
|diff --git a/drivers/media/cec/cec-adap.c b/drivers/media/cec/cec-adap.c
|index 6a04d19a96b2..fcb29dba8de1 100644
|--- a/drivers/media/cec/cec-adap.c
|+++ b/drivers/media/cec/cec-adap.c
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
*********** FAILED COMMAND ***********
patch -d $(echo "${PKG_BUILD}" | cut -f1 -d\ ) -p1 1>&${VERBOSE_OUT}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/build linux:host has failed!

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLG12.arm-9.80-devel/.threads/logs/43.log

>>> linux:host seq 43 >>>
[044/265] [FAIL] build   linux:host

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLG12.arm-9.80-devel/.threads/logs/43.log

Parallel build failure - see log for details. Time of failure: Wed Oct 14 10:00:06 CEST 2020
make: *** [Makefile:12: image] Fehler 1

```

```
oli @ ~/LibreELEC.tv - [master] $ git branch
* master
oli @ ~/LibreELEC.tv - [master] $ git remote show
origin
oli @ ~/LibreELEC.tv - [master] $ git remote show origin 
* Remote-Repository origin
  URL zum Abholen: https://github.com/chewitt/LibreELEC.tv.git
  URL zum Versenden: https://github.com/chewitt/LibreELEC.tv.git
  Hauptbranch: master
  Remote-Branches:
    amlogic                  gefolgt
    amlogic-upstream         gefolgt
    appletv-7.0              gefolgt
    bump-connman             gefolgt
    connman-wireguard-92     gefolgt
    connman-wireguard-master gefolgt
    drop-wg-module           gefolgt
    fix-bcmsta               neu (wird bei nächstem "fetch" in remotes/origin gespeichert)
    hyperion-ng              gefolgt
    hyperion-ng-92           gefolgt
    leiot-8.0                gefolgt
    libreelec-7.0            gefolgt
    libreelec-8.0            gefolgt
    libreelec-8.2            gefolgt
    libreelec-9.2            gefolgt
    master                   gefolgt
    misc-bits                gefolgt
    openvfd                  gefolgt
    realtek                  gefolgt
    samsung                  gefolgt
    wyse-wifi                gefolgt
  Lokaler Branch konfiguriert für 'git pull':
    master führt mit Remote-Branch master zusammen
  Lokale Referenz konfiguriert für 'git push':
    master versendet nach master (lokal nicht aktuell)

```

chewitt, 2020-10-14:
I guess you're not doing something right, because it works for me ™️

damnms, 2020-10-14:
i checked out: git clone https://github.com/chewitt/LibreELEC.tv.git
cd LibreELEC.tv
PROJECT=Amlogic DEVICE=AMLG12 ARCH=arm make image

same error again.
maybe its an environmental problem. i use fedora 33 beta (gcc10 etc. - pretty bleeding edge)

chewitt, 2020-10-14:
I'd guess you are not checking out the amlogic branch in my repo, thus you are building master, not master + changes for 5.9

```
git clone https://github.com/chewitt/LibreELEC.tv.git
cd LibreELEC.tv
git checkout amlogic
PROJECT=Amlogic DEVICE=AMLGX ARCH=arm UBOOT_SYSTEM=odroid-n2 make image
```

^ AMLG12 no longer exists in my branch (should be obvious from the commits) as we are now using mesa/panfrost and can combine everything to a single image for GXBB > GXL > GXM > G12A > G12B > SM1 devices.

damnms, 2020-10-14:
oh dang! you are right! thanks! i was on master all the time

chewitt, 2020-10-14:
That build command will result in 11x different .img files being created, so if you only want N2, append UBOOT_SYSTEM as per the command in my previous post.

damnms, 2020-10-14:
now i get another error :/ sorry for disturbing:

```
mv -f .deps/mips.Tpo .deps/mips.Po
/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-g++ -W -Wall    -Wstack-usage=262144 -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -frandom-seed=dwp -I/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/gold/../zlib -pthread -g -O2   -o dwp dwp.o libgold.a ../libiberty/libiberty.a     -L./../zlib -lz -ldl 
/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-g++ -W -Wall    -Wstack-usage=262144 -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -frandom-seed=ld-new -I/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/gold/../zlib -pthread -g -O2   -o ld-new main.o i386.o x86_64.o sparc.o powerpc.o arm.o arm-reloc-property.o tilegx.o mips.o aarch64.o aarch64-reloc-property.o s390.o libgold.a ../libiberty/libiberty.a     -L./../zlib -lz -ldl 
/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-g++ -W -Wall    -Wstack-usage=262144 -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -frandom-seed=incremental-dump -I/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/gold/../zlib -pthread -g -O2 -pthread  -o incremental-dump incremental-dump.o i386.o x86_64.o sparc.o powerpc.o arm.o arm-reloc-property.o tilegx.o mips.o aarch64.o aarch64-reloc-property.o s390.o libgold.a ../libiberty/libiberty.a    -L./../zlib -lz -ldl 
/usr/bin/ld: /usr/lib64/libssh.so.4: undefined reference to `EVP_KDF_derive@OPENSSL_1_1_1b'
/usr/bin/ld: /usr/lib64/libssh.so.4: undefined reference to `EVP_KDF_ctrl@OPENSSL_1_1_1b'
/usr/bin/ld: /usr/lib64/libssh.so.4: undefined reference to `EVP_KDF_CTX_new_id@OPENSSL_1_1_1b'
/usr/bin/ld: /usr/lib64/libcurl.so.4: undefined reference to `SSLv3_client_method@OPENSSL_1_1_0'
/usr/bin/ld: /usr/lib64/libssh.so.4: undefined reference to `EVP_KDF_CTX_free@OPENSSL_1_1_1b'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:934: readelf] Error 1
make[5]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux/binutils'
make[4]: *** [Makefile:1134: all-recursive] Error 1
make[4]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux/binutils'
make[3]: *** [Makefile:761: all] Error 2
make[3]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux/binutils'
make[2]: *** [Makefile:3633: all-binutils] Error 2
make[2]: *** Waiting for unfinished jobs....
make[5]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux/gold'
make[4]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux/gold'
make[3]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux/gold'
make[2]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux'
make[1]: *** [Makefile:853: all] Error 2
make[1]: Leaving directory '/home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/binutils-2.34/.x86_64-redhat-linux'
FAILURE: scripts/build binutils:host during make_host (package.mk)
*********** FAILED COMMAND ***********
make
**************************************
FAILURE: scripts/build binutils:host has failed!

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/.threads/logs/44.log

>>> binutils:host seq 44 >>>
[045/266] [FAIL] build   binutils:host

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/.threads/logs/44.log

Parallel build failure - see log for details. Time of failure: Wed Oct 14 11:56:17 CEST 2020
make: *** [Makefile:12: image] Fehler 1

```

maybe thats now because of fedora, but for completness i poste it here. when i find a solution i will also poste it here.

chewitt, 2020-10-14:
I've just done a full/clean rebuild with no issues. Images are here https://chewitt.libreelec.tv/testing/9.80/

damnms, 2020-10-24:
i just started a new build, guess the old error has been fixed but now i get another one. any idea? fedora 33 beta

```
checking if /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc supports -fno-rtti -fno-exceptions... no
checking for /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc option to produce PIC... -fPIC -DPIC
checking if /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc PIC flag -fPIC -DPIC works... yes
checking if /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc static flag -static works... yes
checking if /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc supports -c -o file.o... yes
checking if /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc supports -c -o file.o... (cached) yes
checking whether the /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc linker (ld -m elf_x86_64 -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... no
checking whether to build static libraries... yes
checking whether make supports nested variables... (cached) yes
checking for /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/toolchain/bin/host-gcc option to accept ISO C99... none needed
checking whether ln -s works... yes
checking for gawk... (cached) gawk
checking for nroff... /usr/bin/nroff
checking for tgetent in -lcurses... no
checking for tgetent in -lncurses... no
configure: error: libcurses or libncurses are required!
configure: error: /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/build/heimdal-7.7.0/lib/libedit/configure failed for lib/libedit
FAILURE: scripts/build heimdal:host during configure_host (default)
*********** FAILED COMMAND ***********
${PKG_CONFIGURE_SCRIPT} ${HOST_CONFIGURE_OPTS} ${PKG_CONFIGURE_OPTS_HOST}
**************************************
FAILURE: scripts/build heimdal:host has failed!

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/.threads/logs/70.log

>>> heimdal:host seq 70 >>>
[077/265] [FAIL] build   heimdal:host

The following log for this failure is available:
  /home/oli/LibreELEC.tv/build.LibreELEC-AMLGX.arm-9.80-devel/.threads/logs/70.log

Parallel build failure - see log for details. Time of failure: Sun Oct 25 00:19:28 CEST 2020
make: *** [Makefile:12: image] Fehler 1

```

Karl-opec, 2020-10-27:
This is the well known heimdal issue #4457

erbasandrade, 2020-11-17:
> I've just done a full/clean rebuild with no issues. Images are here https://chewitt.libreelec.tv/testing/9.80/

I have a small problem with the initial Kodi settings. SSH and SAMBA.
I try to enable ssh or samba and it is disabled.
LibreELEC-AMLGX.arm-9.80.7-box. 2020-Nov-15
Thank you friend.

heitbaum, 2021-02-04:
@damnms - there is a PR that may resolve your fedora build issue. https://github.com/LibreELEC/LibreELEC.tv/pull/5089 current master should work for your Odroid N2

heitbaum, 2021-02-04:
@erbasandrade - with the current https://chewitt.libreelec.tv/testing/9.80/ are you still facing the same issues? Settings has been updated a couple times since and should be improved. I have tested on an Odroid N2+

chewitt, 2021-03-08:
As the main branch is on 5.11.3 (for Amlogic) I'm going to close this out.

closed 2021-03-08 by chewitt