LibreELEC/LibreELEC.tv #4607: [BUG] Licensing and authors before 2009 (the initial import is a fork)

Skywalker13, 2020-10-15:
Hello guys,

The first import by Stephan Raue (2009) is an altered fork of GeeXboX, see https://github.com/LibreELEC/LibreELEC.tv/blob/de71016fb4c7787cf55e336b000a8b53192e0853/config/issue

You can compare the files, it's very similar between the first import and GeeXboX, other example (mars 2009):

OpenELEC 2009
https://github.com/LibreELEC/LibreELEC.tv/blob/de71016fb4c7787cf55e336b000a8b53192e0853/scripts/install

GeeXboX 2009
https://github.com/OpenBricks/openbricks/blob/0fa96ba955fe365dc3d4fe13dba6463e9a709c00/scripts/install

GeeXboX (OpenBricks) is using GPLv2 license. Why the authors of GeeXboX (like me by the way) are just killed in OpenELEC and (de facto) LibreELEC? That's not very kind.. 😞 

Yes, GeeXboX is dead... it's just for the posterity, and the respect of all the work which were done on GeeXboX by many people

Thanks,
Regards

lrusak, 2020-10-15:
What would you like us to do? Put a blurb in the readme about the history of forks?

Skywalker13, 2020-10-16:
Just respect the GPLv2 license of the original project; nothing less, nothing more. A fork cannot remove the original authors.

The problem is here:
[image]

It's impossible to retrieve the authors in LibreELEC before 2009 because Stephan Raue has removed all copyright entries and violates the GPLv2 license by the way.

[image]
OK for new code, but not for the original code since 2009. You can't take the old copyrights, it's based on scripts / code copyrighted by the GeeXboX's members.

See https://github.com/OpenBricks/openbricks/tree/0fa96ba955fe365dc3d4fe13dba6463e9a709c00 for COPYING, COPYRIGHT and AUTHORS files

Just restore the history please, and note that the git history is lost intentionally by Stephan Raue.

lrusak, 2020-10-16:
We are all for respecting copyright and will do what we can (within reason of course).

We can only do so much with what we have inherited and the project has diverged significantly it's almost impossible to see any resemblance. None of us were even around when the OpenELEC fork was created.

So short of a blurb in the readme about the history of forks and copyright information I'm not sure what else we can do to remedy this situation?

Please be clear about what you think we should do.

Skywalker13, 2020-10-16:
> So short of a blurb in the readme about the history of forks and copyright information I'm not sure what else we can do to remedy this situation?

Looks good to me

CC @gxben @davide125 @guillaumelecerf

guillaumelecerf, 2020-10-16:
LGTM too, please at least credit us to respect our work.

davide125, 2020-10-16:
+1, I would love to see our work in GeeXboX credited properly here. Thanks @Skywalker13 for raising this!

stefansaraev, 2020-10-18:
lol, really ?

btw, at the time I joined OpenELEC, there was almost nothing left from the original geexbox code. so lets be honest here... with all due respect to your work on GeeXboX, it is dead, 

LibreELEC is set of build scripts, recipes and so on, that have absolutely nothing to do with the gheexbox's codebase or past work. 

my 2 cents.

Skywalker13, 2020-10-18:
Just read the history of LibreELEC projet : https://github.com/LibreELEC/LibreELEC.tv/commit/082b1d4dd03d2fccfea7cc86182117597549efb1

It's easy to base a work on an existing work... it's much harder to build everything from the beginning

stefansaraev, 2020-10-18:
I know very well what OpenELEC was based on, thank you, and as said - at the time I joined, there was almost nothing left from the original "initial import" codebase. we basicaly built verything from scratch, so please dont tell me how hard it is..

Coming here (at a fork of a fork...) 10+ years later and asking for credits makes no sense. The world does not revolve around us, those who do not do code contributions anymore to the projects we love. time to move on.

MilhouseVH, 2020-10-18:
The original OpenELEC developer is @sraue who claimed copyright on much (if not, all) of the early OpenELEC code that appears to have included original GeeXboX code prior to it being significantly reworked/replaced by @sraue and others (including @stefansaraev) in the following years. It is this code that LibreELEC inherited following a fork in March 2016.

As such, I'm pretty sure the GeeXboX beef regarding any removal of copyright should be with @sraue and nobody else, and certainly **not** those that came afterwards to assist @sraue on OpenELEC in blissful ignorance of any prior GeeXboX involvement, considering most of the GeeXboX references had already been removed by that point.

I'm not saying the GeeXboX complaint is not justified, just that it's way off target, especially considering the elapsed time and that LibreELEC is a fork of OpenELEC (of which there are now several) from a time when virtually all GeeXboX code had already been removed/replaced in OpenELEC several years previously.

I'm guessing LibreELEC has been contacted because it's still active, and because you know the original OpenELEC project is dead and that the original developer who has caused this issue can no longer be contacted? You can still open issues at [OpenELEC](https://github.com/OpenELEC/OpenELEC.tv/issues) even if it's likely to be ignored - you would at least be staking your claim at the source, and leaving it there for posterity! :smile:

Requesting that individual GeeXboX developers now be name checked in LibreELEC doesn't really seem reasonable, as there's always going to be someone else that wants adding to the list in another few years, with nobody to verify their involvement.

In the 2009 GeeXboX [COPYRIGHT](https://github.com/GeeXboX/geexbox/blob/0fa96ba955fe365dc3d4fe13dba6463e9a709c00/COPYRIGHT) (which is relevant for the period under discussion) it mentions the following:
```
GeeXboX as a whole is copyrighted by the GeeXboX team, 2002-2009.`
```
so on that basis the best that Team LibreELEC could/should do now to address the past sins of an ex-developer of an ex-upstream project is to acknowledge (most likely in the README.md) that "In 2009 the [GeeXboX project](https://github.com/GeeXboX/geexbox) served as the inspiration for the original OpenELEC build system which itself evolved over many years into the current LibreELEC build system".

And then hopefully everyone can move on.

chewitt, 2020-10-19:
I've submitted #4611 to address this. The current project team (represented on this thread by myself and @lrusak) has no issue making a simple credit, and we are always open to neighbourly collaboration on matters of mutual interest, e.g. Kodi.

guillaumelecerf, 2020-10-19:
LGTM, thanks 👍

Skywalker13, 2020-10-19:
> I'm guessing LibreELEC has been contacted because it's still active, and because you know the original OpenELEC project is dead and that the original developer who has caused this issue can no longer be contacted?

I don't know if the original OpenELEC developer can be contacted or not (IIRC he was contacted ~10 years ago by @gxben and it was difficult to talk with him (@sraue)). It's true that I've open this "issue" because LibreELEC is still active.

And why now? Because after long years, I'm working again on cross-toolchains and embedded systems..

chewitt, 2020-10-19:
> I don't know if the original OpenELEC developer can be contacted or not (IIRC he was contacted ~10 years ago by @gxben and it was difficult to talk with him (@sraue)). It's true that I've open this "issue" because LibreELEC is still active.

Stephan's contact details are known but we're not in his friends list and he doesn't acknowledge messages when we ping him to point out the OE forum/website needs a reboot etc. - so I wouldn't get hopes up.

> And why now? Because after long years, I'm working again on cross-toolchains and embedded systems..

LE's build-system will look familiar but has the advantage of parallel build which can shave hours off compile times. Forks and PRs that find their way back upstream are welcome :)

PR #4611 (2020-10-20): README: revise general text and credit OpenBricks/GeeXboX build system in copyright
This PR does some overdue cleanup to the README file: we forked 4.5 years ago, and we swapped from forum tickets to GH issues recently. It also rewords the copyright section for brevity and credits OpenBricks/GeekBox as the original source and inspiration for the build system we later inherited, to resolve #4607.
closed 2020-10-20 by CvH