LibreELEC/LibreELEC.tv #4610: [BUG] only use HTTP2 with SSL

KiPSOFT, 2020-10-18:
### Describe the bug
https://github.com/xbmc/xbmc/issues/18572

Until a few days ago cloudflare stopped allowing non-SSL requests for http2 requests. For this reason, problems arise for providers that do not have SSL. A backport for kodi 19 has been released. Can we do this for librelec 9.x?

Kodi pull reqeust: https://github.com/xbmc/xbmc/pull/18577

### To Reproduce
Steps to reproduce the behavior:
You have to open a non-SSL stream using cloudflare.

### Informations
 - LE Version: 9.2
 - Hardware Platform: RPi1

### Log file

### Additional context
Kodi pull reqeust: https://github.com/xbmc/xbmc/pull/18577

mebkopcznski, 2020-10-18:
What the odds of having a new official version with the solution ? It seems to be affecting not only IPTV but also others add-ons like Gismeteo, TMDB, etc.

CvH, 2020-10-18:
Depends how much is actually effected.
Currently the fix is 2 days old so doing a "stable" release today is super risky because we have no idea if it breaks something. 
If needed we do a release, currently the situation is still a bit unclear.

mebkopcznski, 2020-10-18:
Got it. Thanks!!!

Just puting here a temporary workaround that may works fore some providers...

have found a temporary workaround but it will depends on how your provider accept connection using "httpS".
Since the problem is with http2 protocol the URLs should work with https connection.

1 - Download the m3u file using your regular brownser.
2 - Open the file on your notepad (it is a simple text file).
2 - Replace all URLs from "http://" to "https://"
3 - Replace (if there is the port number on the URL) from ".com:80" to ".com:443"
4 - Save the file in a place where you can access from Kodi (can be on Kodi itself or in your network).
5 - On Kodi Simple IPTV Configuration change the m3u URL from "Internet" to "Local Connection"
6 - Select the file that you have saved.
7 -Restart

If your provider update the m3u file you will have to do all over again.

KiPSOFT, 2020-10-18:
> Depends how much is actually effected.
> Currently the fix is 2 days old so doing a "stable" release today is super risky because we have no idea if it breaks something. 
> If needed we do a release, currently the situation is still a bit unclear.

At least the 10.x version can be tested on the alpha channel.  Without waiting for it to be stable.

shahinco, 2020-10-18:
Hi @mebkopcznski 
Step 3 doesn't work if your or some IPTV Provender doesn't supports HTTPS Ports ( 443 ) 
I think best way to wait fix this bug from core.. 

Someone knows when will be release new Alpha or Stable version ? 
Thanks. 

Shahin

mebkopcznski, 2020-10-18:
Yes I know.
As I said it depends on how your provider accepts https connection.

shahinco, 2020-10-18:
Someone said edit and add in `advancedsettings.xml` below setting going to fix this issue. But on LibreELEC
`-rw-r--r--    1 root     root           369 Jul 29 21:57 advancedsettings.xml` permissions setup only for Read-only file. 

`<advancedsettings>
  <network>
    <disablehttp2>false</disablehttp2>
  </network>
</advancedsettings>`

Someone tried this fix ?

queeup, 2020-10-18:
This workaround is for Kodi 19. LibreElec is kodi 18.8 atm. That advanced settings option not exist on LibreElec version.

mebkopcznski, 2020-10-18:
It works only with v19.
Libreelec uses v.18

mebkopcznski, 2020-10-19:
Hi Guys...
I have found this in a Cloudflare forum.
To people that have a different port than 80. Could give a try in other ports to understand if they are open or not to connection.

To avoid to have to replace de several times you can give it a try on your browser.

[quote]
Simple , 
Change streaming port for one of cloudflares proxy ports like: 

HTTP ports supported by Cloudflare:

80
8080
8880
2052
2082
2086
2095
HTTPS ports supported by Cloudflare:

443
2053
2083
2087
2096
8443
[\quote]

chewitt, 2020-10-19:
Fixes have already been posted by Kodi and there will be an 18.9 release to resolve this; and once that ships we will package it into an LE 9.2 bump. Until that happens there's not much more to say .. we're aware, it will be resolved.

shahinco, 2020-10-19:
Hi @mebkopcznski 
Yes this is nice post.. 

I have checked all random HTTPS ports ( 443, 4443, 8443, 8181 etc) but i think my IPTV Provider doesn't use different ports they had setup only for HTTP 8080. 
I have asked to my Provider still waiting official answer for HTTPS Port and which one they used.. 

Cheers 

Shahin

shahinco, 2020-10-19:
@chewitt 
Thanks for fast feedback. Do you have any ETA ? for that.. 

Regards 

Shahin

chewitt, 2020-10-19:
No ETA, but sooner than later.

jncastro, 2020-10-20:
> Someone said edit and add in `advancedsettings.xml` below setting going to fix this issue. But on LibreELEC
> `-rw-r--r-- 1 root root 369 Jul 29 21:57 advancedsettings.xml` permissions setup only for [...]

Does not work on Libreelec 9.0.4 Kodi 18.2 (<advancedsettings> <network> <disablehttp2>false</disablehttp2> </network> </advancedsettings>` or true)

solution in kodi 18.2 e 18.8 oficial ??

shahinco, 2020-10-20:
Hi @jncastro 
I learned from @queeup  this solution only works for [Kodi-19.xx](https://github.com/LibreELEC/LibreELEC.tv/issues/4610#issuecomment-711431533) version.

jncastro, 2020-10-20:
> Hi @jncastro
> I learned from @queeup this solution only works for [Kodi-19.xx](https://github.com/LibreELEC/LibreELEC.tv/issues/4610#issuecomment-711431533) version.

What is the solution for kodi 18.xx version ?

thestupidbloke, 2020-10-24:
Hi All, this is fixed for me now without any update to libreelec or my provider... I guess cloudflare made a change at their end. Anyone else experiencing the same?

KiPSOFT, 2020-10-24:
Kodi v18.9 has released. When release new libreelec?

lrusak, 2020-10-24:
Please use the forum for support.

closed 2020-10-24 by lrusak

mebkopcznski, 2020-10-24:
I have search it but didn’t find anything...
There is any update ?

KiPSOFT, 2020-10-24:
Where is info for this bug in forum ? @lrusak

lrusak, 2020-10-24:
> Where is info for this bug in forum ? @lrusak 

Make a new thread if you can't find one.