LibreELEC/LibreELEC.tv #4620: [BUG] Inputstream.* not compatible with latest Kodi Matrix

heitbaum, 2020-10-24:
### Describe the bug

Inputstream.Adaptive not compatible with latest Kodi Matrix 
Inputstream.ffmpegdirext not compatible with latest Kodi Matrix 
Inputstream.rtmp not compatible with latest Kodi Matrix 

### To Reproduce
Steps to reproduce the behavior:
1. Update to the latest test image

### Informations
 - LE Version: LibreELEC (community): nightly-20201024-636e32e
 - Hardware Platform: Generic.x86_64

### Solution
Include the following in libreElec

https://github.com/peak3d/inputstream.adaptive/issues/537
https://github.com/xbmc/xbmc/pull/18352

https://github.com/xbmc/inputstream.rtmp/pull/69
https://github.com/xbmc/inputstream.ffmpegdirect/pull/64
https://github.com/peak3d/inputstream.adaptive/pull/507

### Log file

LibreELEC:~/.kodi/addons # cat inputstream.*/addon.xml | pastebinit 
http://ix.io/2BON
LibreELEC:~/.kodi/addons # ls -la inputstream.*/addon.xml
-rw-r--r--    1 root     root          3699 Sep 12 14:59 inputstream.adaptive/addon.xml
-rw-r--r--    1 root     root          8425 Sep 23 22:34 inputstream.ffmpegdirect/addon.xml
-rw-r--r--    1 root     root          1560 Oct 18 18:00 inputstream.rtmp/addon.xml

### Additional context

https://forum.kodi.tv/showthread.php?tid=343069&pid=2984095#pid2984095

HiassofT, 2020-10-24:
The regular API breakages in Kodi are quite unfortunate...

Please use the nightly addon builds from https://test.libreelec.tv/addons/ until the addons in the LE repo have been updated.

inputstream.adaptive 2.6.0.2 from there has been reported to work fine - see https://forum.libreelec.tv/thread/22846-inputstream-adaptive-in-nightly-generic-x86-64/

heitbaum, 2020-10-24:
@HiassofT I can confirm that all there modules a have loaded with nightly-20201024-636e32e (Generic.x86_64) without error

inputstream.adaptive-2.6.0.2.zip
inputstream.ffmpegdirect-1.16.0.1.zip
inputstream.rtmp-3.2.0.1.zip 

==

2020-10-25 00:04:52.359 T:37209    INFO <general>: CAddonMgr::FindAddon: inputstream.adaptive v2.6.0.2 installed
2020-10-25 00:05:10.461 T:37210    INFO <general>: CAddonMgr::FindAddon: inputstream.ffmpegdirect v1.16.0.1 installed
2020-10-25 00:05:10.547 T:37238    INFO <general>: CPythonInvoker(18, /storage/.kodi/addons/inputstream.ffmpegdirect/resources/lib/runner.py): script successfully run
2020-10-25 00:05:27.264 T:37209    INFO <general>: CAddonMgr::FindAddon: inputstream.rtmp v3.2.0.1 installed

lrusak, 2020-10-24:
Please use the forums for support.

closed 2020-10-24 by lrusak

CvH, 2020-10-24:
repository is now updated with the latest addon versions