LibreELEC/LibreELEC.tv #4630: [BUG] CEC not working on raspberry pi 3

flubshi, 2020-10-27:
### Describe the bug
Similar [to this report](https://forum.libreelec.tv/thread/22811-pulse-cec-not-working-on-latest-rpi2-nightly-build/), CEC stopped working on RPi3 on latest master for me as well.
CEC (control via TV remote) worked before. Unfortunately, [test.libreelec.tv](http://test.libreelec.tv/) provides only a few images from past. Thus, I cannot reproduce when it stopped working.

### To Reproduce
Steps to reproduce the behavior:
1. Install latest nightly image on RPi3
2. Connect to a CEC supported TV
3. Enable CEC adapter in Kodi settings
4. Try to use TV remote to control Kodi

### Informations
 - LE Version: LibreELEC-RPi2.arm-9.80-nightly-20201027-e5d254c.tar
 - Hardware Platform: RPi3

### Log file
See: https://pastebin.com/jdrCN1eY

### Additional context

HiassofT, 2020-10-27:
This is a known issue in the RPi linux kernel.

We are currently waiting for the proposed fix https://github.com/raspberrypi/linux/pull/3917 to be merged to the RPi kernel, when that happens we'll include it in LibreELEC.

jairoxyz, 2020-11-29:
@flubshi  I build from latest sources including https://github.com/raspberrypi/linux/pull/3917 and CEC works fine. I can share the tar if you don't want to wait for that PR to be merged.

flubshi, 2020-11-29:
Thanks @jairoxyz for testing and the offer providing the build. Unfortunately, I have currently no time to test. But due to your encourage it is now merged upstream 👍

HiassofT, 2020-11-29:
PR with 5.9 kernel update including CEC fix is here: #4700

closed 2020-11-29 by CvH