LibreELEC/LibreELEC.tv #4641: [BUG] Auto updater doesn't work

MartinX3, 2020-11-04:
### Describe the bug
Since a long time I am needed to manual trigger the updater by choosing the newer version by hand in the menu.
It did work in the past a long time ago and I thought it will be patched by a newer update.
That's the reason why I report it now.

### To Reproduce
Waiting for the updater mess

### Informations
 - LE Version: 9.2.4
 - Hardware Platform: RPi3

### Log file
http://ix.io/2CYm

### Additional context
I wasn't able to execute the log function in the settings menu.
It stay on "Pasting..." forever.

But logging in via ssh was successful.
```
LibreElec:~ # pastekodi
error=1 error_msg="Command not registered"
Use 'vcgencmd commands' to get a list of commands
http://ix.io/2CYm
```

chewitt, 2020-11-04:
The auto-update function requires one of the project staff to manually update release information server-side. Only after this step are clients told that a new release is available and the download/update behaviour is triggered. As our current release manager consistently forgets about this step (all the focus is on building/uploading the files) the auto-update function works fine but isn't be triggered. This is a long-standing human/process issue not an actual bug.

NB: Current behaviour will likely continue until we rewrite the settings add-on (which needs to be done for Python3) to move the auto-update logic client-side and eliminate humans from the process.

ping @CvH

MartinX3, 2020-11-04:
Thank you very much for your response and I wish you all the best eliminating all humans. <3

mglae, 2020-11-05:
Updater is already fixed in #4547.

CvH, 2020-11-09:
The updater was borked and didn't worked properly. After fixing we forgot to bump the package at LE 9.2 branch so the fix was never merged into any release.

svenfoo, 2020-11-24:
Keeping the update information separate from he availability of the release on github actually seems to be a good idea. It allows you to make a release and then wait a few days before providing it as an automatic update to all users. Only some interested and technically versatile users will manually update during the first days, which gives you a chance to catch bugs before they hit the broad mass of devices out there. With a little logic on the update server you could also implement a phased roll-out of automatic updates. Just something you might want to consider ...

chewitt, 2020-11-24:
LE has no plans to release files on GitHub as we already use a global list of mirrors and it would technically be an abuse of GitHub TOS, and we already do the two stage release process where files are posted and then later (when someone remembers) the auto-update config is changed. This needs to change at some point to move the logic wholly into the settings add-on so the dependency on forgetful staff and need for an update server to exist (beyond stats collection, it's second purpose) are removed.

MartinX3, 2020-11-24:
Looks like it was https://github.com/LibreELEC/LibreELEC.tv/pull/4633 ?

Hopefully we get a 9.2.7 release with this update. :)

chewitt, 2021-03-08:
I'm closing this ticket out as the issue was resolved in LE 10.0 codebase and there will be no LE 9.2.7 release.

closed 2021-03-08 by chewitt