LibreELEC/LibreELEC.tv #4644: [BUG] Squeezelite package in multimedia-tools outdated

Nelisknip, 2020-11-05:
I'm running Libreelec on an Rpi 4 with Hifiberry DAC to great satisfaction, exept for the Squeezelite for Kodi plugin not entirely functioning as it should.

I'm experiencing playback issues when including the Squeezelite player from this Rpi in the synchronisation with three other Pis in the network that run updated Picoreplayer software with latest Squeezelite. Using the three PcP players synced work beautifully, but including the Kodi one is where things go wrong.

I have done some prior investigation and am quite sure updating Squeezelite to latest version (**or at least above 1.9.3**) will solve my issue. If anyone could perhaps update the multimedia-addons package in Libreelec, that would be greatly appreciated!

Multiroom setup (all directly connected by ethernet over the same switch):

Rpi4 PcP with LMS and Hifiberry Amp2 - Squeezelite version v1.9.6-1206-pCP

Rpi4 PcP with hifiberry DAC+ pro v1.9.6-1206-pCP

Rpi Zero WH PcP with Hifiberry DAC+ zero and LAN dongle v1.9.6-1206-pCP

Rpi4 PcP Libreelec with Hifibery DAC+ pro - Squeezelite v1.9.0-1113 -> This one gives issues

Many thanks in advance,

Nelisknip

CvH, 2020-11-05:
tx for hint, indeed the package is rather out of date
I have a look this weekend.

Nelisknip, 2020-11-05:
That would be awesome! Thanks for your effort in advance.

Nelisknip, 2020-11-15:
Hi,

Any news on this?

Gr, Nelis

CvH, 2020-11-17:
https://github.com/LibreELEC/LibreELEC.tv/pull/4663

Nelisknip, 2020-11-18:
Thank you very much for this!

Is there any way I can update this package in Kodi manually now? I suppose otherwise it will be available later in a next release? Apologies for the noob-questions...

CvH, 2020-11-18:
[RPi4-virtual.multimedia-tools-9.2.0.110.zip](https://github.com/LibreELEC/LibreELEC.tv/files/5560707/RPi4-virtual.multimedia-tools-9.2.0.110.zip)
I compiled it for LE9.2 RPi4 so it should work, pls report back if it fixes it.

Nelisknip, 2020-11-19:
Hi! That's awesome. 

I uninstalled the outdated package, and installed your zip without problems. The squeezelite version shows the current 1.9.8 in the LMS server now. 

I updated all other squeezeboxes in the network to latest (same) version too and at first test it all seems to work stable now, no playback issues anymore. It seems finally my multiroom audio project is complete! 

I bought you a beer by donating to the project.

Cheers, Niels.

joswart, 2021-01-02:
Is there a way to get this for Intel hardware as well?

CvH, 2021-01-02:
hmm for some reason it was not build for generic, I have a look and report back after I fixed it

CvH, 2021-01-06:
build the addons two times to realize that I forgot to pr the changes to LE9.2 ... doh

heitbaum, 2021-04-14:
Closing issue. Updated now. Please let us know if there is still an open issue.

commit 085c0649 (2020-11-17): Merge pull request #4663 from CvH/10.0/squeeze

update multimedia-tools fixes #4644
closed 2021-04-14 by heitbaum