LibreELEC/LibreELEC.tv #4646: [BUG]

bestouff, 2020-11-07:
### Describe the bug
After watching a movie, the fonts are lost. Or something like that. Everything that is written disappears, only the pictures, video and logos are remaining. UI texts and subtitles are lost.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Films
2. Play a movie until the end
3. Play another movie: texts in UI and subtitles have disappeared, only a reboot will fix that

### Informations
 - LE Version: 9.2.4
 - Hardware Platform: Zotac Zbox (generic x86 PC) without hard drive, boots from PXE

CvH, 2020-11-07:
Did you checked that you have the correct resolution set upped at kodi (1920x1080) ?
I had similar issues because the intel driver (at that kernel version) refuses to detect my TV correctly, resulting in exactly that issue.

chewitt, 2021-03-08:
I'm closing this ticket as there will be no more LE 9.2 releases. Please retest with a current LE10 nightly and either reopen the ticket or raise a new one. Thanks.

closed 2021-03-08 by chewitt