LibreELEC/LibreELEC.tv #4669: [HELP Wanted] LE Settings add-on port dbus to Python3

CvH, 2020-11-18:
Currently the LE Settings add-on is broken for configuring WLAN and BT. That part was not yet ported to py3.
The python dbus bindings we are currently using are EOL and we want to switch to dbussy (already at LE master https://github.com/LibreELEC/LibreELEC.tv/pull/4568).

@edit4ever tried to port the settings add-on to dbussy but eventually run out of free time

Currently **there is nobody** that has time and skills at LE who could fix that. Due the forthcoming release of Kodi 19.0 we are really looking for someone who want to help to make it work again.

If additional information are needed, just ask :)

heitbaum, 2020-11-19:
Hi @CvH / @edit4ever - 

In looking into this, I assume we are looking at updating 
https://github.com/LibreELEC/service.libreelec.settings/ from dbus to dbussy 

Specifically
- oe.py
- bluetooth.py
- connman.py
- xdbus.py

I have had a look around at the existing @edit4ever forks - just wondering how far we got?

chewitt, 2020-11-19:
I think we got to the research part and then stalled due to lack of time, and also due to it being uncharted territory.

heitbaum, 2020-11-19:
I’ll have a bit further of a look at the code on the weekend - see what is required.

johnpaul7, 2020-11-21:
I'm running a Pine64+ and with the latest nightlies it crashes on every boot (#4563). Just to get it working, I had been looking into updating the package version for pygobject and including one of its required dependencies, gobject-introspection, but now that I've found this issue I can see that the best way forward is to remove the LE settings dependency on gobject entirely. Let me know if there's anything I can do to help with this issue.

heitbaum, 2020-11-26:
Refer to https://github.com/LibreELEC/service.libreelec.settings/pull/188 for the latest updates on dbus.

gd71, 2020-12-11:
#4733 looks to fix wlan configuration and connection on my RPI4 with no magic and smooth processing.
BT still failing (but looks expected for now).
Warm big thanks guys, I can now switch on LE10 dev for my daily usage :)

chewitt, 2021-03-08:
Huge thanks to @thoradia for doing the initial lift reworking the settings add-on codebase around dbussy and moving this long-overdue task forwards. Further kudos to everyone else who tested and contributed changes. There are still some minor improvements to do, but it's now in a useable, releasable (and merged) state, so it's time to close out this ticket :)

closed 2021-03-08 by chewitt