LibreELEC/LibreELEC.tv #4672: memory cache using Advancedsettings.xml does not work anymore when upgrading to v9.2.6

gnu644, 2020-11-19:
### Describe the bug
Memory cache based on advanced settings configuration does obviously not work any more when upgrading from LE 9.2.5 to LE 9.2.6. => REGRESSION 

### To Reproduce
Steps to reproduce the behavior:
1. Switch to HD TV channel 
2. Video stops repeatedly after a few seconds for reloading buffer

### Informations
 - LE Version: 9.2.6
 - Hardware Platform: RPi3b

### advancedsettings.xml residing in /storage/.kodi/userdata
<advancedsettings version="1.0">
<cache>
<memorysize>104857600</memorysize>
</cache>
</advancedsettings>

### Additional context
Memory cache without HD channel interruptions worked perfectly with same configuration and HW when using LE v9.2.5.
Regression occurred just after upgrade to v9.2.6 without any configuration changes,

Please provide fix or workaround!

sopparus, 2020-12-10:
Please provide log

gnu644, 2020-12-14:
During testing for producing the log, the issue could be clarfied: When having upgraded my installation, I had by incident isochronously installed a new LED lamp that uses a switching power supply. Due to using Powerlan for my LIBREElec installation, I swirched off the LED lamp and surprisingly evrything works fine again. Sorry for this false positive, but I was very surprised that a small switching power supply for producing the right LED DC voltage could have so dramatic impact on the powerlan stability and throughput. Thanks for your support anyway.

heitbaum, 2021-02-04:
Closing resolved ticket.

closed 2021-02-04 by heitbaum