LibreELEC/LibreELEC.tv #4677: [BUG] Dependency failed for Broadcom sdio firmware update for BCM43430A1

dtmaster79, 2020-11-20:
I tried the last nightly builds from November 13th. I have an RPi2. None of the builds work. When starting, this message always comes and the start process is canceled.

Dependency failed for Broadcom sdio firmware update for BCM43430A1

HiassofT, 2020-11-20:
The dependency failed message may be a bit annoying, but it's harmless.

The latest nightly https://test.libreelec.tv/LibreELEC-RPi2.arm-9.80-nightly-20201120-5d0dc8a.img.gz is booting fine here on RPi2. Please note that upgrading from LE 9.2 isn't supported yet so do a fresh install with that image.

If the problem persists remove "quiet" from cmdline.txt and add "ssh" instead. Then you should be able to log in and grab logs (run "pastekodi").

dtmaster79, 2020-11-20:
I didn't have a 9.2 on the RPi2. I had the last Milehouse build from July on the RPi2. But he doesn't do any more.

Can the databases etc. be backed up before a fresh installation? I don't feel like setting up everything from scratch.

HiassofT, 2020-11-20:
Manually copying databases should work fine.

I strongly recommend though using a separate SD card for RPi2 testing ATM as there are still several major issues (eg CEC not working on RPi2/3, no seeking when using hardware decoding) - quite a lot of things have changed since the last milhouse builds

chewitt, 2021-03-08:
Closing as this is not really a bug report (and dated).

closed 2021-03-08 by chewitt