LibreELEC/LibreELEC.tv #4681: [BUG]

cmoulliard, 2020-11-22:
### Describe the bug

The application `LibreELEC USB-SD Creator` cannot be installed on macos. The following error is reported

```
PasteBoard: Error creating pasteboard: com.apple.pasteboard.clipboard [-4960]
PasteBoard: Error creating pasteboard: com.apple.pasteboard.find [-4960]
2020-11-22 18:28:14.103 LibreELEC USB-SD Creator[85682:2536826] The application with bundle ID com.apple.ScriptEditor.id.LibreELEC is running setugid(), which is not allowed. Exiting.
```

### Additional info
- MacOs version: 10.15.7

broglea, 2021-03-11:
This also happens on macOS Big Sur 11.2.1

heitbaum, 2021-04-17:
Please review forum post (WIP) https://forum.libreelec.tv/thread/21333-usb-sd-creator-error-in-macos-catalina/?postID=154244

heitbaum, 2021-04-23:
Also https://forum.libreelec.tv/thread/22207-libreelec-usb-sd-creator-macos-will-not-run-on-macos-catalina/?postID=154625&highlight=Catalina#post154625

heitbaum, 2024-03-11:
Recent updates to this have been done. Please reopen if still an issue. https://github.com/LibreELEC/usb-sd-creator

closed 2024-03-11 by heitbaum