LibreELEC/LibreELEC.tv #4684: [BUG] OVA virtual image installer can't find disk

CvH, 2020-11-22:
Starting the OVA image and running the installer at VMWare can't find any disk.

[image]

with pr https://github.com/LibreELEC/LibreELEC.tv/pull/4615 from @mglae 
http://ix.io/2F7Z

LE Master
http://ix.io/2F81

CvH, 2020-11-22:
you need to type run to make it work instead of installer

mglae, 2020-11-22:
The vmware disk is seen if configured as "SATA".

Default HD mode is supported by adding `CONFIG_FUSION_SPI=y` to the kernel.

More worse: I did not get any sound until adding `CONFIG_SND_ENS1371=y`.

Time for a convenience PR...

CvH, 2020-11-23:
[image]

looks like vmware detect it as IDE

mglae, 2020-11-23:
When creating a virtual HD you can choose the type:

[image]

You seem to have "IDE" configured. Is this default on Windows?

"SATA" is supported (kernel driver ata_piix). The CONFIG_FUSION_SPI above is for the recommended "SCSI".

Can you test your configuration (I'm interested in checking audio too)? Boot installer, then Alt-F3 for debug console and:
```
mount -o rw,remount /flash
lspci -v >/flash/lspci.log
poweroff
```

CvH, 2020-11-23:
https://pastebin.com/fKJCenMs
btw I double-click at the ova then it get opened, there is no possibility to change anything
here is he ova template https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/Generic/config/ovf.template

mglae, 2020-11-23:
OK, got it. You are using the generated OVA file. I should not answer late at night ;-)

I can confirm the included simulated HW is supported and you have to use "run" because of already using the HD.

I did use vmplayer for some PXE boot tests and run into the issues with the default simulated HW.

Let us keep this issue closed.

mglae, 2023-03-24:
You have to use the `run` feature because there is only one disk defined in the OVA.

If motivated you can try to define a second empty disk but then the boot order may be an issue.

mglae, 2023-03-30:
- Use "Edit virtual machine settings" and add a second IDE HD. 
- Boot into installer and perform the installation.
- Press F2 at reboot to open BIOS.
- Change boot order to boot from second HD.

heitbaum, 2026-02-13:
Closing as stale - Please reopen if not.

closed 2026-02-13 by heitbaum