LibreELEC/LibreELEC.tv #4719: [BUG] ytop fails to build when cross compiling

heitbaum, 2020-12-05:
### Describe the bug

On compiling ytop on master the cross compiler is not picking up the glibc.

I’m pretty sure the issue is that the build host **Ubuntu 18.04** has not got libc 2.32 - though the cross compiler target should not need this.

``` 
BUILD      ytop (target)
    TOOLCHAIN      manual
   Compiling libc v0.2.69
...
error: failed to run custom build command for `libc v0.2.69`

Caused by:
  process didn't exit successfully: `/home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build` (exit code: 1)
--- stderr
/home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build)
```

### To Reproduce
Steps to reproduce the behavior:
Execute the build for ytop
PROJECT=Generic ARCH=x86_64 scripts/build ytop

I have tried a few things to understand the **cargo** cross compile but to no avail. 

E.g.
```
PKG_BUILD=/home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build
. ./rust-1.45.0/cargo/env
cd ytop*
cargo build     --release     --locked     --all-features     --target=x86_64-unknown-linux-gnu
cargo build     --release     --locked     --all-features
export LD_LIBRARY_PATH=/home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/
cargo build     --release     --locked     --all-features     --target=x86_64-unknown-linux-gnu
cargo build     --release     --locked     --all-features
```

When hard setting LD LIBRARY PATH and checking with **ldd** - I get different unresolvable linking errors.
```
LD_LIBRARY_PATH=/home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/ ldd -v /home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build
/bin/bash: relocation error: /home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/libc.so.6: symbol _dl_fatal_printf version GLIBC_PRIVATE not defined in file ld-linux-x86-64.so.2 with link time reference

ldd -v /home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build | grep not
/home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.80-devel/build/ytop-5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc/.x86_64-libreelec-linux-gnu/release/build/libc-c09da03183744d59/build-script-build)
                libc.so.6 (GLIBC_2.32) => not found

### rust / cargo is showing that the cross compiler is installed ok.
rustup target list | grep installed
x86_64-unknown-linux-gnu (installed)

```

I have included the diff below for the Latest ytop, and have tried hard coding the library path in for cargo.

### Informations
 - LE Version: LE10 Head
 - Hardware Platform: Generic

### Log file

### Additional context

```
diff --git a/packages/addons/addon-depends/system-tools-depends/ytop/package.mk b/packages/addons/addon-depends/system-tools-depends/ytop/package.mk
index 87ad992b67..17df321691 100644
--- a/packages/addons/addon-depends/system-tools-depends/ytop/package.mk
+++ b/packages/addons/addon-depends/system-tools-depends/ytop/package.mk
@@ -2,8 +2,9 @@
 # Copyright (C) 2020-present Team LibreELEC (https://libreelec.tv)
 
 PKG_NAME="ytop"
-PKG_VERSION="0.6.2"
-PKG_SHA256="d9fd6ce00e27de894bc0790947fbeab40e81e34afa5ead5a53d126c458d50e99"
+PKG_VERSION="5d1e2fd1dc3edff9ea1025c93fcce8adc1f64ecc"
+#PKG_VERSION="0.6.2"
+#PKG_SHA256="d9fd6ce00e27de894bc0790947fbeab40e81e34afa5ead5a53d126c458d50e99"
 PKG_LICENSE="MIT"
 PKG_SITE="https://github.com/cjbassi/ytop"
 PKG_URL="https://github.com/cjbassi/ytop/archive/$PKG_VERSION.tar.gz"
@@ -16,6 +17,7 @@ make_target() {
   cargo build \
     --release \
     --locked \
+    --target=x86_64-unknown-linux-gnu \
     --all-features
 }
```

heitbaum, 2020-12-05:
Reference to cross compiling with rust - https://github.com/japaric/rust-cross

heitbaum, 2020-12-05:
Further investigation compiling the bottom package which is an an alternative to ytop (as ytop is deprecated) has the same issues. https://github.com/ClementTsang/bottom.

CvH, 2020-12-05:
librespot is not working too, at the buildbot it looks like it works perfect for everything besides generic

heitbaum, 2020-12-05:
From my reading around rustup - there is something missing in the target (ie the x86) is not using the tool chain.

heitbaum, 2020-12-05:
I think the fix is something around setting - rustc-link-search
Not sure if it can be done in rustup override or other. 

See:
- https://github.com/diwic/dbus-rs/issues/184
- https://doc.rust-lang.org/cargo/reference/config.html

mglae, 2020-12-13:
Rust is using the host compiler to create the `build-script-build` binaries and execute them for building the targets.

In the Generic case host triple is equal target triple (both x86_64-unknown-linux-gnu). It is not possible to use different tool chains for this case.

We do set up a GLIBC_2.32 tool chain but GLIBC_2.32 is not available in host system and the build-script-build binaries cannot be executed.

I'm afraid this was only working by accident in the past.

There seem only to be two ugly work arounds:

1) Use a host system with GLIBC_2.32.
2) Use rust nightly features to set up a x86_64-libreelec-linux-gnu.json definition and use it for the target architecture.

heitbaum, 2021-02-04:
Rust issues are noted as:
- https://github.com/rust-lang/cargo/issues/6582
- https://github.com/rust-lang/rust/issues/58394
- https://github.com/rust-lang/cargo/pull/6909
- https://github.com/Homebrew/linuxbrew-core/pull/21804

they provide insight on potential workaround / fix. 

ideally the cross compile would actually force a cross-compile regardless of the triple.

mglae, 2021-02-05:
I'm seeing no solution for us.

CvH, 2021-02-05:
So a dirty workaround would be a host os that has the same glib version?

mglae, 2021-02-05:
Yes, I've already successful tested with a groovy docker build container.

SupervisedThinking, 2021-02-14:
Well so even the latest rust version 1.50.0 is screwed? I tried to compile a recent versio of librsvg and ended with:

```
   Compiling language-tags v0.2.2
error: failed to run custom build command for `libc v0.2.80`

Caused by:
  process didn't exit successfully: `/home/supervisedthinking/development/LibreELEC-RR/build.LibreELEC-Generic.x86_64-9.80-devel/build/librsvg-2.51.0/.x86_64-libreelec-linux-gnu/release/build/libc-a9532ed5331eaa05/build-script-build` (exit code: 1)
  --- stderr
  /home/supervisedthinking/development/LibreELEC-RR/build.LibreELEC-Generic.x86_64-9.80-devel/build/librsvg-2.51.0/.x86_64-libreelec-linux-gnu/release/build/libc-a9532ed5331eaa05/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/supervisedthinking/development/LibreELEC-RR/build.LibreELEC-Generic.x86_64-9.80-devel/build/librsvg-2.51.0/.x86_64-libreelec-linux-gnu/release/build/libc-a9532ed5331eaa05/build-script-build)
warning: build failed, waiting for other jobs to finish...
error: build failed
FAILURE: scripts/build librsvg during make_target (package.mk)
*********** FAILED COMMAND ***********
cargo build --release --target=x86_64-unknown-linux-gnu --no-default-features
```

jameshilliard, 2021-04-02:
Possible fix for this in [#9322](https://github.com/rust-lang/cargo/pull/9322).

heitbaum, 2021-04-03:
Given the LE package dos not build rustc compiler. This needs a little bit of help / testing (not a just trivial .patch file.)

notes: 
1. Howto: https://github.com/rust-lang/rustc-dev-guide/blob/master/src/building/how-to-build-and-run.md
2. Grab this patch - https://github.com/rust-lang/cargo/pull/9322/commits/1597f843f4d3c7eb9e6f92487716608ed52ccda0.patch
3. Use the rust current release - https://github.com/rust-lang/rust/releases/tag/1.51.0 as the base
4. Build rustc (with/without the patch and test)
5. Compile librespot on x86 for x86

jameshilliard, 2021-04-03:
>     4\. Build rustc (with/without the patch and test)

FYI the patch is for [cargo](https://github.com/rust-lang/cargo/), not [rustc](https://github.com/rust-lang/rust/).

I'm not really familiar with LibreELEC myself so someone else will probably need to test this here, this bug appears to be identical to one I encountered when cross compiling with rust+cargo using [buildroot](https://buildroot.org/).

heitbaum, 2021-04-03:
This is our current package.mk script

https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/addons/addon-depends/librespot-depends/rust/package.mk

so essentially a download of the binaries. Need to move that to a compile to get the patch included as I understand it.

```
  "$(get_build_dir rustup.rs)/rustup-init.sh" \
    --default-toolchain "${PKG_VERSION}" \
    --no-modify-path \
    --profile minimal \
    --target "${RUST_TARGET_TRIPLE}" \
    -y
```

jameshilliard, 2021-04-03:
> Need to move that to a compile to get the patch included as I understand it.

I think all you have to do is build cargo from source with my patch after installing the prebuilt toolchain with rustup. AFAIK rustup includes both rustc and cargo, you only need to replace cargo.

heitbaum, 2021-04-10:
Using @jameshilliard patch and the following script. It is possible to build librespot on U18. It is a massive hack. But need to have cargo to build cargo. And the u18 cargo is not building cargo ...

I’m not at all happy with this hack. But maybe some groundwork we can use to workaround the cross compile until @jameshilliard patch is mainlined? Remembering the background to the failure is that cargo does not honour target cross compile if host=target. This came about because the LE glibc was advanced of the buildhost gcc.

here are the commits required to LE10 to get the below to work:
(https://github.com/heitbaum/LibreELEC.tv/commits/cargohack)
- https://github.com/heitbaum/LibreELEC.tv/commit/b7a8f28e58e2b18d34b266e4ad2efd6f6bcf4b5f
- https://github.com/heitbaum/LibreELEC.tv/commit/6d3672b2c68b18f9c367938919b20fdb9095ccd0

```
scripts/build rust:host
scripts/build cargo
scripts/clean rust
scripts/build rust

cp -p build.LibreELEC-Generic.x86_64-10.0-devel/build/cargo-0.52.0/target/release/cargo build.LibreELEC-Generic.x86_64-10.0-devel/build/rust-1.51.0/cargo/toolchains/1.51.0-x86_64-unknown-linux-gnu/bin/cargo

s/create_addon librespot
....
ALL ADDONS BUILT SUCCESSFULLY
```

heitbaum, 2021-06-19:
I have update rust to nightly, now that https://github.com/rust-lang/cargo/pull/9322 GS been merged, using these instructions.
- https://rust-lang.github.io/rustup/installation/index.html

**rust version 1.55.0-nightly (ce1d5611a 2021-06-18)**

@jameshilliard  - Do we still need to wait on https://github.com/rust-lang/cargo/pull/9532 before this will start working? Still getting the libraries from the cross not being referenced.

```
$ s/build bottom     
UNPACK      bottom
BUILD      bottom (target)
    TOOLCHAIN      manual
    Updating crates.io index
…
error: failed to run custom build command for `memchr v2.4.0`

Caused by:
  process didn't exit successfully: `/storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/bottom-0.6.1/.x86_64-libreelec-linux-gnu/release/build/memchr-b98b661f62721450/build-script-build` (exit status: 1)
  --- stderr
  /storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/bottom-0.6.1/.x86_64-libreelec-linux-gnu/release/build/memchr-b98b661f62721450/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/bottom-0.6.1/.x86_64-libreelec-linux-gnu/release/build/memchr-b98b661f62721450/build-script-build)
  /storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/bottom-0.6.1/.x86_64-libreelec-linux-gnu/release/build/memchr-b98b661f62721450/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/bottom-0.6.1/.x86_64-libreelec-linux-gnu/release/build/memchr-b98b661f62721450/build-script-build)
warning: build failed, waiting for other jobs to finish...
error: build failed
FAILURE: s/build bottom during make_target (package.mk)
*********** FAILED COMMAND ***********
cargo build --release --locked --all-features
**************************************
```

```
$ s/build rust:host
BUILD      rust (host)
    TOOLCHAIN      manual
info: downloading installer
info: profile set to 'minimal'
info: default host triple is x86_64-unknown-linux-gnu
info: skipping toolchain installation

Rust is installed now. Great!

To get started you need Cargo's bin directory 
(/storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/rust-nightly/cargo/bin)
in your PATH
environment variable. This has not been done automatically.

To configure your current shell, run:
source 
/storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/rust-nightly/cargo/env
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'
info: latest update on 2021-06-19, rust version 1.55.0-nightly (ce1d5611a 2021-06-18)
info: downloading component 'cargo'
info: downloading component 'clippy'
info: downloading component 'rust-std'
 22.0 MiB /  22.0 MiB (100 %)  10.8 MiB/s in  2s ETA:  0s
info: downloading component 'rustc'
 50.1 MiB /  50.1 MiB (100 %)  11.0 MiB/s in  4s ETA:  0s
info: installing component 'cargo'
info: installing component 'clippy'
info: installing component 'rust-std'
 22.0 MiB /  22.0 MiB (100 %)  17.9 MiB/s in  1s ETA:  0s
info: installing component 'rustc'
 50.1 MiB /  50.1 MiB (100 %)  20.2 MiB/s in  2s ETA:  0s

  nightly-x86_64-unknown-linux-gnu installed - rustc 1.55.0-nightly (ce1d5611a 2021-06-18)

info: default toolchain set to 'nightly-x86_64-unknown-linux-gnu'
info: checking for self-updates

$ s/build rust:target
BUILD      rust (target)
    TOOLCHAIN      manual
info: downloading installer
warning: it looks like you have an existing installation of Rust at:
warning: /storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/rust-nightly/cargo/bin
warning: rustup should not be installed alongside Rust. Please uninstall your existing Rust first.
warning: Otherwise you may have confusion unless you are careful with your PATH
warning: If you are sure that you want both rustup and your already installed Rust
warning: then please reply `y' or `yes' or set RUSTUP_INIT_SKIP_PATH_CHECK to yes
warning: or pass `-y' to ignore all ignorable checks.
error: cannot install while Rust is installed
warning: continuing (because the -y flag is set and the error is ignorable)
info: profile set to 'minimal'
info: default host triple is x86_64-unknown-linux-gnu
info: skipping toolchain installation
warning: ignoring requested target: x86_64-unknown-linux-gnu

Rust is installed now. Great!

To get started you need Cargo's bin directory 
(/storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/rust-nightly/cargo/bin)
in your PATH
environment variable. This has not been done automatically.

To configure your current shell, run:
source 
/storage/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/rust-nightly/cargo/env
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'
info: latest update on 2021-06-19, rust version 1.55.0-nightly (ce1d5611a 2021-06-18)
info: component 'clippy' for target 'x86_64-unknown-linux-gnu' is up to date

  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.55.0-nightly (ce1d5611a 2021-06-18)

info: checking for self-updates
```

jameshilliard, 2021-06-19:
Are you setting `-Z target-applies-to-host` in the cli and `CARGO_TARGET_APPLIES_TO_HOST=false` in the environment?

jameshilliard, 2021-06-19:
> Do we still need to wait on rust-lang/cargo#9532 before this will start working? Still getting the libraries from the cross not being referenced.

no

Are you setting `-Z target-applies-to-host` in the cli and `CARGO_TARGET_APPLIES_TO_HOST=false` in the environment?

heitbaum, 2021-06-19:
> no
>
> Are you setting `-Z target-applies-to-host` in the cli and `CARGO_TARGET_APPLIES_TO_HOST=false` in the environment?

Thanks for the howto. Draft PR updated and it works 😎 - #5446

closed 2021-08-03 by CvH