LibreELEC/LibreELEC.tv #4722: [BUG] Missing configuration parameter for DWC2 at RPI4

alex1702, 2020-12-06:
### Describe the bug
In the current 9.2.6 and also in the master the configuration for dwc2 'CONFIG_USB_DWC2=y' is missing in the 'projects/RPi/devices/RPi4/linux/linux.arm.conf' is not set.
Without this option there is no kernel module to switch the Power-USB-C port of the Raspberry Pi 4 to OTG mode via DWC2.
I have the new case [DeskPi Pro](https://deskpi.com/products/deskpi-pro-for-raspberry-pi-4), this one connects the power button, the fan control and a USB hub via the OTG port.

I have 2 pull requests for it in which I adjust the file directly, but I don't know if it is correct there, because the file says "Automatically generated file; DO NOT EDIT". If it has to be adapted somewhere else, please explain briefly, then I adapt the pull requests.

### Informations
Libreelec 9.2.6
RPi4
DeskPi Pro Case

HiassofT, 2020-12-06:
can you test if adding `otg_mode=1` to config.txt fixes this? Please test this both on LE 9.2.6 and also on the latest master nightly build from test.libreelec.tv

alex1702, 2020-12-06:
I should only enter the string "otg_mode=1" alone in the file? At least I do not know this option. 

The installation script from the DeskPi Pro Team ( [link to the installation script](https://github.com/DeskPi-Team/deskpi/blob/master/install.sh) ) adds the line `dtoverlay=dwc2,dr_mode=host`, I tested with that. I had also tried the alternative option `dtoverlay=dwc-otg`, but with that it did not recognize USB devices.

HiassofT, 2020-12-06:
Just add a line with `otg_mode=1` to the end of /flash/config.txt, just like you'd add the dtoverlay line.

This option will enable the builtin XHCI controller on the USB-C port instead of the DWC2 controller. You can find more info about that option on the RPi forums.

I haven't tried this myself yet, but as the last beta bootloader added support for USB booting using the XHCI controller - see https://github.com/raspberrypi/rpi-eeprom/commit/3702e706aff73a0b0ad545223f9aa4b7d98aeb5c - chances are good that it might just work fine (at least on the latest master builds, which contain latest kernel and firmware)

alex1702, 2020-12-06:
> can you test if adding `otg_mode=1` to config.txt fixes this? Please test this both on LE 9.2.6 and also on the latest master nightly build from test.libreelec.tv

Does not work in libreelec 9.2, but in the current master the option works.

For me I think it would be enough to use this new option and continue using libreelec master. Is the current master already stable?
Can I update from 9.2.6 via img file without problems?

Offtopic:
Unfortunately I couldn't get into the IRC channel yesterday because I couldn't get it with the registration and after 5 new nicknames I still couldn't get in. Otherwise you could have solved it faster there.

HiassofT, 2020-12-07:
Thanks a lot for testing, it's good to know this is working fine in master.

RPi4 master builds are rather usable, but still experimental and has some known issues. As there are many changes compared to 9.2 we can't guarantee that upgrading will work fine. It may, but you probably need to adjust several settings.

I highly recommend you create a full backup of your existing installation as there's no easy way to go back to 9.2 in case you run into issues.

BTW: can you close your 9.2 and master PRs? We currently have no plans to do further 9.2 releases and the kernel config changes are incomplete, too.

alex1702, 2020-12-08:
Thanks a lot for the information, that sounds quite good.
The listed problems do not concern me at the moment. I will try my setup to see if it is compatible with the latest master version.

I closed the PR's, but to learn a little bit more, what was missing in the kernel adjustments?

closed 2020-12-08 by alex1702

HiassofT, 2020-12-08:
kernel configs should be changed with `make menuconfig` or by running `make oldconfig` and answering the questions after changing some option(s). Often kernel options enable other options, eg when enabling CONFIG_USB_DWC2=y the full kernel config diff would look like this:

```
--- a/projects/RPi/devices/RPi4/linux/linux.arm.conf
+++ b/projects/RPi/devices/RPi4/linux/linux.arm.conf
@@ -4744,7 +4744,15 @@ CONFIG_USB_UAS=y
 # CONFIG_USB_CDNS3 is not set
 # CONFIG_USB_MUSB_HDRC is not set
 # CONFIG_USB_DWC3 is not set
-# CONFIG_USB_DWC2 is not set
+CONFIG_USB_DWC2=y
+CONFIG_USB_DWC2_HOST=y
+
+#
+# Gadget/Dual-role mode requires USB Gadget support to be enabled
+#
+# CONFIG_USB_DWC2_PCI is not set
+# CONFIG_USB_DWC2_DEBUG is not set
+# CONFIG_USB_DWC2_TRACK_MISSED_SOFS is not set
 # CONFIG_USB_ISP1760 is not set
 
 #
@@ -5363,6 +5371,7 @@ CONFIG_RESET_SIMPLE=y
 # CONFIG_PHY_PXA_28NM_USB2 is not set
 # CONFIG_PHY_MAPPHONE_MDM6600 is not set
 # CONFIG_PHY_OCELOT_SERDES is not set
+# CONFIG_PHY_SAMSUNG_USB2 is not set
 # end of PHY Subsystem
 
 # CONFIG_POWERCAP is not set
```