LibreELEC/LibreELEC.tv #4728: [BUG] Videos (Amazon VOD, Netflix) won't start on RPi 4, endless spinner, stopping is not possible

dkadioglu, 2020-12-07:
### Describe the bug
Whatever video I want to play, nothing happens and I only see the spinner, endless. I can still navigate through KODI, but stopping is not possible. Therefor, playing another item is also not possible and the only thing which helps is restarting KODI.

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'Whatever video in Netflix or Amazon VOD'
2. Play 'Any video'
3. Endless spinner is visible

### Informations
 - LE Version: 9.80
 - Hardware Platform: RPi4

### Log file
http://ix.io/2H3q

### Additional context
Already filed against Netflix Addon -> https://github.com/CastagnaIT/plugin.video.netflix/issues/964

BugDeliverer, 2020-12-09:
As already mentioned in the referenced issue this bug also affects RPi 3. Tried a clean install of newest testing images. While showing loading spinner high cpu usage by process kodi.bin.

HiassofT, 2020-12-11:
Please update to the latest nightly and make sure inputstream.adaptive has been updated to version 2.6.6.2 (2.6.6.1 has this hanging issue). You may need to restart kodi after the inputstream.adaptive update.

Netflix is working fine here on RPi4 with these versions

closed 2020-12-11 by dkadioglu