LibreELEC/LibreELEC.tv #4737: [BUG] incomplete dbussy installation with PYTHONDONTWRITEBYTECODE=1

mklein-de, 2020-12-11:
### Describe the bug
When PYTHONDONTWRITEBYTECODE=1 is set, the dbussy package does not install any python modules.

### To Reproduce
Steps to reproduce the behavior:
```
$ PYTHONDONTWRITEBYTECODE=1 ./scripts/install dbussy
BUILD      dbussy (target)
    TOOLCHAIN      manual
running build
running build_py
warning: build_py: byte-compiling is disabled, skipping.

running install
running build
running build_py
warning: build_py: byte-compiling is disabled, skipping.

running install_lib
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
copying build/lib/dbussy.py -> /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
copying build/lib/ravel.py -> /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
warning: install_lib: byte-compiling is disabled, skipping.

running install_egg_info
Writing /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages/DBussy-1.3-py3.8.egg-info
INSTALL      dbussy (target)

$ find build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3                                                       
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages/DBussy-1.3-py3.8.egg-info
build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/.libreelec-package
```

This looks better:
```
$ PYTHONDONTWRITEBYTECODE=0 ./scripts/install dbussy                                                                       
BUILD      dbussy (target)
    TOOLCHAIN      manual
running build
running build_py
running install
running build
running build_py
running install_lib
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8
creating /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
copying build/lib/dbussy.py -> /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
copying build/lib/ravel.py -> /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages
byte-compiling /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages/dbussy.py to dbussy.cpython-38.pyc
byte-compiling /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages/ravel.py to ravel.cpython-38.pyc
running install_egg_info
Writing /home/michael/LibreELEC.tv/build.LibreELEC-H2-plus.arm-9.80-devel/install_pkg/dbussy-1.3/usr/lib/python3.8/site-packages/DBussy-1.3-py3.8.egg-info
INSTALL      dbussy (target)
```

### Informations
 - LE Version: 9.80-devel (05c81a0e7132f53ef38d3b74d4db7455193f7fc8)
 - Hardware Platform: Allwinner H2 Plus

HiassofT, 2020-12-11:
I don't see a bug here.

First, you are not supposed to run scripts/install - this is an internal script used by the build system.

Second, you are setting some non-standard environment variable - the build system doesn't clear all environment variables so this means you just shot yourself into your foot.

There are only a few environment variables you are supposed to set when building LE, eg ARCH, PROJECT, DEVICE, DEBUG and a few others, anything else will result in undefined behaviour

closed 2020-12-11 by HiassofT

mklein-de, 2020-12-12:
> First, you are not supposed to run scripts/install - this is an internal script used by the build system.

These are just the minimal steps to demonstrate the issue. It's the same behavior if you run it though the build system. If you have `PYTHONDONTWRITEBYTECODE=1` in your shell to avoid .pyc clutter in the filesystem, at least one python-related package is broken.
 
> Second, you are setting some non-standard environment variable - the build system doesn't clear all environment variables so this means you just shot yourself into your foot.

What are the standard environment variables? There are of course many ways to break the build system by setting some obscure variable. Since the build system already clears `PYTHONHOME`, `PYTHONPATH` and `PYTHONSTARTUP`, I don't see why it should not also clear `PYTHONDONTWRITEBYTECODE`, if this is known to cause breakage.
 
> There are only a few environment variables you are supposed to set when building LE, eg ARCH, PROJECT, DEVICE, DEBUG and a few others, anything else will result in undefined behaviour

HiassofT, 2020-12-12:
Standard environment variables are those set up by a minimal distribution installation. So if you are using a distribution that uses eg `PYTHONDONTWRITEBYTECODE=1` in it's default env please tell us so.

There's a reason we recommend building LibreELEC in a minimal environment (container/chroot/VM/dedicated box/...), see the basic build documentation on our wiki https://libreelec.wiki/development-1/build-basics#environment 

There are simply too many ways host packages, environment settings etc can leak into the LibreELEC build and result in odd build issues. I'm all in for making LibreELEC build more stable and isolating it from the build host environment, but I don't consider build breakage introduced by custom user settings a bug. Build breakage caused by installed host packages probably qualify as a bug but even that is a tedious cat-and-mouse game - therefore the pragmatic approach of recommending to use a clean, minimal build env.

thoradia, 2020-12-12:
I has issues with [this](https://github.com/thoradia/LibreELEC.tv/commit/39eac758745a1e1b6544e695bcf1cb1ecc249b7e)

HiassofT, 2020-12-12:
BTW: I'm not opposed at all to making the LE build system more robust (completely on the contrary). I just see this more as an improvement rather than an actual bug - so it's best to keep the discussion in your PR (which didn't exist when I started commenting on this issue)

mklein-de, 2020-12-12:
> BTW: I'm not opposed at all to making the LE build system more robust (completely on the contrary). I just see this more as an improvement rather than an actual bug - so it's best to keep the discussion in your PR (which didn't exist when I started commenting on this issue)

Fair enough. I didn't add the Bug tag on purpose, that's just left from the issue template.