LibreELEC/LibreELEC.tv #4742: [ffplay] hevc_v4l2m2m @ 0xa9fc1b10] Could not find a valid device

sknbl, 2020-12-12:
RPI4

[hevc_v4l2m2m @ 0xa9fc1b10] Could not find a valid device
[hevc_v4l2m2m @ 0xa9fc1b10] can't configure decoder

./configure \
 --enable-neon\
 --enable-vdpau\
 --enable-vaapi\
 --enable-v4l2_m2m --enable-libdrm\
 --enable-hwaccels\
 --enable-shared\
 --disable-static\
 --enable-decoder=hevc_rpi\
 --prefix=$USR_PREFIX\
 --libdir=$LIB_PREFIX\
 --incdir=$INC_PREFIX\
 --cpu=cortex-a7\
 --enable-debug --disable-stripping\
 --disable-thumb\
 --disable-mmal\
 --disable-rpi --enable-sand\
 --disable-hwaccel=h264_v4l2request \
 --disable-hwaccel=mpeg2_v4l2request \
 --disable-hwaccel=vp8_v4l2request \
 --disable-hwaccel=vp9_v4l2request \
 --enable-libudev --enable-v4l2-request\
 --enable-vout-drm\
 --extra-cflags="-ggdb $RPI_KEEPS $RPI_DEFINES $RPI_INCLUDES"\
 --extra-cxxflags="$RPI_DEFINES $RPI_INCLUDES"\
 --extra-ldflags="$RPI_LIBDIRS"\
 --extra-libs="-Wl,--start-group -lbcm_host -lmmal -lmmal_util -lmmal_core -lvcos -lvcsm -lvchostif -lvchiq_arm"\

lrusak, 2020-12-12:
Can you be more clear? What are you actually doing? Are you self building? Are you using a nightly build?

sknbl, 2020-12-12:
> Can you be more clear? What are you actually doing? Are you self building? Are you using a nightly build?

hi, @lrusak! Thank you for quick response. I'm trying to build FFmpeg-4.3.1-Matrix-Beta1 separately

lrusak, 2020-12-12:
> > Can you be more clear? What are you actually doing? Are you self building? Are you using a nightly build?
> 
> hi, @lrusak! Thank you for quick response. I'm trying to build FFmpeg-4.3.1-Matrix-Beta1 separately

So are you using LibreELEC at all? Are you using the correct v4l2 patchsets? Do you have v4l2 support in your kernel?

sknbl, 2020-12-12:
To be precise, i want to be able to play hevc files with ffplay on my rpi4 with archarm on it

lrusak, 2020-12-12:
closing as this isn't relevant to LibreELEC. Please use the forum for support.

closed 2020-12-12 by lrusak