LibreELEC/LibreELEC.tv #4779: [BUG] Final output image overflows by 1 pixel to the right, pixels wrap around to the left of the screen

BreakPoints, 2020-12-19:
### Describe the bug
Using, as of now, the latest nightly build all output from the RPi4 is offset by one pixel to the right leading to the final column of pixels to wrap around to the left side of the screen. This happens in the UI and in video playback but is most noticeable when light content is displayed on the right of the screen and dark content on the left. This leads to a line of bright pixels on the left of the screen.

This can easily be seen when browsing the Settings -> Display options as the white scroll bar on the far right overflows into the left of the screen

### To Reproduce
Steps to reproduce the behavior:
1. Flash 9.80-nightly-20201219-c7aa22f to an SD card
2. Boot RPi4
3. Issue can be observed in the UI and in video playback. Most easily noticeable in settings menus with a scroll bar on the right (example: System Settings -> Display)

### Informations
 - LE Version: 9.80-nightly-20201219-c7aa22f
 - Hardware Platform: RPi4

### Log file
https://pastebin.com/epNMRjUp

### Additional context
Unfortunately this cannot be captured with the screenshot function so I have attached both the screenshot that does not show the issue and taken the retro route of pointing my phones camera at the screen to give a visual.

White scroll bar and blue highlight can be seen overflowing to the far left
[image]

Screenshot of the same screen using Kodi screenshot feature
[image]

HiassofT, 2020-12-19:
This is a known issue in the RPi4 video driver, see here https://github.com/raspberrypi/linux/issues/3667

BreakPoints, 2020-12-19:
> This is a known issue in the RPi4 video driver, see here [raspberrypi/linux#3667](https://github.com/raspberrypi/linux/issues/3667)

Ah, did not realise this was a video driver issue and not libreELEC. Thanks for pointing me to the right place.
Closing

closed 2020-12-19 by BreakPoints

BreakPoints, 2020-12-19:
As an aside, does this mean Kodi renders video to a quad that goes off screen? Would that not also be an issue with Kodi as this happens with videos

HiassofT, 2020-12-19:
The description in the RPi kernel bug report may be a bit misleading and probably was the first test case of the reporter where it showed up.

The problem is that the right-most pixel column wraps around to the left and as this is often blank (eg on text screens) or in uniform background colour it's often not visible when that happens.

Also the issue is a bit random, sometimes display is fine sometimes not (which makes it a bit hard to debug...).

For a while we carried a workaround to mask that issue (it simply avoided any content in the left-most and right-most columns), it wasn't always working but we are currently thinking about adding at back until the underlying problem is fixes as the issue is somewhat distracting (both in menus and when watching movies)

dkadioglu, 2021-01-23:
Are there any news regarding a fix for that? Or what's your decision about reintroducing the workaround? I can only second that it is distracting. Just to mention, I see that on the most recent Libreelec nightly for RPi 4.Thank you very much for an update on this. And for sure for your hard work in constantly improving support for RPi 4!