LibreELEC/LibreELEC.tv #4780: [BUG] LCDd doesn't work with PiFaceCAD, an upstream fix is available

kisik21, 2020-12-21:
### Describe the bug

### To Reproduce
Steps to reproduce the behavior:
1. Get a PiFace Control and Display unit
2. Set LCDd driver to "hd44780", tweak LCDd.conf to set appropriate driver "pifacecad"
3. Run lcdd.start from SSH (to see the error directly)
4. Observe the error and the backlit but empty display

### Informations
 - LE Version: 9.2.6
 - Hardware Platform: RPi2

### Log file
It spews a lot of copies of this string:

```
HD44780: PiFaceCAD: mcp23s17_write_reg: There was a error during the SPI transaction: Invalid argument
```

### Additional context

[lcdd](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/addons/service/lcdd/package.mk) should be updated to at least [3a3d622d9bb74c44fa67bc20573751a207514134](https://github.com/lcdproc/lcdproc/commit/3a3d622d9bb74c44fa67bc20573751a207514134) - this should fix the error.

heitbaum, 2020-12-23:
@kisik21 - Updated lcdd - https://github.com/heitbaum/LibreELEC.tv/commit/3a75a7c74f618a538cdec9dc85f48f73bf664af2

heitbaum, 2020-12-23:
@kisik21  - RPi2 add on built, could you please test. https://github.com/heitbaum/LibreELEC.tv/raw/b0897b6a1fb044f7d14788f147ffdbfa5b1b1682/service.lcdd-9.2.0.107.zip

kisik21, 2020-12-24:
@heitbaum can confirm that it works!

closed 2020-12-29 by CvH