LibreELEC/LibreELEC.tv #4825: [BUG] ConnMan does not respect proxy configuration

Kariton, 2021-01-01:
### Describe the bug

There is at minimum one *thing* not respecting the configured proxy.
```[...] SRC=LibreELEC DST=212.227.81.55 LEN=60 TOS=0x00 PREC=0x00 TTL=63 ID=4503 DF PROTO=TCP SPT=37236 DPT=80 WINDOW=64240 RES=0x00 SYN URGP=0```

according to [this thread](https://forum.libreelec.tv/thread/21962-which-url-s-are-needed-for-libreelec-core/)
this is coused by the ```ConnMan``` service for checking the internet connectivity.
after reading the thread it would explain my strange update behavior.

I asume this is caused by ```ConnMan```

### To Reproduce
Steps to reproduce the behavior:
1. Fresh LibreELEC install from the [LibreELEC-Generic.x86_64-9.2.6.img.gz](https://libreelec.tv/downloads_new/generic-pc/) behind 'block all' firewall.
2. configure proxy under 'settings -> system -> internet access'
3. See firewall log: ```[...] SRC=LibreELEC DST=212.227.81.55 [...] PROTO=TCP [..] DPT=80 [...]```

### Informations
 - LE Version: Kodi 18.9 (Linux Kernel 5.1.x)
 - Hardware Platform: Intel x86

### Log file

-
### Additional context

due to the nature of ```ConnMan``` there will be no updates applied if this check fails.
Nevertheless i got past that requirement by simply navigate to 'Add-ons' and open the window.
i gont instandly the pop-up info for the new updates (previously enabled in the settings)

Kariton, 2021-01-01:
```
LibreELECliving:~ # connmanctl services
*AR Wired                ethernet_70[...]_cable

LibreELECliving:~ # connmanctl services ethernet_70[...]_cable
/net/connman/service/ethernet_70[...]_cable
  Type = ethernet
  Security = [  ]
  State = ready
  Favorite = True
  Immutable = False
  AutoConnect = True
  Name = Wired
  Ethernet = [ Method=auto, Interface=eth0, Address=[...], MTU=1500 ]
  IPv4 = [ Method=manual, Address=[...], Netmask=[...], Gateway=[...] ]
  IPv4.Configuration = [ Method=manual, Address=[...], Netmask=[...], Gateway=[...] ]
  IPv6 = [  ]
  IPv6.Configuration = [ Method=off ]
  Nameservers = [ 8.8.8.8, 8.8.4.4 ]
  Nameservers.Configuration = [ 8.8.8.8, 8.8.4.4 ]
  Timeservers = [ 0.pool.ntp.org, 1.pool.ntp.org, 2.pool.ntp.org, 3.pool.ntp.org ]
  Timeservers.Configuration = [ 0.pool.ntp.org, 1.pool.ntp.org, 2.pool.ntp.org ]
  Domains = [  ]
  Domains.Configuration = [  ]
  Proxy = [ Method=direct ]
  Proxy.Configuration = [  ]
  mDNS = False
  mDNS.Configuration = False
  Provider = [  ]
```
#apply configuration manually
```
LibreELECliving:~ # connmanctl config ethernet_70[...]_cable --proxy manual http://[...]:3128/

LibreELECliving:~ # connmanctl services ethernet_70[...]_cable
/net/connman/service/ethernet_70[...]_cable
  Type = ethernet
  Security = [  ]
  State = ready
  Favorite = True
  Immutable = False
  AutoConnect = True
  Name = Wired
  Ethernet = [ Method=auto, Interface=eth0, Address=[...], MTU=1500 ]
  IPv4 = [ Method=manual, Address=[...], Netmask=[...], Gateway=[...] ]
  IPv4.Configuration = [ Method=manual, Address=[...], Netmask=[...], Gateway=[...] ]
  IPv6 = [  ]
  IPv6.Configuration = [ Method=off ]
  Nameservers = [ 8.8.8.8, 8.8.4.4 ]
  Nameservers.Configuration = [ 8.8.8.8, 8.8.4.4 ]
  Timeservers = [ 0.pool.ntp.org, 1.pool.ntp.org, 2.pool.ntp.org, 3.pool.ntp.org ]
  Timeservers.Configuration = [ 0.pool.ntp.org, 1.pool.ntp.org, 2.pool.ntp.org ]
  Domains = [  ]
  Domains.Configuration = [  ]
  Proxy = [ Servers=[ http://[...]:3128/ ], Excludes=[  ], Method=manual ]
  Proxy.Configuration = [ Servers=[ http:/[...]:3128/ ], Method=manual ]
  mDNS = False
  mDNS.Configuration = False
  Provider = [  ]
  ```
  
  seems not to help - still got hits on the firewall

chewitt, 2021-01-01:
Kodi proxy settings are only applied to Kodi itself and they have no effect upon services that run in the OS layer, so ConnMan bypasss is expected within the current design (not a bug). I can't speak for ConnMan bypassing its own proxy settings, but that's a question for the ConnMan maintainers. @igaw ^ any comment?

Kariton, 2021-01-01:
@chewitt 
Thank you for your fast reply.

This is understandable.
is there a way to change the behavor of kodi itself?
like a DNS request / check to be "online"?

is there somethink i can do locally in my installation in general about that?

i mean i could opening the firewall?
if that IS needed it would be possible but i would like to avoid that.
i think in this case i woul periodically open the add-on manager.

igaw, 2021-01-04:
Is pacrunner used as proxy daemon? IIRC ConnMan will only use proxy settings with pacrunner, see 

  https://git.kernel.org/pub/scm/network/connman/pacrunner.git/tree/doc/architecture.txt

chewitt, 2021-01-16:
LE does not implement pacrunner so I'd guess that's the explanation.

chewitt, 2021-03-08:
I'm going to close this ticket as there is no plan to alter how the current OS works (PRs to review are always welcome) and I think we've been able to clarify current behaviour. Thanks.

closed 2021-03-08 by chewitt