LibreELEC/LibreELEC.tv #4841: [BUG] Does not announce _ssh._tcp

probonopd, 2021-01-03:
### Describe the bug

### To Reproduce
Steps to reproduce the behavior:
1. Go to a Zeroconf enabled Terminal application (such as the one on the Mac)
2. You will not see OpenELEC there

### Additional context

Currently there is only `/etc/avahi/services/sftp-ssh.service` which announces `_sftp-ssh._tcp`.
There should also be `/etc/avahi/services/ssh.service` which should announce `_ssh._tcp`.

Since the filesystem is read-only, users cannot easily fix this manually.

lrusak, 2021-01-03:
We purposely remove this file. See, https://github.com/LibreELEC/LibreELEC.tv/blob/d37672b621a90657ce94b978e8c455ead08322c7/packages/network/avahi/package.mk#L74

I'm not sure the original reason to remove it (it's been like this for 8+ years). I don't really have an issue with adding this file as security isn't really as much of a concern now that we have a configurable root password.

probonopd, 2021-01-03:
Please do not remove this file, since you are already advertising (thank you!) sftp the security vector is identical.

lrusak, 2021-01-03:
> Please do not remove this file, since you are already advertising (thank you!) sftp the security vector is identical.

Please file a PR if you would like it changed 😃

chewitt, 2021-01-04:
What does not-removing it achieve? i.e. what use-case is enabled/served by not-removing it vs. the existing ssh/sftp service which is broadcasting basically the same info?

HiassofT, 2021-03-19:
closing as fix has been merged to master

closed 2021-03-19 by HiassofT

probonopd, 2021-03-20:
Thank you very much everyone. Really helpful.