LibreELEC/LibreELEC.tv #4997: [BUG] ISO playback (DVD) does not have audio streams

heitbaum, 2021-01-18:
### Describe the bug

ISO Playback does not detect audio streams.

Taking the same iso and extracting the mpeg-2 stream into an .mpg file **does** work.

**mplayer -identify dvd://2 -nocache -dvd-device l.iso -dumpstream -dumpfile l.mpg**

**ffprobe l.mpg**
Input #0, mpeg, from 'l.mpg':
  Duration: 01:56:44.99, start: 0.287267, bitrate: 5697 kb/s
    Stream #0:0[0x1e0]: Video: mpeg2video (Main), yuv420p(tv, top first), 720x576 [SAR 64:45 DAR 16:9], 25 fps, 25 tbr, 90k tbn, 50 tbc
    Stream #0:1[0x80]: Audio: ac3, 48000 Hz, 5.1(side), fltp, 448 kb/s
    Stream #0:2[0x81]: Audio: ac3, 48000 Hz, stereo, fltp, 192 kb/s

### To Reproduce
Steps to reproduce the behavior:
1. Go to play an ISO
2. Error as attached log
(Unsure where this bug has crept in) will need to do some rollback and further testing.

### Informations
 - LE Version: LE10 - https://github.com/heitbaum/LibreELEC.tv/tree/dev
 - Hardware Platform: Generic

### Log file

- http://ix.io/2MqR - ISO DVD (not working) with no audio
- http://ix.io/2Mr6 - Title 2 of the ISO as an MPG (works)

**mplayer -identify dvd://2 -dvd-device l.iso**
...
ID_DVD_CURRENT_TITLE=2
There are 1 angles in this DVD title.
audio stream: 0 format: ac3 (5.1) language: en aid: 128.

ID_AUDIO_ID=128
ID_AID_128_LANG=en
audio stream: 1 format: ac3 (stereo) language: en aid: 129.

ID_AUDIO_ID=129
ID_AID_129_LANG=en
number of audio channels on disk: 2.
...
MPEG-PS file format detected.
MPEG: No audio stream found -> no sound. <—- think this is a false error.

**lsdvd l.iso**
...
Title: 02, Length: 01:56:45.000 Chapters: 29, Cells: 30, Audio streams: 02, Subpictures: 03
        Audio: 1, Language: en - English, Format: ac3, Frequency: 48000, Quantization: drc, Channels: 6, AP: 0, Content: Undefined, Stream id: 0x80
        Audio: 2, Language: en - English, Format: ac3, Frequency: 48000, Quantization: drc, Channels: 2, AP: 0, Content: Undefined, Stream id: 0x81

### Additional context

Issue detected in the last few days with master. 
MKV, MPG and MP4 fine.

CvH, 2021-01-18:
@DaVukovic afaik you had similar problems ?

heitbaum, 2021-01-19:
I can confirm that the build c3d5c93 (which was head on Jan 18, 2021) does work. The regression was introduced in https://github.com/howie-f/libdvdnav/releases/tag/6.1.0-Matrix-Beta-2 (which has yet to be committed to master.)

libdvdcss: update to 1.4.3-Matrix-Beta-2 - works
libdvdread: update to 6.1.1-Matrix-Beta-2 - works
libdvdnav: update to 6.1.0-Matrix-Beta-2 - BROKEN (no audio) (https://github.com/howie-f/libdvdnav/releases/tag/6.1.0-Matrix-Beta-2)
libdvdnav: 6.0.0-Leia-Alpha-3 - works (https://github.com/xbmc/libdvdnav/releases/tag/6.0.0-Leia-Alpha-3)

https://github.com/xbmc/libdvdnav/compare/6.0.0-Leia-Alpha-3...howie-f:6.1.0-Matrix-Beta-2

the working log is:
```
2021-01-19 16:46:23.214 T:1267     INFO <general>: VideoPlayer::OpenFile: l.iso
2021-01-19 16:46:23.215 T:3920     INFO <general>: Creating InputStream
2021-01-19 16:46:23.268 T:3920     INFO <general>:   msg: libdvdnav: Using dvdnav version 6.0.0
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg: libdvdnav: vm: DVD Title: l
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg: libdvdnav: DVD disk reports itself with Region mask 0x00000000. Regions:
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  1
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  2
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  3
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  4
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  5
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  6
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  7
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg:  8
2021-01-19 16:46:23.290 T:3920     INFO <general>:   msg: libdvdread: Get key for /VIDEO_TS/VIDEO_TS.VOB at 0x00000127
2021-01-19 16:46:23.291 T:3920     INFO <general>:   msg: libdvdread: Elapsed time 0
2021-01-19 16:46:23.291 T:3920     INFO <general>:   msg: libdvdread: Get key for /VIDEO_TS/VTS_01_0.VOB at 0x000001d4
2021-01-19 16:46:23.292 T:3920     INFO <general>:   msg: libdvdread: Elapsed time 0
2021-01-19 16:46:23.292 T:3920     INFO <general>:   msg: libdvdread: Get key for /VIDEO_TS/VTS_01_1.VOB at 0x00001d58
2021-01-19 16:46:23.292 T:3920     INFO <general>:   msg: libdvdread: Elapsed time 0
2021-01-19 16:46:23.292 T:3920     INFO <general>:   msg: libdvdread: Get key for /VIDEO_TS/VTS_02_0.VOB at 0x002f48a8
2021-01-19 16:46:23.292 T:3920     INFO <general>:   msg: libdvdread: Elapsed time 0
2021-01-19 16:46:23.292 T:3920     INFO <general>:   msg: libdvdread: Get key for /VIDEO_TS/VTS_02_1.VOB at 0x002f48ff
2021-01-19 16:46:23.293 T:3920     INFO <general>:   msg: libdvdread: Elapsed time 0
2021-01-19 16:46:23.293 T:3920     INFO <general>:   msg: libdvdread: Found 2 VTS's
2021-01-19 16:46:23.293 T:3920     INFO <general>:   msg: libdvdread: Elapsed time 0
2021-01-19 16:46:23.298 T:3920     INFO <general>: VideoPlayer: playing a file with menu's
2021-01-19 16:46:23.298 T:3920     INFO <general>: Creating Demuxer
2021-01-19 16:46:23.298 T:3920  WARNING <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
2021-01-19 16:46:23.298 T:3920  WARNING <general>: CDVDMessageQueue(video)::Put MSGQ_NOT_INITIALIZED
2021-01-19 16:46:23.298 T:3920  WARNING <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
2021-01-19 16:46:23.298 T:3920  WARNING <general>: CDVDMessageQueue(video)::Put MSGQ_NOT_INITIALIZED
2021-01-19 16:46:24.298 T:3920  WARNING <general>: could not find codec parameters for l.iso
2021-01-19 16:46:24.299 T:3920     INFO <general>: GetVideoAspectRatio - Aspect wanted: 3, Scale permissions: 2
2021-01-19 16:46:24.299 T:3920     INFO <general>: VideoPlayer: next stream, wait for old streams to be finished
2021-01-19 16:46:24.299 T:3920     INFO <general>: Creating Demuxer
2021-01-19 16:46:24.299 T:3920     INFO <general>: GetVideoAspectRatio - Aspect wanted: 3, Scale permissions: 2
2021-01-19 16:46:24.299 T:3920     INFO <general>: GetVideoAspectRatio - Aspect wanted: 3, Scale permissions: 2
2021-01-19 16:46:24.321 T:3920     INFO <general>: GetVideoAspectRatio - Aspect wanted: 3, Scale permissions: 2
2021-01-19 16:46:24.322 T:3920     INFO <general>: Opening stream: 1 source: 256
2021-01-19 16:46:24.322 T:3920     INFO <general>: Finding audio codec for: 86019
2021-01-19 16:46:24.322 T:3920     INFO <general>: Creating audio thread
2021-01-19 16:46:24.322 T:3932     INFO <general>: running thread: CVideoPlayerAudio::Process()
2021-01-19 16:46:24.322 T:3920     INFO <general>: Opening stream: 0 source: 256
2021-01-19 16:46:24.322 T:3920     INFO <general>: GetVideoAspectRatio - Aspect wanted: 3, Scale permissions: 2
2021-01-19 16:46:24.322 T:3920     INFO <general>: Creating video codec with codec id: 2
2021-01-19 16:46:24.322 T:3920     INFO <general>: CDVDVideoCodecFFmpeg::Open() Using codec: MPEG-2 video
2021-01-19 16:46:24.322 T:3920     INFO <general>: Creating video thread
2021-01-19 16:46:24.322 T:3933     INFO <general>: running thread: video_thread
2021-01-19 16:46:24.323 T:3933     INFO <general>: CVideoPlayerVideo - Stillframe left, switching to normal playback
2021-01-19 16:46:24.325 T:3933     INFO <general>: CDVDVideoCodecFFmpeg::Open() Using codec: MPEG-2 video
2021-01-19 16:46:24.332 T:3932     INFO <general>: CAEStreamParser::TrySyncAC3 - AC3 stream detected (6 channels, 48000Hz)
2021-01-19 16:46:24.332 T:3932     INFO <general>: Creating audio stream (codec id: 86019, channels: 6, sample rate: 48000, pass-through)
[... cut ...]
2021-01-19 17:08:26.905 T:3967     INFO <general>: Closing stream player 3
2021-01-19 17:08:26.908 T:1144     INFO <general>: VideoPlayer: finished waiting
2021-01-19 17:08:26.908 T:1144     INFO <general>: CVideoPlayer::CloseFile()
2021-01-19 17:08:26.908 T:1144     INFO <general>: VideoPlayer: waiting for threads to exit
2021-01-19 17:08:26.908 T:1144     INFO <general>: VideoPlayer: finished waiting
2021-01-19 17:08:26.919 T:4611     INFO <general>: Deleting settings information for files l.iso
```

heitbaum, 2021-01-19:
@afedchin / @howie-f - I have used your tagged releases in the LE10 dev tree (been updating all the packages in preparation for Matrix and LE10 release) but have identified a regression in libdvdnav between 6.0 and 6.1 that the audio streams are not available. Please let me know where to log the issue

fuzzard, 2021-01-19:
Matrix won't be using any of those bumped packages. They were done by @howie-f purely to attempt to handle another issue that was not resolved by the bumps anyways.

heitbaum, 2021-01-19:
Good to know. I’ll close the bug and draft PR off. Will put a mental note in place about the regression.

closed 2021-01-19 by heitbaum

howie-f, 2021-01-19:
will remove the forks shortly as they're leftovers

heitbaum, 2021-01-19:
CSS and READ have worked fine, but NAV is definitely :-(