LibreELEC/LibreELEC.tv #5038: librespot failed with newest test version of libreelec on RPI4

elschman, 2021-01-25:
logs are at http://ix.io/2NjV

If I choose librespot via spotify at my mobile to play a song it jumps over to the next song till the playlist end is reached without playing anz kind of song

elschman, 2021-01-25:
2021-01-25 23:15:04.268 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z INFO  librespot_connect::spirc] Fetching autoplay context uri
2021-01-25 23:15:04.302 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z INFO  librespot_connect::spirc] Autoplay uri resolved to <"spotify:station:album:7DcHWOXFDrfWLhAp02FMQp">
2021-01-25 23:15:04.374 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z INFO  librespot_playback::player] Loading <Rauchen> with Spotify URI <spotify:track:79yOuInT3K0iBU4pcNf26N>
2021-01-25 23:15:04.490 T:943      INFO <general>: Librespot: thread '<unnamed>' panicked at 'attempted to zero-initialize type `vorbisfile_sys::ov_callbacks`, which is invalid', /rustc/e1884a8e3c3e813aada8254edfa120e85bf5ffca/library/core/src/mem/mod.rs:623:9
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot: suspended sink 1
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot: event panic 1/3
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    0:   0x901934 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    1:   0x92a818 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    2:   0x8f995c - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    3:   0x9045cc - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    4:   0x9041d0 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    5:   0x904de0 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    6:   0x904908 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    7:   0x901ed4 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    8:   0x904888 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:    9:   0x9271f4 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:   10:   0x927150 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:   11:   0x5b87b4 - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:   12:   0x5c4bdc - <unknown>
2021-01-25 23:15:04.584 T:943      INFO <general>: Librespot:   13:   0x5c32fc - <unknown>
2021-01-25 23:15:04.585 T:943      INFO <general>: Librespot:   14:   0x90b82c - <unknown>
2021-01-25 23:15:04.585 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z WARN  librespot_playback::player] Unable to load <SpotifyId { id: 312455127656784245021068882369821925381, audio_type: Track }>
2021-01-25 23:15:04.585 T:943      INFO <general>: Librespot: Skipping to next track
2021-01-25 23:15:04.585 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z INFO  librespot_connect::spirc] Resolved 50 tracks from <"spotify:album:7DcHWOXFDrfWLhAp02FMQp">
2021-01-25 23:15:04.649 T:943      INFO <general>: Librespot: suspended sink 1
2021-01-25 23:15:04.650 T:943      INFO <general>: Librespot: librespot stopped
2021-01-25 23:15:04.692 T:943      INFO <general>: Librespot: librespot started
2021-01-25 23:15:04.699 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z INFO  librespot] librespot 819404a454 (2021-01-16). Built on 2021-01-16. Build ID: K7eOWhqX
2021-01-25 23:15:04.730 T:943      INFO <general>: Librespot: *** WARNING *** The program 'librespot' uses the Apple Bonjour compatibility layer of Avahi.
2021-01-25 23:15:04.730 T:943      INFO <general>: Librespot: *** WARNING *** Please fix your application to use the native API of Avahi!
2021-01-25 23:15:04.730 T:943      INFO <general>: Librespot: *** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
2021-01-25 23:15:04.803 T:943      INFO <general>: Librespot: [2021-01-25T22:15:04Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-e-6f5c.ap.spotify.com:4070"
2021-01-25 23:15:05.026 T:943      INFO <general>: Librespot: [2021-01-25T22:15:05Z INFO  librespot_core::session] Authenticated as "7wrank501ok1pu5oqtcghyfp2" !
2021-01-25 23:15:05.027 T:943      INFO <general>: Librespot: [2021-01-25T22:15:05Z INFO  librespot_core::session] Country: "DE"
2021-01-25 23:15:05.027 T:943      INFO <general>: Librespot: country=DE

heitbaum, 2021-02-14:
Could you please try with the latest 126 addon version and the current nightly release, there have been quite a few changes since the matrix release 2021-01-17 that you have been running.

elschman, 2021-02-15:
There is currently no RPI4 librespot addon version 126 available under https://test.libreelec.tv/addons/9.80.9/

CvH, 2021-02-15:
pls try again

elschman, 2021-02-15:
different Error but same result.. doesnt work

2021-02-15 23:38:02.367 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z INFO  librespot_connect::spirc] Fetching autoplay context uri
2021-02-15 23:38:02.401 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z WARN  librespot_connect::spirc] No autoplay_uri found
2021-02-15 23:38:02.401 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
2021-02-15 23:38:02.473 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z INFO  librespot_playback::player] Loading <Money for Nothing> with Spotify URI <spotify:track:4yqtwO7MQIIXqoiRBPHAgR>
2021-02-15 23:38:02.628 T:926      INFO <general>: Librespot: thread '<unnamed>' panicked at 'attempted to zero-initialize type `vorbisfile_sys::ov_callbacks`, which is invalid', /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/mem/mod.rs:623:9
2021-02-15 23:38:02.712 T:926      INFO <general>: Librespot: suspended sink 1
2021-02-15 23:38:02.712 T:926      INFO <general>: Librespot: event panic 2/3
2021-02-15 23:38:02.712 T:926      INFO <general>: Librespot:    0:   0x8db908 - <unknown>
2021-02-15 23:38:02.712 T:926      INFO <general>: Librespot:    1:   0x905eb0 - <unknown>
2021-02-15 23:38:02.712 T:926      INFO <general>: Librespot:    2:   0x8d34cc - <unknown>
2021-02-15 23:38:02.712 T:926      INFO <general>: Librespot:    3:   0x8de130 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:    4:   0x8ddb5c - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:    5:   0x8de944 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:    6:   0x8de46c - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:    7:   0x8dbe84 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:    8:   0x8de3ec - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:    9:   0x9028e0 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:   10:   0x90283c - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:   11:   0x5988b0 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:   12:   0x5a5de4 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:   13:   0x5a3304 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot:   14:   0x8e4f64 - <unknown>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z WARN  librespot_playback::player] Unable to load <SpotifyId { id: 198948220676562962479203615179475644189, audio_type: Track }>
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot: Skipping to next track
2021-02-15 23:38:02.713 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z INFO  librespot_playback::player] Loading <Another Pyramid> with Spotify URI <spotify:track:0DMSXhLnI1tat6UBtnZU7o>
2021-02-15 23:38:02.789 T:926      INFO <general>: Librespot: suspended sink 1
2021-02-15 23:38:02.790 T:926      INFO <general>: Librespot: librespot stopped
2021-02-15 23:38:02.826 T:926      INFO <general>: Librespot: librespot started
2021-02-15 23:38:02.833 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z INFO  librespot] librespot 96636fcf00 (2021-02-14). Built on 2021-02-14. Build ID: vZtigsgf
2021-02-15 23:38:02.863 T:926      INFO <general>: Librespot: *** WARNING *** The program 'librespot' uses the Apple Bonjour compatibility layer of Avahi.
2021-02-15 23:38:02.863 T:926      INFO <general>: Librespot: *** WARNING *** Please fix your application to use the native API of Avahi!
2021-02-15 23:38:02.863 T:926      INFO <general>: Librespot: *** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
2021-02-15 23:38:02.942 T:926      INFO <general>: Librespot: [2021-02-15T22:38:02Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-e-jnqn.ap.spotify.com:4070"
2021-02-15 23:38:03.166 T:926      INFO <general>: Librespot: [2021-02-15T22:38:03Z INFO  librespot_core::session] Authenticated as "7wrank501ok1pu5oqtcghyfp2" !
2021-02-15 23:38:03.167 T:926      INFO <general>: Librespot: [2021-02-15T22:38:03Z INFO  librespot_core::session] Country: "DE"
2021-02-15 23:38:03.167 T:926      INFO <general>: Librespot: country=DE
2021-02-15 23:38:37.134 T:735      INFO <general>: Loading skin file: SettingsCategory.xml, load type: KEEP_IN_MEMORY
2021-02-15 23:38:46.354 T:926      INFO <general>: Librespot: [2021-02-15T22:38:46Z INFO  librespot_connect::spirc] Fetching autoplay context uri
2021-02-15 23:38:46.391 T:926      INFO <general>: Librespot: [2021-02-15T22:38:46Z WARN  librespot_connect::spirc] No autoplay_uri found
2021-02-15 23:38:46.391 T:926      INFO <general>: Librespot: [2021-02-15T22:38:46Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
2021-02-15 23:38:46.391 T:926      INFO <general>: Librespot: [2021-02-15T22:38:46Z INFO  librespot_playback::player] Loading <Another Pyramid> with Spotify URI <spotify:track:0DMSXhLnI1tat6UBtnZU7o>
2021-02-15 23:38:46.524 T:926      INFO <general>: Librespot: thread '<unnamed>' panicked at 'attempted to zero-initialize type `vorbisfile_sys::ov_callbacks`, which is invalid', /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/mem/mod.rs:623:9
2021-02-15 23:38:46.619 T:926      INFO <general>: Librespot: suspended sink 1
[... cut ...]
2021-02-15 23:38:46.885 T:740      INFO <general>:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
2021-02-15 23:38:46.885 T:740      INFO <general>:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
2021-02-15 23:38:46.885 T:740      INFO <general>:         m_streamTypes     : No passthrough capabilities
2021-02-15 23:38:47.223 T:765      INFO <general>: Python interpreter stopped

elschman, 2021-02-15:
My audio device is the AVR connected via HDMI

heitbaum, 2021-02-21:
I’m have done a build of librespot 0.1.5 (127) for you to test. I don’t have a premium account - so can’t do any more than a validation test.

PROJECT=RPi ARCH=arm DEVICE=RPi4 s/create_addon librespot

test addon can be downloaded - https://github.com/heitbaum/LibreELEC.tv/blob/librespot-bin/service.librespot-9.80.9.127.zip

heitbaum, 2021-02-22:
Updated to 0.1.6 - https://github.com/heitbaum/LibreELEC.tv/blob/librespot-bin/service.librespot-9.80.9.-0.1.6-127.zip please test.

elschman, 2021-02-22:
what kind of logs should I prepare for you ?

 

Immo Wetzel
PGP Fingerprint (GMX) : D894 088D 3CE5 D769 7B89 518B E009 4219 129E 4642

 
 

Gesendet: Montag, 22. Februar 2021 um 10:32 Uhr

heitbaum, 2021-02-22:
Just a pastekodi please.

is the error still `attempted to zero-initialize type vorbisfile_sys::ov_callbacks, which is invalid `

As I this the following may have identified the vorbis error

https://github.com/RustAudio/lewton/issues/89
https://github.com/tomaka/vorbis-rs/issues/19

elschman, 2021-02-23:
http://ix.io/2QsK

 

should I create a premium account for you ? Just feel as a family member as long as I dont have all accounts running .

Immo Wetzel
PGP Fingerprint (GMX) : D894 088D 3CE5 D769 7B89 518B E009 4219 129E 4642

 
 

Gesendet: Montag, 22. Februar 2021 um 10:32 Uhr

heitbaum, 2021-02-23:
yup same bug. Let me read into rust first first about patching crates before I volunteer to try and patch this 👍

2021-02-23 07:45:34.364 T:4495     INFO <general>: Librespot: thread '<unnamed>' panicked at 'attempted to zero-initialize type `vorbisfile_sys::ov_callbacks`, which is invalid', /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/mem/mod.rs:623:9

NofanTasi, 2021-03-03:
same here on CoreELEC
Librespot: thread '<unnamed>' panicked at 'attempted to zero-initialize type `vorbisfile_sys::ov_callbacks`, which is invalid', /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/mem/mod.rs:623:9

heitbaum, 2021-03-25:
#5166 updated and New test build available at https://github.com/heitbaum/LibreELEC.tv/blob/librespot-bin2/service.librespot-9.80.11.127.zip

@elschman 
@NofanTasi 

PROJECT=RPi ARCH=arm DEVICE=RPi4 scripts/create_addon librespot

please test and let me know.

heitbaum, 2021-03-26:
X86 test addon.zip @
https://github.com/heitbaum/LibreELEC.tv/blob/librespot-bin2/service.librespot-9.80.11.127.x86.zip

elschman, 2021-03-26:
You're awesome...

 

For me it works like a charm.
 

Immo Wetzel
PGP Fingerprint (GMX) : D894 088D 3CE5 D769 7B89 518B E009 4219 129E 4642

 
 

Gesendet: Donnerstag, 25. März 2021 um 13:33 Uhr

closed 2021-03-27 by chewitt

jmbreuer, 2021-03-31:
> > X86 test addon.zip @
> > https://github.com/heitbaum/LibreELEC.tv/blob/librespot-bin2/service.librespot-9.80.11.127.x86.zip
> 
> Not available (anymore), any place to get a testing build for the time being?

... found this:

https://github.com/ldsz/plugin.audio.spotify/releases

Works for me, YMMV.

heitbaum, 2021-03-31:
> > > X86 test addon.zip @
> > > https://github.com/heitbaum/LibreELEC.tv/blob/librespot-bin2/service.librespot-9.80.11.127.x86.zip
> > 
> > 
> > Not available (anymore), any place to get a testing build for the time being?
> 
> ... found this:
> 
> https://github.com/ldsz/plugin.audio.spotify/releases
> 
> Works for me, YMMV.

@jmbreuer - I have republished the test version at - it will end up in the LE repo once we resolve a glibc build issue - https://github.com/heitbaum/LibreELEC.tv/blob/dev/service.librespot-9.80.11.127.zip

PR #5166 (2021-03-27): librespot: update to 0.1.6 and addon (127)
Update to try and resolve #5038 

- release: https://github.com/librespot-org/librespot/releases/tag/v0.1.5
- release: https://github.com/librespot-org/librespot/releases/tag/v0.1.6
- diff: https://github.com/librespot-org/librespot/compare/v0.1.3...v0.1.5
- diff: https://github.com/librespot-org/librespot/compare/v0.1.5...v0.1.6