LibreELEC/LibreELEC.tv #5048: [BUG] 661683fed (rpi4-pixelwrap) causes black screen with some videos

rschupp, 2021-01-27:
### Describe the bug
With 661683fed582bf420dd9431565b30ac0b8af7bbc on RPi4 some videos show a black screen (audio works).

### To Reproduce
View some videos.

Big Buck Bunny (Full HD) is good: ff-mpeg4-drm_prime (SW), yuv420p, 1920x1080 px, 1.78 AR
Lots of 1280x720 streams are affected: ff-h264_v4l2m2m-drm_prime (HW), 1280x720 px, 1.78 AR

Reverting 661683fed582bf420dd9431565b30ac0b8af7bbc makes the problem go away.

### Informations
 - LE Version: master
 - Hardware Platform: RPi4 B

### Log file
```
2021-01-27 12:08:35.997 T:712      INFO <general>: VideoPlayer::OpenFile: redacted.mp4
2021-01-27 12:08:36.003 T:917      INFO <general>: Creating InputStream
2021-01-27 12:08:36.040 T:917      INFO <general>: Creating Demuxer
2021-01-27 12:08:36.816 T:917      INFO <general>: Opening stream: 0 source: 256
2021-01-27 12:08:36.816 T:917      INFO <general>: Creating video codec with codec id: 27
2021-01-27 12:08:36.816 T:917      INFO <general>: CDVDVideoCodecDRMPRIME::Open - using decoder V4L2
 mem2mem H.264 decoder wrapper
2021-01-27 12:08:36.845 T:917      INFO <general>: Creating video thread
2021-01-27 12:08:36.845 T:918      INFO <general>: running thread: video_thread
2021-01-27 12:08:36.846 T:917      INFO <general>: Opening stream: 1 source: 256
2021-01-27 12:08:36.846 T:917      INFO <general>: Finding audio codec for: 86018
2021-01-27 12:08:36.847 T:917      INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened au
dio decoder aac
2021-01-27 12:08:36.847 T:917      INFO <general>: Creating audio thread
2021-01-27 12:08:36.848 T:919      INFO <general>: running thread: CVideoPlayerAudio::Process()
2021-01-27 12:08:36.848 T:917      INFO <general>: Opening stream: 2 source: 256
2021-01-27 12:08:36.849 T:918      INFO <general>: CVideoPlayerVideo - Stillframe left, switching to
 normal playback
2021-01-27 12:08:36.859 T:919      INFO <general>: Creating audio stream (codec id: 86018, channels:
 2, sample rate: 44100, no pass-through)
2021-01-27 12:08:36.867 T:726      INFO <general>: CActiveAESink::OpenSink - initialize sink
2021-01-27 12:08:36.959 T:712     ERROR <general>: CDRMAtomic::DrmAtomicCommit - test commit failed:
 (Invalid argument) - falling back to last successful atomic request
2021-01-27 12:08:36.959 T:712     ERROR <general>: 
                                                   Object: plane        ID: 89
                                                   Object: plane        ID: 68
                                                     Property: SRC_X    ID: 9   Value: 0
                                                     Property: SRC_Y    ID: 10  Value: 0
                                                     Property: SRC_W    ID: 11  Value: 83798699
                                                     Property: SRC_H    ID: 12  Value: 47185920
                                                     Property: CRTC_X   ID: 13  Value: 0
                                                     Property: CRTC_Y   ID: 14  Value: 0
                                                     Property: CRTC_W   ID: 15  Value: 1918
                                                     Property: CRTC_H   ID: 16  Value: 1080
                                                     Property: FB_ID    ID: 17  Value: 228
                                                     Property: CRTC_ID  ID: 20  Value: 74
                                                     Property: COLOR_ENCODING   ID: 72  Value: 1
                                                     Property: COLOR_RANGE      ID: 73  Value: 0
2021-01-27 12:08:37.008 T:712     ERROR <general>: CDRMAtomic::DrmAtomicCommit - test commit failed: (No space left on device) - falling back to last successful atomic request
2021-01-27 12:08:37.008 T:712     ERROR <general>: 
                                                   Object: plane        ID: 89
                                                     Property: FB_ID    ID: 17  Value: 224
                                                   Object: plane        ID: 68
                                                     Property: FB_ID    ID: 17  Value: 228
                                                     Property: CRTC_ID  ID: 20  Value: 74
2021-01-27 12:08:37.047 T:712     ERROR <general>: CDRMAtomic::DrmAtomicCommit - test commit failed: (No space left on device) - falling back to last successful atomic request
2021-01-27 12:08:37.047 T:712     ERROR <general>: 
                                                   Object: plane        ID: 89
                                                     Property: FB_ID    ID: 17  Value: 223
                                                   Object: plane        ID: 68
                                                     Property: FB_ID    ID: 17  Value: 228
                                                     Property: CRTC_ID  ID: 20  Value: 74
```
(repeats over and over)

HiassofT, 2021-01-27:
I could not reproduce the issue here with several 1280x720 files.

Please provide a link to a sample file that provokes the issue and post a link to a full debug log:

Enable debug logging in kodi, restart kodi, reproduce the issue, then run `pastekodi` on the console or use the logfile upload function in LE settings

rschupp, 2021-01-27:
### debug log 
I had to gzip kodi.log, otherwise the upload went into the big bucket): http://ix.io/2Nt2

### samples 
sometimes the screen is black, sometimes there's heavy flickering
- local copy of https://file-examples-com.github.io/uploads/2017/04/file_example_MP4_1280_10MG.mp4
- any stream from plugin.video.ardmediathek_de (e.g. "New" and then the top entry)

HiassofT, 2021-01-27:
PR #5051 should fix this issue

PR #5051 (2021-01-27): linux (RPi): add patch to fix zoom in video player
This also fixes playback issues of 720p videos if no 720p modes
are whitelisted or refresh switch is disabled after the pixel-wrap
workaround commit 01e90a2882.

Fixes: #5048
closed 2021-01-27 by CvH