LibreELEC/LibreELEC.tv #5082: [BUG] Bluetooth page in LE settings add-on crashes Kodi

KOPRajs, 2021-02-01:
### Describe the bug

Kodi crashes on the Bluetooth page in the LE settings add-on if the list of devices gets updated while the context menu of the device is already open and any action is executed from such context menu.

### To Reproduce
Steps to reproduce the behavior:
1. Go to LE settings add-on
2. Open Bluetooth page (start scanning)
3. Open context menu of a device (just click on any device)
4. Wait until scanning process in the background refreshes the device list
5. Execute any action from the opened device menu (e.g. Connect)
6. Kodi crashes

### Informations
 - LE Version: 9.2.6 (the bug is present from the early days of OpenELEC)
 - Hardware Platform: Arm (but I think all platforms are affected)

### Log file

Will add it later if needed, but I don't think so because it is easily reproducible crash.

### Additional context

I hate this bug for a long time but I've always forgotten to report it. It happens to me when I'm connecting my Bluetooth gamepad. Nowadays every TV set has a Bluetooth enabled by default which makes the list of devices long and changing all the time. It makes connecting of a device to LE a challenge. You need to be quick and lucky to not crash the Kodi.

I think the problem is that you are executing an action on the item which has disappeared (or has been re-created / re-indexed) in the meantime. Possible solution might be to stop the device list refresh when the context menu is opened (or even stop the Bluetooth scanning if needed).

KOPRajs, 2021-02-01:
Sorry, I currently can't test LE 10.

chewitt, 2021-02-01:
Okay. When you do have time please report back and we'll (re)open the ticket if necessary.

closed 2021-02-01 by chewitt

KOPRajs, 2021-05-16:
Ok, so I've finally got hands on a LE10 capable HW and tested it. Unfortunately the bug is still there.

Testing environment:
LE 10.0b2 running on RK3399

Steps to reproduce:
1. scan for BT device
2. open the context menu
3. wait for the list in background to update
4. execute any action from the opened menu
5. Kodi crashes

heitbaum, 2021-05-16:
Hi @KOPRajs - a fix was added to the nightlies yesterday - https://github.com/LibreELEC/LibreELEC.tv/pull/5374 could you please try that image. Need to make sure the image is at least this one (29f573f) 

- https://test.libreelec.tv/

KOPRajs, 2021-05-16:
Yesterday? Really? It has been there for several years and when I finally get time to properly report it I find out it was fixed yesterday :-D

https://github.com/LibreELEC/service.libreelec.settings/commit/d3d7a7822683129f6da3ace522817a596517204d indeed looks like the fix I was hoping for :-)

Please consider backporting this to 9.2 if there is going to be any maintenence release.