LibreELEC/LibreELEC.tv #5117: Unable to boot LibreELEC.

Tobiah10, 2021-02-07:
Hello, I have recently tried using LibreELEC on a Raspberry Pi4 with great success and had it running for a few days. The. I tried to update the EEPROM in the settings and rebooted the Pi. Now all I get is a black screen. No sounds, no image, nothing. I tried to re-flash the image using the LibreELEC usb-sd installer and the screen is still black. I then edited the config file and uncomented the HDMI force hot, to see if that was the issue, but nothing changed. I've tried earlier versions with no luck and even tried to make a USB image with no luck. The Pi still boots Raspian fine and I can change the EEprom there, which I think is the issue. I believe the EEPROM is the issue because I was able to boot LibreELEC on a different Pi that hadn't been booted in a few months, and was able to get through the set up and everything. I then tried to do a system update as well as the EEPROM update and the same issue started again. Black screen with no sounds. From then on I could not get any version to boot on that Pi either. I hope my input is useful. If any additional information is needed please let me know and I will do my best to get it.

Tobiah10, 2021-02-08:
Update, follow steps to update and upgrade the raspberry pi through Raspian along with the EEPROM worked to fix the issue. I had to go through steps to install the EEPROM again and got an SD card with just LebreELEC on it to boot up. I'm now trying to get a USB to boot it.

heitbaum, 2021-02-13:
@Tobiah10 - just checking how you went? Can we please close this issue?

Tobiah10, 2021-02-14:
I did some testing and found that the EEPROM update did prevent the Pi from booting properly sometimes. The only way I found to fix it was to use a Raspian OS on an SD card and re-install the EEPROM. This did fix the issue and I was able to boot into other OS's and get LibreELEC to successfully boot from a USB drive and a USB SSD drive using NOOBS to install it as the LibreELEC installer won't recognize the SSD drive as a USB device.

closed 2021-02-14 by lrusak