LibreELEC/LibreELEC.tv #5118: [Question]: Syntax for ipv6 address in config file

ds2k5, 2021-02-08:
Hi,
using libreelec on RPI3.

Wanna use Wireguard.

Config file: /storage/.config/wireguard/myvpn.config

[global]
Name = MYVPN
Description = MY VPN configuration

```
[provider_wireguard]
Type = WireGuard
Name = myvpn
Host = 1.2.3.4
WireGuard.Address = 192.168.100.2/32,fd00:0000:0000:0000:0000:0000:0000:0001/8
WireGuard.ListenPort = 51194
WireGuard.PrivateKey = 1111111111111111111111111111111111111111111111
WireGuard.PublicKey = 22222222222222222222222222222222222222222222222
WireGuard.DNS = 8.8.8.8, 1.1.1.1
WireGuard.AllowedIPs = 0.0.0.0/0, ::/0
WireGuard.EndpointPort = 51194
WireGuard.PersistentKeepalive = 25

```

LibreELEC:~ # connmanctl connect vpn_1_2_3_4
Error /net/connman/service/vpn_vpn_1_2_3_4: Input/output error

Without IPV6 address it work.

What do I wrong ?

chewitt, 2021-02-08:
ConnMan has incomplete support for IPv6 and as a result WireGuard won't work to an IPv6 server. It's nothing we can fix, and upstream is aware of the issue. The fix is non-trivial and not likely to happen quickly. Ping @igaw

ds2k5, 2021-02-08:
thanks a lot chewitt

chewitt, 2021-03-08:
Closing as this was a Question not a Bug report.

closed 2021-03-08 by chewitt