LibreELEC/LibreELEC.tv #5172: RC6 IR needs to press key twice

m-busch, 2021-02-21:
Setup:
I fresh installed LibreElec 9.2.6 on a RPi4. I activated WLAN via mywifi.config and SSH via cmdline.txt.
Then I tried to make my IR (a vuplus 4kuno) work. I have attached a DVBS2 receiver (DVBSky S960) and a IR Receiver (an MSL Digital Solutions Remote Pi Board). I followed the notes in the current LibreELEC.wiki.

Trouble:
I setup some KEYs for testing, but I need to press each Key twice to work.
In the Log (see below) I see every second time the event EV_KEY(0x01) is missing.
Any help would be very appreciated because I would love to build my own SettopBox, but this renders my project Useless. 

Logs:
**ir-keytable**
Found /sys/class/rc/rc1/ (/dev/input/event5) with:
        Name: DVBSky S960/S860
        Driver: dvb_usb_dvbsky, table: rc-dvbsky
        lirc device: /dev/lirc1
        Supported protocols: rc-5
        Enabled protocols: rc-5
        bus: 3, vendor/product: 0572:6831, version: 0x0000
        Repeat delay = 500 ms, repeat period = 125 ms
Found /sys/class/rc/rc0/ (/dev/input/event4) with:
        Name: gpio_ir_recv
        Driver: gpio_ir_recv, table: rc-rc6-mce
        lirc device: /dev/lirc0
        Supported protocols: lirc rc-5 rc-5-sz jvc sony nec sanyo mce_kbd rc-6 sharp xmp imon rc-mm
        Enabled protocols: lirc rc-6
        bus: 25, vendor/product: 0001:0001, version: 0x0100
        Repeat delay = 500 ms, repeat period = 125 ms

**ir-ctl -f**
Receive features /dev/lirc0:
 - Device can receive raw IR
 - Can report decoded scancodes and protocol
 - Can set receiving timeout min:1 microseconds max:1250000 microseconds
Send features /dev/lirc0:
 - Device cannot send

**ir-keytable -t**
11540.702238: lirc protocol(rc6_6a_32): scancode = 0x80529059
11540.702256: event type EV_MSC(0x04): scancode = 0x80529059
11540.702256: event type EV_SYN(0x00).

11543.268912: lirc protocol(rc6_6a_32): scancode = 0x80521059
11543.268941: event type EV_MSC(0x04): scancode = 0x80521059
11543.268941: event type EV_KEY(0x01) key_down: KEY_DOWN(0x006c)
11543.268941: event type EV_SYN(0x00).
11543.398879: event type EV_KEY(0x01) key_up: KEY_DOWN(0x006c)
11543.398879: event type EV_SYN(0x00).

11545.735582: lirc protocol(rc6_6a_32): scancode = 0x80529059
11545.735600: event type EV_MSC(0x04): scancode = 0x80529059
11545.735600: event type EV_SYN(0x00).

**ir-ctl -r**
pulse 2716
space 822
pulse 509
space 374
...(50 more lines)
pulse 478
space 405
pulse 482
timeout 10778

ir-ctl -r & ir-keytable -t &
# Testing events. Please, press CTRL-C to abort.
pulse 2724
space 816
pulse 535
...
pulse 509
space 379
pulse 504
11820.015554: lirc protocol(rc6_6a_32): scancode = 0x80529058
timeout 10509
11820.015591: event type EV_MSC(0x04): scancode = 0x80529058
11820.015591: event type EV_KEY(0x01) key_down: KEY_UP(0x0067)
11820.015591: event type EV_SYN(0x00).
11820.145517: event type EV_KEY(0x01) key_up: KEY_UP(0x0067)
11820.145517: event type EV_SYN(0x00).

pulse 949
space 379
pulse 507
space 821
....
pulse 508
space 374
pulse 509
space 374
pulse 508
11880.362195: lirc protocol(rc6_6a_32): scancode = 0x80521058
timeout 13269
11880.362222: event type EV_MSC(0x04): scancode = 0x80521058
11880.362222: event type EV_SYN(0x00).

Thanks for help ...
Mat

HiassofT, 2021-02-22:
Thanks for the report!

Could ypu please confirm which exact remote you are using? The vuplus website lists several IR remotes http://vuplus.com/sub/sub0112.php

Your remote sends alternating scancodes for each button (eg 0x80521059 and 0x80529059 for "DOWN"), like the Microsoft MCE remote does too, but the linux kernel doesn't handle that yet.

Can you please test with this build https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-9.2-devel-20210222113430-8c269ca.tar - you should only see one scancode (eg 0x80521059) with that.

Note that this test build doesn't include support for out-of-tree DVB drivers (they interfere with the IR drivers), but you can safely go back to LE 9.2.6 after testing if you need them.

HiassofT, 2021-03-19:
please report back with the test results and the information I asked for, otherwise the issue will be closed.

donbrew1, 2021-03-20:
Are you using both lirc and gpio-ir?  Should not need lirc if you are using ir-keytable.

m-busch, 2021-03-21:
@donbrew1: 
I am not aware i enabled lirc. If so, it would have been by default.
In the config.txt I just added: 
    dtoverlay=gpio-ir

@HiassofT:
Thank you for your support and sorry for late response (do time due to family).
I can not see any difference with our build. But I am not sure I used it correct.
Easiest for me would be to send you a parcel with the RPI-setup and the Remote Control :-)
Another way would be: can you post a working Remote. So everyone can by this ... one?

Thank You!

HiassofT, 2021-03-21:
@m-busch a simple workaround is to include both scancodes (i.e. the 0x8052**1**... and the 0x8052**9**... ones) in your keymap. This should work with official LE builds, too.

eg:
```
# table vuplus, type: RC6
0x80521058 KEY_UP
0x80529058 KEY_UP
0x80521059 KEY_DOWN
0x80529059 KEY_DOWN
...
```

Please give this a try and report back

m-busch, 2021-03-24:
Indeed a very smart solution! I coded all key twice. No new remote is working perfect. Thank you for your help. Even if it took some time from my side ... I appreciate it allot!!! 
Matthias

heitbaum, 2021-04-14:
Looks fixed. Will close. Please let me know if it is not, and to be reopened.

closed 2021-04-14 by heitbaum