LibreELEC/LibreELEC.tv #5215: [BUG] HDMI-CEC not working on Rock64

seangineer, 2021-03-05:
### Describe the bug
Input from TV remote over CEC does not work on mainline images. Confirmed working on vendor kernel.

### To Reproduce
Steps to reproduce the behavior:
1. Install latest rock64 nightly build
2. Attempt to navigate using a CEC input device

### Informations
 - LE Version: 10.0 nightly 20210305
 - Hardware Platform: rk3328-rock64 (Rockchip)

### Log file
http://ix.io/2RL4

### Additional context
Tested on multiple nightlies and has not worked. Also tried multiple HDMI cables. No problems on LE9/vendor 4.4 kernel.

seangineer, 2021-03-05:
Some more output from the console:

LibreELEC:~ # cec-ctl
Driver Info:
        Driver Name                : dwhdmi-rockchip
        Adapter Name               : dw_hdmi
        Capabilities               : 0x0000010e
                Logical Addresses
                Transmit
                Passthrough
                Connector Info
        Driver version             : 5.10.17
        Available Logical Addresses: 4
        DRM Connector Info         : card 0, connector 51
        Physical Address           : 1.0.0.0
        Logical Address Mask       : 0x8000
        CEC Version                : 1.4
        Vendor ID                  : 0x001582 (Pulse-Eight)
        OSD Name                   : ''
        Logical Addresses          : 1 (Allow Fallback to Unregistered)
        Logical Address          : 15 (Unregistered)
            Primary Device Type    : Record
            Logical Address Type   : Record

LibreELEC:~ # cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.7
no serial port given. trying autodetect:
 path:     /dev/cec0
 com port: Linux

opening a connection to the CEC adapter...
DEBUG:   [              64]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              64]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [              64]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [              64]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [              64]     could not open a connection (try 1)
DEBUG:   [              64]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            1064]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            1064]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [            1065]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [            1065]     could not open a connection (try 2)
DEBUG:   [            1065]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            2065]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            2065]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [            2065]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [            2065]     could not open a connection (try 3)
DEBUG:   [            2065]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            3065]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            3065]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16

knaerzche, 2021-03-05:
Can you give some details on TV / rock64 version? On my rock64 (V3.0) / Samsung TV (one very old, one newer) its working like a charm.

seangineer, 2021-03-05:
Thanks for the tip knaerzche, do you need me to re-run cec-ctl?
The board is a rock64 v2.0 from what I can tell on the PCB, and the TV is a Sharp LC-40CFG6352E.

knaerzche, 2021-03-05:
I've hinted this, because both `cec-client` and `cec-ctl` will try to gain full access over `/dev/cec0` which kodi already has if its running (so these commands will always fail when using them w/o arguments)
I've checked schematics: there are no differences between v3.0 and v2.1 in regards of CEC - could you please run
`systemctl stop kodi` and paste the results  `cec-client` somewhere?

seangineer, 2021-03-05:
Ah got it, understood. 

Output of cec-client w/ kodi stopped below:

LibreELEC:~ # cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.7
no serial port given. trying autodetect:
 path:     /dev/cec0
 com port: Linux

opening a connection to the CEC adapter...
DEBUG:   [              88]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              88]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
DEBUG:   [              88]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_G_PHYS_ADDR - addr=1000
DEBUG:   [              88]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [              88]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=8000 num_log_addrs=1
NOTICE:  [              89]     connection opened
DEBUG:   [              88]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=8000 phys_addr=1000
DEBUG:   [              89]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [              89]     << f0
DEBUG:   [              89]     processor thread started
DEBUG:   [             230]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=f0 opcode=ffffffff
TRAFFIC: [             230]     << f0
DEBUG:   [             230]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=f0 opcode=ffffffff
DEBUG:   [             230]     >> POLL not sent
DEBUG:   [             230]     TV (0): device status changed into 'not present'
DEBUG:   [             230]     registering new CEC client - v4.0.7
DEBUG:   [             231]     SetClientVersion - using client version '4.0.7'
NOTICE:  [             231]     setting HDMI port to 1 on device TV (0)
DEBUG:   [             231]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [             231]     << f0
DEBUG:   [             231]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=f0 opcode=ffffffff
TRAFFIC: [             231]     << f0
DEBUG:   [             231]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=f0 opcode=ffffffff
DEBUG:   [             231]     >> POLL not sent
DEBUG:   [             231]     SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [             231]     detecting logical address for type 'recording device'
DEBUG:   [             231]     trying logical address 'Recorder 1'
DEBUG:   [             231]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             231]     << 11
DEBUG:   [             232]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=11 opcode=ffffffff
TRAFFIC: [             232]     << 11
DEBUG:   [             232]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=11 opcode=ffffffff
DEBUG:   [             232]     >> POLL not sent
DEBUG:   [             232]     using logical address 'Recorder 1'
DEBUG:   [             232]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             232]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             232]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             232]     Recorder 1 (1): CEC version 1.4
DEBUG:   [             232]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             232]     CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [             232]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=1000
DEBUG:   [             235]     CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=8000 num_log_addrs=1
DEBUG:   [             235]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             235]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             235]     using auto-detected physical address 1000
DEBUG:   [             235]     Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             235]     << Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [             235]     << 1f:84:10:00:01
ERROR:   [             235]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [             235]     << 1f:84:10:00:01
ERROR:   [             235]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
NOTICE:  [             235]     CEC client registered: libCEC version = 4.0.7, client version = 4.0.7, firmware version = 0, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0,  compiled on 2021-03-05 07:04:35 by jenkins@prometheus on Linux 4.15.0-118-generic (x86_64), features: P8_USB, DRM, P8_detect, Linux
DEBUG:   [             235]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG:   [             235]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [             235]     << 10
DEBUG:   [             235]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=30 len=1 addr=10 opcode=ffffffff
TRAFFIC: [             235]     << 10
[... cut ...]
DEBUG:   [           37091]     unregistering all CEC clients
DEBUG:   [           37091]     CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [           37096]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=1000
DEBUG:   [           37101]     CLinuxCECAdapterCommunication::Process - stopped - m_fd=3
DEBUG:   [           37101]     CLinuxCECAdapterCommunication::Close - m_fd=3

seangineer, 2021-03-05:
Maybe something interesting: if I turn the TV off and then on while cec-client is running, my console gets spammed like so until I kill the command:

DEBUG:   [          283459]     CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=5 addr=0f opcode=84
TRAFFIC: [          283459]     >> 0f:84:00:00:00
DEBUG:   [          283459]     << Recorder 1 (1) -> broadcast (F): physical address 1000
DEBUG:   [          283459]     >> TV (0) -> Broadcast (F): report physical address (84)
TRAFFIC: [          283459]     << 1f:84:10:00:01
ERROR:   [          283459]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          283459]     << 1f:84:10:00:01
ERROR:   [          283459]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
DEBUG:   [          283650]     CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=6 addr=0f opcode=80
TRAFFIC: [          283650]     >> 0f:80:00:00:10:00
DEBUG:   [          283650]     TV (0): power status changed from 'unknown' to 'on'
DEBUG:   [          283650]     making Recorder 1 (1) the active source
NOTICE:  [          283651]     >> source activated: Recorder 1 (1)
DEBUG:   [          283651]     >> TV (0) -> Broadcast (F): routing change (80)
DEBUG:   [          283651]     sending active source message for 'Recorder 1'
DEBUG:   [          283651]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [          283651]     << 10:8c
ERROR:   [          283651]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          283651]     << 10:8c
ERROR:   [          283651]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
NOTICE:  [          283651]     << powering on 'TV' (0)
TRAFFIC: [          283651]     << 10:04
ERROR:   [          283651]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          283651]     << 10:04
ERROR:   [          283651]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
NOTICE:  [          283651]     << Recorder 1 (1) -> broadcast (F): active source (1000)
TRAFFIC: [          283651]     << 1f:82:10:00
ERROR:   [          283651]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          283651]     << 1f:82:10:00
ERROR:   [          283651]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
DEBUG:   [          283651]     failed to make 'Recorder 1' the active source. will retry later
DEBUG:   [          283789]     CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=4 addr=0f opcode=86
TRAFFIC: [          283789]     >> 0f:86:10:00
DEBUG:   [          283789]     >> TV (0) sets stream path to physical address 1000
DEBUG:   [          283789]     Recorder 1 (1) was already marked as active source
NOTICE:  [          283789]     >> source activated: Recorder 1 (1)
NOTICE:  [          283789]     << Recorder 1 (1) -> broadcast (F): active source (1000)
DEBUG:   [          283789]     >> TV (0) -> Broadcast (F): set stream path (86)
TRAFFIC: [          283789]     << 1f:82:10:00
ERROR:   [          283789]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          283789]     << 1f:82:10:00
ERROR:   [          283789]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
DEBUG:   [          284790]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [          284790]     << 10:8c
ERROR:   [          284790]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          284790]     << 10:8c
ERROR:   [          284790]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
NOTICE:  [          284790]     << powering on 'TV' (0)
TRAFFIC: [          284790]     << 10:04
ERROR:   [          284790]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
TRAFFIC: [          284790]     << 10:04
ERROR:   [          284790]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
NOTICE:  [          284790]     << Recorder 1 (1) -> broadcast (F): active source (1000)
TRAFFIC: [          284790]     << 1f:82:10:00

knaerzche, 2021-03-05:
If you seeing any traffic there, its likely not an HW issue.
Out of curiosity: Do you have any other devices connected to this TV? If so: please detach and try again with latest image: https://test.libreelec.tv/LibreELEC-RK3328.arm-10.0-nightly-20210305-d86c890-rock64.img.gz

seangineer, 2021-03-05:
I didn't have any other HDMI connections to the TV, but disconnected everything for good measure, still same result. Same with removing peripherals from the board.

The image you listed is the one that was flashed, I just flashed it today.

I would also guess non-HW issue given that it's working properly on the 4.4 vendor kernel (Armbian, LE 9, etc).

knaerzche, 2021-03-08:
From tomorrow on nightly images will contain a workaround which addresses some CEC-issues: so please retest.

Looking at your `cec-client` log again, I doubt it will help, since tx attempts are failing, while rx seems to be OK and this workaround is targeting failed rx attempts.
I guess that your TV's CEC implementation is not compatible with mainline kernel cec-core or libCEC, since it is working great with my TVs and rock64 and you also can see actual cec traffic happing. Please report this issue over there, since there is not much we can do within this project and it will help more people if it gets resolved there.

Feel free to report here if it helped for you or not but also understand that I have to close this issue now, since it is not related to rock64 hardware.

closed 2021-03-08 by knaerzche

SupervisedThinking, 2021-03-08:
@knaerzche whatever your patch did... it also changed the behaviour of my ROCKPro64 - when I've rebooted my Vim3 my AVR switched the input to it but never did the ROCKPro64. I've installed the latest version of my build with kernel 5.10.21 & your fixes... et voila now the ROCKPro64 grabs the AVR input after booting up.

So I guess there was definitely something wrong bevor.

seangineer, 2021-03-09:
Console output for reference:
```
LibreELEC:~ # cec-ctl -S
Driver Info:
        Driver Name                : dwhdmi-rockchip
        Adapter Name               : dw_hdmi
        Capabilities               : 0x0000010e
                Logical Addresses
                Transmit
                Passthrough
                Connector Info
        Driver version             : 5.10.21
        Available Logical Addresses: 4
        DRM Connector Info         : card 0, connector 51
        Physical Address           : 1.0.0.0
        Logical Address Mask       : 0x0002
        CEC Version                : 1.4
        Vendor ID                  : 0x001582 (Pulse-Eight)
        OSD Name                   : ''
        Logical Addresses          : 1 (Allow Fallback to Unregistered)

          Logical Address          : 1 (Recording Device 1)
            Primary Device Type    : Record
            Logical Address Type   : Record

        System Information for device 0 (TV) from device 1 (Recording Device 1):
                CEC Version                : 1.4
                Physical Address           : 0.0.0.0
                Primary Device Type        : TV
                Vendor ID                  : Tx, OK, Rx, OK, Feature Abort
                OSD Name                   : 'TV'
                Menu Language              : eng
                Power Status               : On

        Topology:

        0.0.0.0: TV
            1.0.0.0: Recording Device 1

LibreELEC:~ # systemctl stop kodi
LibreELEC:~ # cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.7
no serial port given. trying autodetect:
 path:     /dev/cec0
 com port: Linux

opening a connection to the CEC adapter...
DEBUG:   [              92]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              92]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
DEBUG:   [              92]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_G_PHYS_ADDR - addr=1000
DEBUG:   [              93]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [              93]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=8000 num_log_addrs=1
NOTICE:  [              94]     connection opened
DEBUG:   [              94]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=8000 phys_addr=1000
DEBUG:   [              94]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [              95]     << f0
DEBUG:   [              94]     processor thread started
DEBUG:   [             406]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=1 addr=f0 opcode=ffffffff
DEBUG:   [             406]     >> POLL sent
DEBUG:   [             406]     TV (0): device status changed into 'present'
DEBUG:   [             406]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [             406]     << f0:8c
DEBUG:   [             475]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=2 addr=f0 opcode=8c
DEBUG:   [            1475]     expected response not received (87: device vendor id)
TRAFFIC: [            1475]     << f0:8c
DEBUG:   [            1529]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=2 addr=f0 opcode=8c
DEBUG:   [            2529]     expected response not received (87: device vendor id)
DEBUG:   [            2529]     registering new CEC client - v4.0.7
DEBUG:   [            2529]     SetClientVersion - using client version '4.0.7'
[... 84 lines cut ...]
DEBUG:   [           11004]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
DEBUG:   [           11004]     CheckKeypressTimeout T:2174.698
DEBUG:   [           11004]     Key up: idle (duration:0) (1) timeout:16980072ms (rel:500,rep:0,prs:500,rel:0)
DEBUG:   [           11324]     CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=2 addr=01 opcode=45
TRAFFIC: [           11324]     >> 01:45
DEBUG:   [           11324]     key released: up (1) D:320ms
DEBUG:   [           11325]     >> TV (0) -> Recorder 1 (1): user control release (45)
^Csignal caught: 2 - exiting
DEBUG:   [           14637]     unregistering all CEC clients
NOTICE:  [           14637]     unregistering client: libCEC version = 4.0.7, client version = 4.0.7, firmware version = 0, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0,  compiled on 2021-03-09 05:52:59 by jenkins@tsuboi on Linux 4.19.0-8-amd64 (x86_64), features: P8_USB, DRM, P8_detect, Linux
DEBUG:   [           14637]     Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [           14637]     Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [           14637]     Recorder 1 (1): CEC version unknown
DEBUG:   [           14637]     Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [           14637]     marking Recorder 1 (1) as inactive source
DEBUG:   [           14637]     Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [           14637]     CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [           14637]     CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
[... cut ...]
DEBUG:   [           14637]     unregistering all CEC clients
DEBUG:   [           14637]     CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [           14642]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=1000
DEBUG:   [           15649]     CLinuxCECAdapterCommunication::Process - stopped - m_fd=3
DEBUG:   [           15649]     CLinuxCECAdapterCommunication::Close - m_fd=3
```