LibreELEC/LibreELEC.tv #5223: [BUG] latest version of gnutls rejects some server certificate chains

rschupp, 2021-03-06:
### Describe the bug
I've seen several failures to play media (e.g. podcasts) in the last couple of days.
Log contains messages like
```
2021-03-05 14:52:11.643 T:3025    ERROR <general>: CCurlFile::FillBuffer - Failed: SSL peer certificate or SSH remote key was not OK(60)
2021-03-05 14:52:11.643 T:3025    ERROR <general>: CCurlFile::Open failed with code 0 for https://dcs.megaphone.fm/SONY7085161311.mp3?key=b1a3f2493ab928d6e23a63532bc02a8b:
...
2021-03-05 13:30:21.027 T:2303    ERROR <general>: CCurlFile::FillBuffer - Failed: SSL peer certificate or SSH remote key was not OK(60)
2021-03-05 13:30:21.028 T:2303    ERROR <general>: CCurlFile::Open failed with code 0 for https://nyt.simplecastaudio.com/9257fd86-22fe-4866-89a9-12c73a57a552/episodes/f4ae8000-62fe-40a2-a24d-439a8360a7b6/audio/128/default.mp3?aid=rss_feed&awCollectionId=9257fd86-22fe-4866-89a9-12c73a57a552&awEpisodeId=f4ae8000-62fe-40a2-a24d-439a8360a7b6&feed=W1rB_kgL:

```

### To Reproduce
Steps to reproduce the behavior:

Running `curl` (using curl **on the target**) for the above URLs shows that it fails to validate the server certificates.
However, pasting the URLs into Firefox or Chrome successfully plays or downloads the files.
Checking the servers with `openssl s_client` (using openssl **on the target**) also successfully verifies the server certificates.
On closer inspection of the output of `s_client` shows that the certificate chains are not 100 percent up to spec, some contain duplicate certificates (e.g. instead of a chain "server cert, intermediate cert, CA cert" the server sends the server cert twice "server cert, server cert, intermediate cert, CA cert").

But LibreELEC builds `curl` and `libcurl`, hence Kodi, against GnuTLS. Unfortunately `gnutls-cli` - the equivalent of `openssl s_client` for GnuTLS - isn't installed, but I managed to build it for the target, and running it for the above examples
```
gnutls-cli --starttls-proto=https --x509cafile=/run/libreelec/cacert.pem dcs.megaphone.fm
gnutls-cli --starttls-proto=https --x509cafile=/run/libreelec/cacert.pem nyt.simplecastaudio.com
```
indeed fails to verify the server certificates. 

I think this is regression in GnuTLS 3.6.x → 3.7.0 (introduced by f1250a8a89097ddb678a5e510132dc9f54a95d21).
It's reported upstream as https://gitlab.com/gnutls/gnutls/-/issues/1131 and a fix has been commited, but not yet released.

### Informations
 - LE Version: master/HEAD
 - Hardware Platform:RPi4

SupervisedThinking, 2021-03-06:
https://github.com/LibreELEC/LibreELEC.tv/pull/5222

rschupp, 2021-03-06:
Rats, didn't check in the last couple hours :smile:
That totally obsoletes my #5224, but #5222 is the better solution long term.

closed 2021-03-08 by chewitt