LibreELEC/LibreELEC.tv #5258: BLANK screen in setup

donbrew1, 2021-03-19:
This is a long standing problem.  When installing a fresh LE on a RPI4 with an old TV (without CEC) the screen goes blank after the LE splash (sometimes after the RPI splash) and seems to crash.  I think it has been fixed in Raspbian/Raspberry Pi OS.

Seems to be either CEC or overscan is the problem.  My TV has no CEC and I must do the video calibration in Kodi to adjust the screen because the TV's controls don't fix it.

Seems to be a big problem that nobody will admit.  There is a hard to find chapter in the Kodi wiki about how to troubleshoot blank screen that basically presents this solution.  Sorry, I came across it on accident and don't recall the exact particulars.

The solution with my TV, a 2006 Polaroid 42 inch is this config.txt.  It gets you a picture, at least, then you can tweak.

```
gpu_mem=320
force_turbo=0
hdmi_ignore_edid=0xa5000080
disable_overscan=1
framebuffer_width=1920
framebuffer_height=1080
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
hdmi_ignore_cec=1 
dtparam=audio=on
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
```

HiassofT, 2021-03-19:
Please fill out the bug report issue template when creating bug reports. You neither did tell us which LE version you are using nor attached any logs.

From your config.txt snippet it looks like you might be using LE 9.2 which is now EOL. Test with a plain install of LE 10 and note that none of the config.txt options you added will be relevant to LE10. If a plain LE 10 installation doesn't work please post on the forums.

closed 2021-03-19 by HiassofT

donbrew1, 2021-03-20:
All versions of LE for RPI4., INCLUDING LE 10.

All I get from the forum is the usual "adjust the TV overscan".