LibreELEC/LibreELEC.tv #5261: [BUG] Several issues with BLUETOOTH on LibreELEC (Matrix) 10.0 BETA1

euquiq, 2021-03-20:
I'm using an OLD INTEL NUC (running Celeron N2820, with 2gb of RAM and 60gb SSD drive, hdmi out at 1920 x 1080, IR port, usb 3.0 and 2.0).

When entering the LibreELEC settings, on the bluetooth panel, I have the following issues:

1) The discovered BT clients list reloads, very fast, several times.
2) The list sometimes includes the same device more than once
3) The list sometimes includes empty rows (with no device name or description)
4) When trying to pair a device, sometimes LibreELEC "reboots" into KODI
5) It takes several tries to finally get a device paired and functional. When it does, "it works", but in my case, after several minutes (half an hour or more) the whole BT subsystem seems to "shutdown" (all my BT devices stop working) and I have to resource into using an IR remote.

I am testing this with the following two BT devices: 

An old BT "amazon firetv" remote control
A Microsoft Sculpt Keyboard

Thank you very much for your work. I hope I can help with this !

heitbaum, 2021-03-20:
Please use the forum for support. Thanks!

Though in saying that - there is a BT fix being included in LE10b2 - https://github.com/LibreELEC/LibreELEC.tv/pull/5247 could you please try that.

you will need to share your logs, dmesg and in the case of a crash, the crash log.

euquiq, 2021-03-20:
> Please use the forum for support. Thanks!
> 

I was just following the "Please report bugs via GitHub Issues." on the code tab. I will try latest nightly, thanks.

closed 2021-03-20 by euquiq

RalphMW, 2021-08-31:
Just for completeness pairing a bt keyboard doesnt _show_ the code you have to type to pair.