LibreELEC/LibreELEC.tv #5262: [BUG]

Karl-opec, 2021-03-20:
### Describe the bug
I'm running Chewitt's libreelec Branch amlogic on lepotato device.
with the vnsi pvr addon.
My issue that it shows for SD TV channels only a green screen, audio is working. all other
TV resolutions in HD either 720p or 1024i are working well with hardware accel.

software decoding shows SD channels well.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE 10.0 -git chewitt amlogic branch
 - Hardware Platform: lepotato, odroid-c2, odroid-n2

### Log file
v4l2-ctl --list-formats-out shows:
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'VP90' (VP9, compressed, dyn-resolution)
        [1]: 'H264' (H.264, compressed, dyn-resolution)
        [2]: 'MPG1' (MPEG-1 ES, compressed)
        [3]: 'MPG2' (MPEG-2 ES, compressed)

### Additional context

```
2021-03-20 16:33:05.287 T:877      INFO <general>: VideoPlayer::OpenFile: pvr://channels/tv/Alle%20Kan%c3%a4le/pvr.vdr.vnsi_1763192291.pvr
2021-03-20 16:33:05.287 T:1719     INFO <general>: Creating InputStream
2021-03-20 16:33:05.291 T:1719     INFO <general>: AddOnLog: pvr.vdr.vnsi: Logged in at '1616254385+3600' to 'VDR-Network-Streaming-Interface (VNSI) Server' Version: '1.8.0' with protocol version '13'
2021-03-20 16:33:05.533 T:1719     INFO <general>: Skipped 1 duplicate messages..
2021-03-20 16:33:05.533 T:1719     INFO <general>: Creating Demuxer
2021-03-20 16:33:06.525 T:1719     INFO <general>: Opening stream: 168 source: 256
2021-03-20 16:33:06.525 T:1719     INFO <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 576, fps: 25.000, 3D: false
2021-03-20 16:33:06.526 T:1719     INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (21) (weight: 0.000)
2021-03-20 16:33:06.563 T:1719  WARNING <general>: OpenStream - Unsupported stream 168. Stream disabled.
2021-03-20 16:33:06.563 T:1719     INFO <general>: Opening stream: 137 source: 256
2021-03-20 16:33:06.563 T:1719     INFO <general>: Finding audio codec for: 86016
2021-03-20 16:33:06.563 T:1719     INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder mp2
2021-03-20 16:33:06.563 T:1719     INFO <general>: Creating audio thread
2021-03-20 16:33:06.564 T:1720     INFO <general>: running thread: CVideoPlayerAudio::Process()
2021-03-20 16:33:06.564 T:1719     INFO <general>: Opening stream: 70 source: 256
2021-03-20 16:33:06.564 T:1719     INFO <general>: waiting for teletext data thread to exit
2021-03-20 16:33:06.566 T:1719     INFO <general>: Creating teletext data thread
2021-03-20 16:33:06.566 T:1721     INFO <general>: running thread: CDVDTeletextData
2021-03-20 16:33:07.103 T:1719     INFO <general>: Opening stream: 168 source: 256
2021-03-20 16:33:07.103 T:1719     INFO <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 576, fps: 25.000, 3D: false
2021-03-20 16:33:07.104 T:1719     INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (21) (weight: 0.000)
2021-03-20 16:33:07.107 T:1719     INFO <general>: Creating video codec with codec id: 2
2021-03-20 16:33:07.107 T:1719     INFO <general>: CDVDVideoCodecDRMPRIME::Open - using decoder V4L2 mem2mem MPEG2 decoder wrapper
2021-03-20 16:33:07.118 T:1719     INFO <general>: Creating video thread
2021-03-20 16:33:07.118 T:1722     INFO <general>: running thread: video_thread
2021-03-20 16:33:07.119 T:1719     INFO <general>: Opening stream: 137 source: 256
2021-03-20 16:33:07.119 T:1719     INFO <general>: Opening stream: 70 source: 256
2021-03-20 16:33:07.130 T:1720     INFO <general>: Creating audio stream (codec id: 86016, channels: 2, sample rate: 48000, no pass-through)
2021-03-20 16:33:07.139 T:888      INFO <general>: CActiveAESink::OpenSink - initialize sink
2021-03-20 16:33:07.340 T:888      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "@:CARD=LIBRETECHCC,DEV=0"
2021-03-20 16:33:07.349 T:888      INFO <general>: CAESinkALSA::Initialize - Opened device "sysdefault:CARD=LIBRETECHCC"
2021-03-20 16:33:07.350 T:888      INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2021-03-20 16:33:07.350 T:888      INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE4
2021-03-20 16:33:07.476 T:877      INFO <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 576, fps: 25.000, 3D: false
2021-03-20 16:33:07.477 T:877      INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (21) (weight: -1683950080.000)

```

How can a working hardware accel with MPEG2 decoding be used?

SupervisedThinking, 2021-03-20:
Well the Amlogic video decoders are still WIP.

https://linux-meson.com/doku.php#upstreaming_status

There's a reason why there are currently no official LibreELEC 10.0 Amlogic releases.

Karl-opec, 2021-03-21:
I Know that.
But the actual amlogic branch in chewitt's repo is well working. if MPEG2 hardware decoding would be usable it close to great.
as i understand MPEG2 hardware decoding is there and needs to be enabled. 

v4l2-ctl --list-formats-out shows:
ioctl: VIDIOC_ENUM_FMT
Type: Video Output Multiplanar

    [0]: 'VP90' (VP9, compressed, dyn-resolution)
    [1]: 'H264' (H.264, compressed, dyn-resolution)
    [2]: 'MPG1' (MPEG-1 ES, compressed)
    [3]: 'MPG2' (MPEG-2 ES, compressed)

MPEG-2 decoding looks like being there.

I would like to offer  testing there, i have a wide range of amlogic plattforms ODROID-C2,C4,N2(+) and lepotato and further
TV boxes. M

hardware MPEG2 decoding would greatly expand the usability .

chewitt, 2021-03-21:
We don't need testers or testing. We need developers capable of architecting and coding modern kernel drivers based on the emerging V4L2 APIs. There are not many who qualify, and since Amlogic has fostered zero "community" around their chips (for a long list of reasons) those who do meet the standard are professional developers who rightfully expect to be paid for their efforts. It will take $100k-$150k to complete the current VDEC work; which is not a huge sum, but also not the kind of pocket-change that people have laying around to spend on something with no clear business case behind it - that's the key to making something happen. I don't write code, so I'm powerless to personally "fix" the issues here, and while there are others in the wider LE team who have some of the skills needed, so much abuse has been historically directed towards LE team members taking an interest in Amlogic SoCs that people prefer to spend their time on other SoC targets where their efforts are praised not persecuted.

Karl-opec, 2021-03-21:
Thank you for pointing that out. I would like to support but unfortunately I 'm not able to code.
Yes its a pity