LibreELEC/LibreELEC.tv #5285: [BUG]Problems encountered during compilation

matthuo333, 2021-04-01:
### Describe the bug
The Clone 9.2.6 project was compiled locally from the Ubuntu 20.04 environment, and the result appears in the following image：

[image]

### To Reproduce
Steps to reproduce the behavior:
1. git clone -b 9.2.6  ....
2. cd ./Libreelec
3. make image PROJECT=RPi DEVICE=RPi4 ARCH=arm

### Informations
 - LE Version: 9.2.6
 - Hardware Platform: RPi4 B

### Log file
Not found log file.  If you know which path, please let me know

### Additional context
No

closed 2021-04-01 by lrusak

SupervisedThinking, 2021-04-01:
You're simply running out of RAM, happened to me even in systems with 6GB so close stuff like your browser, increase swap and try again.

matthuo333, 2021-04-01:
> You're simply running out of RAM, happened to me even on systems with 6GB so close stuff like your browser, increase swap and try again.

My Ubuntu is running on a VM virtual machine. Do I need to set up appropriate memory and other resources for the VM virtual machine? What should be the main ones? How much memory should I have? How much should SWAP be?

SupervisedThinking, 2021-04-01:
Well how much RAM did you actually reserve for your VM? 6GB is fine on my dedicated system once you don't run a browser, slack or else. 4GB is fine on a headless system. When you installed the VM you had to define the size of the swap partition, I guess 4-8GB should be fine too & a safe value.