LibreELEC/LibreELEC.tv #5286: [BUG]Download links LibreElec not working

NickFatbob, 2021-04-02:
### Describe the bug
I am not able to download the “LibreELEC USB-SD Creator” app from https://libreelec.tv/downloads_new/
None of the links are working. Am I doing something wrong?  
I am a newbie.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE Version: [e.g. 9.2.1]
 - Hardware Platform: [e.g. RPi3]

### Log file

### Additional context

heitbaum, 2021-04-02:
here are the links from the https://libreelec.tv/downloads_new/.

http://releases.libreelec.tv/LibreELEC.USB-SD.Creator.Win32.exe
http://releases.libreelec.tv/LibreELEC.USB-SD.Creator.Linux-32bit.bin
http://releases.libreelec.tv/LibreELEC.USB-SD.Creator.Linux-64bit.bin
http://releases.libreelec.tv/LibreELEC.USB-SD.Creator.macOS.dmg

Please use the forums for support.

closed 2021-04-02 by heitbaum

CvH, 2021-04-02:
Main problem is chrome/chromium that are now blocking downloads with a lot known extensions like .zip, .exe etc ... .
You need "right click" -> "save link as" to workaround.

NickFatbob, 2021-04-02:
Thanks for your reply, but unfortunately those links are not working (at my pc)

NickFatbob, 2021-04-02:
CvH: thank you. Problem solved. Google is going crazy making decisions like that.