LibreELEC/LibreELEC.tv #5288: [BUG] - failed to find dependency vfs.libarchive

jhonny-oliveira, 2021-04-02:
### Describe the bug
I get the following error while attempting to install "script.module.archive_tool-2.0.2.zip" (https://github.com/zach-morris/repository.zachmorris/tree/master/script.module.archive_tool):

ERROR <general>: CAddonInstallJob[script.module.archive_tool]: failed to find dependency vfs.libarchive

### To Reproduce
Steps to reproduce the behavior:
1. Go to Add-Ons
2. 'Install plugin from zip file': script.module.archive_tool-2.0.2.zip

### Informations
 - LE Version: 9.995.1
 - Hardware Platform: RPi4.arm

### Log file
2021-04-02 14:44:31.962 T:983      INFO <general>: CActiveAESink::OpenSink - initialize sink
2021-04-02 14:44:31.962 T:983      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "default"
2021-04-02 14:44:31.968 T:983      INFO <general>: CAESinkALSA::Initialize - Opened device "default"
2021-04-02 14:44:31.970 T:983      INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2021-04-02 14:44:31.971 T:983      INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
2021-04-02 14:44:32.454 T:4698    ERROR <general>: CAddonInstallJob[script.module.archive_tool]: failed to find dependency vfs.libarchive

mglae, 2021-04-04:
What if you install vfs.libarchive first from LE repository?

jhonny-oliveira, 2021-04-05:
Thank you for the advice! It solved the issue.

chewitt, 2021-04-05:
@zach-morris should add the dependency to his add-on, then it will be installed automagically.

closed 2021-04-05 by chewitt

zach-morris, 2021-04-05:
I'm confused.  Why doesn't this install from the official repo if it's a dependancy?