LibreELEC/LibreELEC.tv #5298: [BUG]

camelreef, 2021-04-07:
### Describe the bug
I'm on LE10 nightly, that I update daily, running off a Raspberry Pi 4 4 GB.
I am trying to play an H.264+AAC in an MP4 container, and it fails, showing the rotating wheel, no message)

I'm adding pastekodi with debug a few lines below.

If I reencode the audio track only, same refusal to play. If I reencode the video track, it plays. So the problem is with the video.

Any other H.264 videos I have play without any issue.

1 min sample file here: https://youplala.net/~nico/le10-bad-video/

### To Reproduce
Steps to reproduce the behavior:
Play the sample file

### Informations
 - LE Version:  10 nightly (still not playing on 20210407)
 - Hardware Platform: RPi 4 4 GB

### Log file
http://ix.io/2UJp

### Additional context
The same video plays without issue on my MacBook Pro using Kodi 19.

Here is a debug log from the Mac, if it helps: https://pastebin.ubuntu.com/p/jKHBGHYzkY/

There is a thread on the LE forums: https://forum.libreelec.tv/thread/23769-h-264-video-does-not-play/

HiassofT, 2021-04-07:
Issue confirmed, the video plays fine on RPi4 LE 9.2.6 with MMAL decoding but not on LE10 with drmprime/V4L2 decoding.

With HW decoding disabled on LE10 the file plays, too.

camelreef, 2021-04-15:
Hello!
I keep updating using the nightly builds, everyday checking my little test video.
I know that LE10 has 10 million points being worked on at the moment, with higher priorities than this. How should I set my expectations regarding a timeline towards a fix for this particular bug?
I wish I could help more, but my skill set is not adequate!

HiassofT, 2021-04-15:
RPi devs are quite busy working on this.

I spare you the tons of messages with gory V4L2 and ffmpeg details that were exchanged the last days :-)

Short summary is that it's clear what's going wrong, but fixing it is quite tricky and could take a bit.

camelreef, 2021-06-16:
Hello devs!

Any progress on this? Is there anything I could offer to help?

On a side note, is there a specific encoding parameter that is triggering the issue and should be avoided?

Thanks!

HiassofT, 2021-06-30:
Can you test the latest nightly (currently 20210630)? We merged a bunch of ffmpeg and kernel fixes and your sample played fine here with these

camelreef, 2021-06-30:
I can confirm that my sample also plays fine at my place using nightly-20210630-d4c4658 (RPi4.arm). YAY!

Another full file plays OK. I haven't played more than a few seconds for testing, but I have great confidence!

A very big thank you to all involved in fixing this!

closed 2021-06-30 by camelreef