LibreELEC/LibreELEC.tv #5303: [BUG] LE settings causes crashes when switching profiles

HiassofT, 2021-04-08:
### Describe the bug

When switching profiles, eg from master user to a separate profile, kodi crashes if LE settings addon is included in system image.

The issue was originally reported on the forum https://forum.libreelec.tv/thread/23657-le-10-0b1-crash-after-profile-change/

### To Reproduce
1. Clean LE installation, accept defaults in wizard
2. Create one (or two) profiles in settings->profiles, accept defaults, enable "Show login screen on startup"
3. reboot LE
4. select master profile in login screen
5. log out of master profile
6. select first created profile in login screen
7. kodi crashes

### Informations
 - LE Version: LE 10b1, master
 - Hardware Platform: RPi4, forum users also reported Generic

### Log file
Kodi crash log: http://ix.io/2Vrs

### Additional context
The crash log points to  python's _asyncio.so lib which indicates the crash is probably caused by the LE settings addon.

Building an image without LE settings addon results in working profile switching without crashes.

Log file from switching between master and both test profiles back and forth several times: http://ix.io/2Vrv

lrusak, 2021-04-08:
Can you enable LE settings add-on debugging and post another log please?

HiassofT, 2021-04-09:
Here's a log with DEBUG=1 in kodi.conf: http://ix.io/2Vt3

lrusak, 2021-04-09:
it's a long shot but can you test this patch? https://termbin.com/y454

HiassofT, 2021-04-09:
the patch didn't help - kodi is still crashing http://ix.io/2VwX

lrusak, 2021-04-09:
Does it crash if you disable/enable the add-on?

HiassofT, 2021-04-09:
enabling/disabling isn't possible in kodi, probably because it's flagged as a system addon

[image]

antonlacon, 2021-05-11:
If I disable Bluetooth (Settings -> Services), I can switch profiles at will.

NeverEatYellowSwissSnow, 2021-05-18:
> 
> 
> If I disable Bluetooth (Settings -> Services), I can switch profiles at will.

Not in my case. Still crashing.

NeverEatYellowSwissSnow, 2021-05-21:
Still crashing on LE 10b3, x86 generic.

romanvm, 2021-06-16:
Multiple profiles support in Kodi is broken for some time and should be removed if someone finds time to do it.

NeverEatYellowSwissSnow, 2021-06-18:
> 
> 
> Multiple profiles support in Kodi is broken for some time and should be removed if someone finds time to do it.

What do you mean it's broken in Kodi? I have several Kodi 19.1 installations which are all running fine with multiple profiles. It's just crashing in LibreELEC, not in Kodi. Don't get me wrong but just because the code is bad doesn't mean it's broken?
It would be too sad if multiple profiles were removed, this is the main reason I am using Kodi/LibreELEC because that way I can manage all of my familiy members individually and keep kids away from things that do not belong to them.

HiassofT, 2021-07-30:
FYI: latest nightly is still crashing with the same _asyncio.so backtrace http://ix.io/3utf

CvH, 2021-08-01:
just a warning, looks like nobody is looking into the issue so LE10 final will have this broken

jfilaa, 2021-09-29:
Raspberry Pi TV HAT not working. RPi4 TV HAT is not detected.

HiassofT, 2023-09-07:
Looks like we just forgot to close the issue. Tested with LE master, profile logoff/logon works fine

closed 2023-09-07 by HiassofT